openwrt/packages #10329: OpenWrt 18.06 - python-cryptography fails due to Python2.7.1.7

BKPepe, 2019-10-22:
Maintainers: @jefferyto, @commodo  
Environment: OpenWrt 18.06, all Turris routers (aarch64_cortex-a53, arm_cortex-a9_vfpv3 and powerpc_8540)

Description:

After we updated Python2 to version python 2.7.17 in the stable branch, locally neither our build server can not compile python-cryptography. When I reverted the update, it works. It seems that this patch https://github.com/pyca/cryptography/pull/4234 should solve it. It doesn't apply and I'm trying to fix failed hunks.

Here is my output:
```
building '_openssl' extension
ccache_cc -fno-strict-aliasing -O2 -I/home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/host/include -I/home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/hostpkg/include -I/home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/target-powerpc_8540_musl/host/include -DNDEBUG -g -O3 -Wall -Wstrict-prototypes -O2 -pipe -mcpu=8540 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/pepe/work/turrisos/turris1x-hbk/build/build_dir/target-powerpc_8540_musl/python-cryptography-2.1.4:python-cryptography-2.1.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/target-powerpc_8540_musl/usr/include -I/home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/target-powerpc_8540_musl/include -I/home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/toolchain-powerpc_8540_gcc-7.3.0_musl/usr/include -I/home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/toolchain-powerpc_8540_gcc-7.3.0_musl/include/fortify -I/home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/toolchain-powerpc_8540_gcc-7.3.0_musl/include -I/home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/target-powerpc_8540_musl/usr/include/python2.7 -fPIC -I/home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/target-powerpc_8540_musl/usr/include/python2.7 -c build/temp.linux2-2.7/_openssl.c -o build/temp.linux2-2.7/build/temp.linux2-2.7/_openssl.o -Wconversion -Wno-error=sign-conversion
build/temp.linux2-2.7/_openssl.c: In function 'SSL_SESSION_get_master_key':
build/temp.linux2-2.7/_openssl.c:2202:23: warning: conversion to 'size_t {aka unsigned int}' from 'int' may change the sign of the result [-Wsign-conversion]
         return session->master_key_length;
                ~~~~~~~^~~~~~~~~~~~~~~~~~~
build/temp.linux2-2.7/_openssl.c:2204:18: warning: conversion to 'size_t {aka unsigned int}' from 'int' may change the sign of the result [-Wsign-conversion]
         outlen = session->master_key_length;
                  ^~~~~~~
build/temp.linux2-2.7/_openssl.c: At top level:
build/temp.linux2-2.7/_openssl.c:2342:6: error: conflicting types for 'X509_get0_signature'
 void X509_get0_signature(ASN1_BIT_STRING **psig, X509_ALGOR **palg,
      ^~~~~~~~~~~~~~~~~~~
In file included from /home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/host/include/openssl/engine.h:96:0,
                 from build/temp.linux2-2.7/_openssl.c:577:
/home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/host/include/openssl/x509.h:919:6: note: previous declaration of 'X509_get0_signature' was here
 void X509_get0_signature(const ASN1_BIT_STRING **psig,
      ^~~~~~~~~~~~~~~~~~~
build/temp.linux2-2.7/_openssl.c:2476:7: error: 'X509_VERIFY_PARAM_set1_host' redeclared as different kind of symbol
 int (*X509_VERIFY_PARAM_set1_host)(X509_VERIFY_PARAM *, const char *,
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/host/include/openssl/x509.h:595:0,
                 from /home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/host/include/openssl/engine.h:96,
                 from build/temp.linux2-2.7/_openssl.c:577:
/home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/host/include/openssl/x509_vfy.h:564:5: note: previous declaration of 'X509_VERIFY_PARAM_set1_host' was here
 int X509_VERIFY_PARAM_set1_host(X509_VERIFY_PARAM *param, const char *name,
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
build/temp.linux2-2.7/_openssl.c:2478:7: error: 'X509_VERIFY_PARAM_set1_email' redeclared as different kind of symbol
 int (*X509_VERIFY_PARAM_set1_email)(X509_VERIFY_PARAM *, const char *,
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/host/include/openssl/x509.h:595:0,
                 from /home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/host/include/openssl/engine.h:96,
                 from build/temp.linux2-2.7/_openssl.c:577:
/home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/host/include/openssl/x509_vfy.h:571:5: note: previous declaration of 'X509_VERIFY_PARAM_set1_email' was here
 int X509_VERIFY_PARAM_set1_email(X509_VERIFY_PARAM *param,  const char *email,
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
build/temp.linux2-2.7/_openssl.c:2480:7: error: 'X509_VERIFY_PARAM_set1_ip' redeclared as different kind of symbol
 int (*X509_VERIFY_PARAM_set1_ip)(X509_VERIFY_PARAM *, const unsigned char *,
       ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/host/include/openssl/x509.h:595:0,
                 from /home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/host/include/openssl/engine.h:96,
                 from build/temp.linux2-2.7/_openssl.c:577:
/home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/host/include/openssl/x509_vfy.h:573:5: note: previous declaration of 'X509_VERIFY_PARAM_set1_ip' was here
 int X509_VERIFY_PARAM_set1_ip(X509_VERIFY_PARAM *param, const unsigned char *ip,
     ^~~~~~~~~~~~~~~~~~~~~~~~~
build/temp.linux2-2.7/_openssl.c:2482:7: error: 'X509_VERIFY_PARAM_set1_ip_asc' redeclared as different kind of symbol
 int (*X509_VERIFY_PARAM_set1_ip_asc)(X509_VERIFY_PARAM *, const char *) = NULL;
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/host/include/openssl/x509.h:595:0,
                 from /home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/host/include/openssl/engine.h:96,
                 from build/temp.linux2-2.7/_openssl.c:577:
/home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/host/include/openssl/x509_vfy.h:575:5: note: previous declaration of 'X509_VERIFY_PARAM_set1_ip_asc' was here
 int X509_VERIFY_PARAM_set1_ip_asc(X509_VERIFY_PARAM *param, const char *ipasc);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
build/temp.linux2-2.7/_openssl.c:2483:8: error: 'X509_VERIFY_PARAM_set_hostflags' redeclared as different kind of symbol
 void (*X509_VERIFY_PARAM_set_hostflags)(X509_VERIFY_PARAM *,
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/host/include/openssl/x509.h:595:0,
                 from /home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/host/include/openssl/engine.h:96,
                 from build/temp.linux2-2.7/_openssl.c:577:
/home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/host/include/openssl/x509_vfy.h:568:6: note: previous declaration of 'X509_VERIFY_PARAM_set_hostflags' was here
[... 307 lines cut ...]
build/temp.linux2-2.7/_openssl.c:53863:10: note: in expansion of macro 'SSL_set_options'
   return SSL_set_options(x0, x1);
          ^~~~~~~~~~~~~~~
/home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/host/include/openssl/ssl.h:577:2: warning: conversion to 'long unsigned int' from 'long int' may change the sign of the result [-Wsign-conversion]
  SSL_ctrl((ssl),SSL_CTRL_OPTIONS,(op),NULL)
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
build/temp.linux2-2.7/_openssl.c:53863:10: note: in expansion of macro 'SSL_set_options'
   return SSL_set_options(x0, x1);
          ^~~~~~~~~~~~~~~
build/temp.linux2-2.7/_openssl.c: In function '_cffi_f_SSL_set_options':
/home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/host/include/openssl/ssl.h:577:34: warning: conversion to 'long int' from 'long unsigned int' may change the sign of the result [-Wsign-conversion]
  SSL_ctrl((ssl),SSL_CTRL_OPTIONS,(op),NULL)
                                  ^
build/temp.linux2-2.7/_openssl.c:53896:14: note: in expansion of macro 'SSL_set_options'
   { result = SSL_set_options(x0, x1); }
              ^~~~~~~~~~~~~~~
/home/pepe/work/turrisos/turris1x-hbk/build/staging_dir/host/include/openssl/ssl.h:577:2: warning: conversion to 'long unsigned int' from 'long int' may change the sign of the result [-Wsign-conversion]
  SSL_ctrl((ssl),SSL_CTRL_OPTIONS,(op),NULL)
  ^
build/temp.linux2-2.7/_openssl.c:53896:14: note: in expansion of macro 'SSL_set_options'
   { result = SSL_set_options(x0, x1); }
              ^~~~~~~~~~~~~~~
error: command 'ccache_cc' failed with exit status 1
make[2]: *** [Makefile:82: /home/pepe/work/turrisos/turris1x-hbk/build/build_dir/target-powerpc_8540_musl/python-cryptography-2.1.4/.built] Error 1
```

commodo, 2019-10-22:
i'll try to take a look
maybe we just need to update openssl?

BKPepe, 2019-10-22:
I was able to compile it locally with this commit https://github.com/BKPepe/packages/commit/cb42523ae1277af00b9255b3ad6dc1f23ea999b6 , which I've pushed to [Turris build bots](https://gitlab.labs.nic.cz/turris/turris-build/commit/1099b5c80000723a7c926a4f5e0cf0e1923b057a), so let's see.

commodo, 2019-10-22:
ack
thanks for the quick-find :)

jefferyto, 2019-10-22:
Would it be better if we updated cryptography to 2.3 (or 2.3.1)? pyca/cryptography#4234 was subsumed by pyca/cryptography#4270, which according to the [changelog](https://github.com/pyca/cryptography/blob/master/CHANGELOG.rst#23---2018-07-18) should be in 2.3. Also 2.3 has a fix for a security issue ([CVE-2018-10903](https://nvd.nist.gov/vuln/detail/CVE-2018-10903)).

@BKPepe I think BKPepe@cb42523 is from pyca/cryptography#4270?

jefferyto, 2019-10-22:
Actually I'm ok with backporting patches for both issues as well if it's not too difficult.

BKPepe, 2019-10-22:
@jefferyto yes, the commit is from the pr, but yeah we can update it to 2.3.1.

jefferyto, 2019-10-22:
Hmm now I'm leaning more towards just backporting patches :joy: we should keep 18.06 as frozen as possible.

jefferyto, 2019-10-22:
@BKPepe I think pyca/cryptography#4342 is the fix for the security issue - would you mind trying/adapting the patch? Then we can include both patches and increment `PKG_RELEASE` only once.

BKPepe, 2019-10-22:
I will try.

BKPepe, 2019-10-22:
Ok, the patch (https://github.com/BKPepe/packages/commit/c6f626d985fc412325aac5d9bf5121128f9b8f0c) applies without a hassle, but I would like to run test both things before I sent a PR.

jefferyto, 2019-10-22:
@BKPepe Thanks for your quick action on this :smile:

neheb, 2019-10-22:
@BKPepe That issue in the beginning looks like you have OpenSSL 1.1.0 in staging_dir while python-cryptography with that version does not support it.

BKPepe, 2019-10-22:
@neheb This issue appeared since an update of Python2 in `openwrt-18.06` branch locally and our Turris build bots. When I reverted Python update, it works for me. Also, the same issue appears on our build bots, so I'm pretty sure that this is going to happen also on OpenWrt build bots. As patch works for compilation, I'm fine with that. However, I didn't run tested it, yet.

neheb, 2019-10-22:
Just looked again. It seems that commit is correct. LibreSSL is used in tools, and hence host:

```
/builder/shared-workdir/build/sdk/staging_dir/host/include/openssl/x509.h:919:6: note: previous declaration of 'X509_get0_signature' was here
 void X509_get0_signature(const ASN1_BIT_STRING **psig,
```

Which reminds me of a false dependency of one other package...

jefferyto, 2019-10-22:
OpenWrt buildbots hit these errors as well, e.g. [mips_24kc](https://downloads.openwrt.org/releases/faillogs-18.06/mips_24kc/packages/python-cryptography/python/compile.txt), [arm_cortex-a5_neon-vfpv4](https://downloads.openwrt.org/releases/faillogs-18.06/arm_cortex-a5_neon-vfpv4/packages/python-cryptography/python/compile.txt)

closed 2019-10-25 by BKPepe