openwrt/packages #10332: <ss> does not build on master

brucealthompson, 2019-10-22:
Maintainer: @\Russell Senior <russell@personaltelco.net>
Environment: (Marvell EBU Armada, Marvell Armada 37x/38x/XP, Linksys WRT1900ACS (Shelby), master)
Description:
I am trying to build the ss package:
https://github.com/openwrt/openwrt/blob/master/package/network/utils/iproute2/Makefile
I get the following error during the build of this package:
Package ss is missing dependencies for the following libraries:
libcap.so.2

I have attached the output of make -j1 V=sc
[make.out.zip](https://github.com/openwrt/packages/files/3756243/make.out.zip)

neheb, 2019-10-24:
ping @RussellSenior

jefferyto, 2019-10-25:
iproute2 is a base package - issues should be reported to https://bugs.openwrt.org/.

Also note that buildbots are building the package just fine: https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3/base/ss_5.3.0-1_arm_cortex-a9_vfpv3.ipk

neheb, 2019-10-25:
@jefferyto that failure is probably related to https://github.com/openwrt/packages/commit/d3908aee91629ed36a4d1a235db2b0b75b90f333 ;)

neheb, 2019-11-04:
I can't replicate. This is probably similar to the Samba problem where selecting libcap and then unselecting causes this error. In your case, it looks like you selected ip-full, built, deselected, and tried building again.

Clean your environment with make clean and try again.

closed 2019-11-04 by neheb

brucealthompson, 2019-11-09:
Rosen,

Clean your environment with make clean and try again.

Thanks greatly for looking into this. I tried a “make clean” and the problem still occurred. I next built a new environment from scratch and the problem is now gone. Everything seems to be working.

 

                Bruce T