openwrt/packages #10334: bluez-daemon page_fault() sending SIGSEGV

BLLuis, 2019-10-22:
Maintainer: @Nicolas Thill <nico@openwrt.org>
Environment: (AR7, TP-LINK mr 3020, OpenWrt 18.06-SNAPSHOT r7885-3b8db97a52)
Description:
Random  "do_page_fault(): sending SIGSEGV to bluetoothd for invalid read access from 00000004"
Some steps to reproduce if not appear:

root@demo:/etc# bluetoothctl
Agent registered

[bluetooth]# power on
[CHG] Controller 00:19:B5:F2:1A:59 Class: 0x00000414
Changing power on succeeded
[CHG] Controller 00:19:B5:F2:1A:59 Powered: yes

[bluetooth]# pairable on
Changing pairable on succeeded

[bluetooth]# discoverable on
Changing discoverable on succeeded
Agent unregistered
[DEL] Controller 10:29:B3:A2:1B:42 DemoSpk [default]
Waiting to connect to bluetoothd...

dmesg log:
[ 1344.189306] do_page_fault(): sending SIGSEGV to bluetoothd for invalid read access from 00000004
[ 1344.196655] epc = 0042d2f5 in bluetoothd[400000+77000]
[ 1344.201759] ra  = 004348fd in bluetoothd[400000+77000]

logread:
Tue Oct 22 19:29:45 2019 daemon.info bluetoothd[1467]: Bluetooth daemon 5.49
Tue Oct 22 19:29:45 2019 daemon.info bluetoothd[1467]: Starting SDP server
Tue Oct 22 19:29:46 2019 daemon.err bluetoothd[1467]: Failed to open RFKILL control device
Tue Oct 22 19:29:46 2019 daemon.info bluetoothd[1467]: Bluetooth management interface 1.14 initialized
Tue Oct 22 19:32:39 2019 kern.info kernel: [ 1344.189306] do_page_fault(): sending SIGSEGV to bluetoothd for invalid read access from 00000004
Tue Oct 22 19:32:39 2019 kern.info kernel: [ 1344.196655] epc = 0042d2f5 in bluetoothd[400000+77000]
Tue Oct 22 19:32:39 2019 kern.info kernel: [ 1344.201759] ra  = 004348fd in bluetoothd[400000+77000]

neheb, 2019-10-22:
ping @diizzyy @br101

neheb, 2019-10-22:
would be useful to use strace

BLLuis, 2019-10-23:
My first time with strace, let me know if you need in some other format.

[STraceWirelessSPK.txt](https://github.com/openwrt/packages/files/3759623/STraceWirelessSPK.txt)

neheb, 2019-10-23:
I see a bunch of

recvmsg(4, {msg_namelen=0}, MSG_CMSG_CLOEXEC) = -1 EAGAIN (Resource temporarily unavailable)

I don't use bluetooth so no idea.

diizzyy, 2019-10-23:
Please use/test master branch of OpenWrt and packages instead

BLLuis, 2019-10-23:
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='18.06.4'
DISTRIB_REVISION='r7808-ef686b7292'
DISTRIB_TARGET='ar71xx/tiny'
DISTRIB_ARCH='mips_24kc'

Same problem:

root@wirelessspk:~# bluetoothctl
Agent registered
[bluetooth]# power on
[CHG] Controller 00:19:B5:F2:1A:59 Class: 0x00000414
Changing power on succeeded
[CHG] Controller 00:19:B5:F2:1A:59 Powered: yes
[bluetooth]# pairable on
Changing pairable on succeeded
[bluetooth]# discoverable on
Changing discoverable on succeeded
Agent unregistered
[DEL] Controller 00:19:B5:F2:1A:59 WirelessSpk [default]
Waiting to connect to bluetoothd...

[  274.596536] do_page_fault(): sending SIGSEGV to bluetoothd for invalid read access from 00000004
[  274.603927] epc = 0042d2f5 in bluetoothd[400000+77000]
[  274.608999] ra  = 004348fd in bluetoothd[400000+77000]

BLLuis, 2019-10-23:
root@wirelessspk:/etc/init.d# /usr/bin/bluetoothd -n -d
bluetoothd[1523]: Bluetooth daemon 5.49
bluetoothd[1523]: src/main.c:parse_config() parsing /etc/bluetooth/main.conf
bluetoothd[1523]: src/main.c:parse_config() Key file does not have key “DiscoverableTimeout” in group “General”
bluetoothd[1523]: src/main.c:parse_config() Key file does not have key “PairableTimeout” in group “General”
bluetoothd[1523]: src/main.c:parse_config() Key file does not have key “Privacy” in group “General”
bluetoothd[1523]: src/main.c:parse_config() name=WirelessSpk
bluetoothd[1523]: src/main.c:parse_config() class=0x240414
bluetoothd[1523]: src/main.c:parse_config() Key file does not have key “DeviceID” in group “General”
bluetoothd[1523]: src/main.c:parse_config() Key file does not have key “ReverseServiceDiscovery” in group “General”
bluetoothd[1523]: src/adapter.c:adapter_init() sending read version command
bluetoothd[1523]: Starting SDP server
bluetoothd[1523]: src/sdpd-service.c:register_device_id() Adding device id record for 0002:1d6b:0246:0531
bluetoothd[1523]: src/plugin.c:plugin_init() Loading builtin plugins
bluetoothd[1523]: src/plugin.c:add_plugin() Loading hostname plugin
bluetoothd[1523]: src/plugin.c:add_plugin() Loading wiimote plugin
bluetoothd[1523]: src/plugin.c:add_plugin() Loading autopair plugin
bluetoothd[1523]: src/plugin.c:add_plugin() Loading policy plugin
bluetoothd[1523]: src/plugin.c:add_plugin() Loading a2dp plugin
bluetoothd[1523]: src/plugin.c:add_plugin() Loading avrcp plugin
bluetoothd[1523]: src/plugin.c:add_plugin() Loading network plugin
bluetoothd[1523]: src/plugin.c:add_plugin() Loading input plugin
bluetoothd[1523]: src/plugin.c:add_plugin() Loading hog plugin
bluetoothd[1523]: src/plugin.c:add_plugin() Loading gap plugin
bluetoothd[1523]: src/plugin.c:add_plugin() Loading scanparam plugin
bluetoothd[1523]: src/plugin.c:add_plugin() Loading deviceinfo plugin
bluetoothd[1523]: src/plugin.c:add_plugin() Loading battery plugin
bluetoothd[1523]: src/plugin.c:plugin_init() Loading plugins /usr/lib/bluetooth/plugins
bluetoothd[1523]: profiles/input/suspend-none.c:suspend_init() 
bluetoothd[1523]: profiles/network/manager.c:read_config() /etc/bluetooth/network.conf: Key file does not have key “DisableSecurity” in group “General”
bluetoothd[1523]: profiles/network/manager.c:read_config() Config options: Security=true
bluetoothd[1523]: src/main.c:main() Entering main loop
bluetoothd[1523]: src/rfkill.c:rfkill_event() RFKILL event idx 0 type 2 op 0 soft 0 hard 0
bluetoothd[1523]: Bluetooth management interface 1.14 initialized
bluetoothd[1523]: src/adapter.c:read_version_complete() sending read supported commands command
bluetoothd[1523]: src/adapter.c:read_version_complete() sending read index list command
bluetoothd[1523]: src/adapter.c:read_commands_complete() Number of commands: 65
bluetoothd[1523]: src/adapter.c:read_commands_complete() Number of events: 35
bluetoothd[1523]: src/adapter.c:read_commands_complete() enabling kernel-side connection control
bluetoothd[1523]: src/adapter.c:read_index_list_complete() Number of controllers: 1
bluetoothd[1523]: src/adapter.c:read_index_list_complete() Found index 0
bluetoothd[1523]: src/adapter.c:index_added() index 0
bluetoothd[1523]: src/adapter.c:btd_adapter_new() System name: WirelessSpk
bluetoothd[1523]: src/adapter.c:btd_adapter_new() Major class: 4
bluetoothd[1523]: src/adapter.c:btd_adapter_new() Minor class: 5
bluetoothd[1523]: src/adapter.c:btd_adapter_new() Modalias: usb:v1D6Bp0246d0531
bluetoothd[1523]: src/adapter.c:btd_adapter_new() Discoverable timeout: 180 seconds
bluetoothd[1523]: src/adapter.c:btd_adapter_new() Pairable timeout: 0 seconds
bluetoothd[1523]: src/adapter.c:index_added() sending read info command for index 0
bluetoothd[1523]: src/adapter.c:read_info_complete() index 0 status 0x00
bluetoothd[1523]: src/adapter.c:clear_uuids() sending clear uuids command for index 0
bluetoothd[1523]: src/adapter.c:clear_devices() sending clear devices command for index 0
bluetoothd[1523]: profiles/audio/a2dp.c:media_server_probe() path /org/bluez/hci0
bluetoothd[1523]: plugins/hostname.c:hostname_probe() 
bluetoothd[1523]: profiles/network/manager.c:panu_server_probe() path /org/bluez/hci0
bluetoothd[1523]: profiles/network/server.c:server_register() Registered interface org.bluez.NetworkServer1 on path /org/bluez/hci0
bluetoothd[1523]: profiles/network/manager.c:gn_server_probe() path /org/bluez/hci0
bluetoothd[1523]: profiles/network/manager.c:nap_server_probe() path /org/bluez/hci0
bluetoothd[1523]: profiles/audio/avrcp.c:avrcp_controller_server_probe() path /org/bluez/hci0
bluetoothd[1523]: src/adapter.c:adapter_service_add() /org/bluez/hci0
bluetoothd[1523]: src/sdpd-service.c:add_record_to_server() Adding record with handle 0x10001
bluetoothd[1523]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00000017-0000-1000-8000-00805f9
bluetoothd[1523]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00000100-0000-1000-8000-00805f9
bluetoothd[1523]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00001002-0000-1000-8000-00805f9
bluetoothd[1523]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 0000110e-0000-1000-8000-00805f9
bluetoothd[1523]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 0000110f-0000-1000-8000-00805f9
bluetoothd[1523]: src/adapter.c:adapter_service_insert() /org/bluez/hci0
[... cut ...]
bluetoothd[1523]: src/adapter.c:dev_class_changed_callback() Class: 0x000414
bluetoothd[1523]: src/adapter.c:new_settings_callback() Settings: 0x00000091
bluetoothd[1523]: src/adapter.c:settings_changed() Changed settings: 0x00000001
bluetoothd[1523]: src/adapter.c:adapter_start() adapter /org/bluez/hci0 has been enabled
bluetoothd[1523]: src/adapter.c:new_settings_callback() Settings: 0x0000009b

then discoverable on

bluetoothd[1523]: src/adapter.c:settings_changed() Changed settings: 0x0000000a
Segmentation fault

br101, 2019-10-23:
hi! yes, please test with bluez version 5.51...
i tried the same commands on ar71xx with bluez 5.51 and dont see this... but i don't know how do you connect to the device?

diizzyy, 2019-10-25:
5.50 worked just fine with ar71xx also fwiw

BLLuis, 2019-10-25:
im still working on 5.51, not finish my tests yet because pulseaudio, at least don't SIGSEGV on discoverable on, but, there are some random errors on commands timeout and I have no audio yet on speaker, even the profile is good on bluetooth on one bluetooth adapter and some other errors on another adapter.

diizzyy, 2019-10-25:
Since you're on a relatively low-end device, you might want to use https://github.com/diizzyy/packages/tree/sound-bluez-alsa/sound/bluez-alsa although you probably want ot bump it to a newer commit.

BLLuis, 2019-10-26:
> Since you're on a relatively low-end device, you might want to use https://github.com/diizzyy/packages/tree/sound-bluez-alsa/sound/bluez-alsa although you probably want ot bump it to a newer commit.

Thank you diizzyy, This very low end device can handle this, plus upnp, samba, sftp, pulseaudio and some other things, maybe not all at the same time, but.... Anyway I will try it, but as I far I understand they have removed a2dp from bluez and now is handled by pulseaudio, so, If I use bluez to alsa what about the a2dp profile?

BLLuis, 2019-10-26:
Well, Im on bluez 5.51 now and running but bluetoothclt "agent NoInputNoOutput" still ask for pincode or confirmation code (changes when I pair different devices, some of them ask me for pair code and another one shows me a random code) then when I play music ask about Authorize Service for profiles, I use to have this automated with old python/dbus agent scripts in the old version of my wirelessspk, but they are removed from master branch and there is no python dbus with the last versions of OpenWrt. What can I do to make this pairable without user interaction? Write my own dbus code? Is there any other utility to do this?

BLLuis, 2020-02-01:
News about what topic, there are many in this post, the last one, I have to manually pair each time I connect, so, because I have no idea how to do in an automated way as I do before, I left the Bluetooth audio once again aside of my wireless speakers, not so very important to me, mostly because I use to play music via WiFi and not too many times via Bluetooth, but I want to finish/close my project and the Bluetooth part is always a big pending. All other topics of this thread are solved in some ways.

closed 2021-08-19 by BLLuis