openwrt/packages #10335: bluez-utils or bluez-examples No Python DBus available

BLLuis, 2019-10-22:
Maintainer:  Nicolas Thill <nico@openwrt.org>
Environment: (AR7, TP-LINK MR-3020, 'OpenWrt 18.06-SNAPSHOT r7885-3b8db97a52')
Description:
Almost all python examples or utilities like simple-agent depends on python-dbus but the package is not available in current distribution.
I migrated my system from a very old OpenWrt version to current and I can pair bluetooth devices, not from OpenWrt to device, just the device (Phone) find the OpenWrt bluetooth adapter and pair with it, with the old version I use to do it with simple-agent script, an unattended way to pair phones with my Bluetooth speaker (OpenWrt) but now I can't find a way to do the same and python examples doesn't work because of dbus

python /usr/bin/bluez/simple-agent
Traceback (most recent call last):
  File "simple-agent", line 7, in <module>
    import dbus
ImportError: No module named dbus

If there are no dbus available for python, examples or utilities should be removed from distro or provide another way to pair devices or use the examples.

Is there another way to pair? Any example?

Regards
Luis

neheb, 2019-10-22:
ping @diizzyy @br101

diizzyy, 2019-10-23:
Not a bug, it's simply not ported
You can pair just fine in 5.50+ using bluetoothctl

jefferyto, 2019-10-24:
Also the examples were removed from master: #9094

diizzyy, 2019-10-25:
..and by that, I'm closing this.

closed 2019-10-25 by diizzyy