openwrt/packages #10338: telldus-core: compile failure on x86/64

stintel, 2019-10-23:
Maintainer: @PeterFromSweden 
Environment: OpenWrt master r11314-d263883796 x86/64

Description:

Package telldus-core fails to compile on x86/64:

```
make[2]: Entering directory '/home/stijn/Development/LEDE/openwrt-packages/utils/telldus-core'
touch /home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/telldus-core-2.1.2/.prepared_74713b46fe116eff737abc33e00ed279_6664517399ebbbc92a37c5bb081b5c53_check
. /home/stijn/Development/LEDE/source/include/shell.sh; gzip -dc /home/stijn/Development/LEDE/source/dl/telldus-core-2.1.2.tar.gz | tar -C /home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/telldus-core-2.1.2/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/telldus-core-2.1.2

Applying ./patches/100-add_includes.patch using plaintext: 
patching file common/Socket_unix.cpp
patching file service/ConnectionListener_unix.cpp

Applying ./patches/110-fix_warnings.patch using plaintext: 
patching file service/ProtocolIkea.cpp
patching file service/ProtocolX10.cpp

Applying ./patches/120-fix_missing_var_conf.patch using plaintext: 
patching file service/SettingsConfuse.cpp

Applying ./patches/900-openwrt_fixes_cmake.patch using plaintext: 
patching file service/CMakeLists.txt
patching file tdadmin/CMakeLists.txt
patching file common/CMakeLists.txt

Applying ./patches/920-openwrt_missing_var_conf.patch using plaintext: 
patching file service/SettingsConfuse.cpp

Applying ./patches/930-usleep.patch using plaintext: 
patching file common/common.h
patching file service/TellStick_libftdi.cpp
touch /home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/telldus-core-2.1.2/.prepared_74713b46fe116eff737abc33e00ed279_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/telldus-core-2.1.2/.configured_*
rm -f /home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/stamp/.telldus-core_installed
mkdir -p /home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/telldus-core-2.1.2
(cd /home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/telldus-core-2.1.2; CFLAGS="-O2 -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/telldus-core-2.1.2=telldus-core-2.1.2 -Wformat -Werror=format-security -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -I/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/include -I/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/include " CXXFLAGS="-O2 -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/telldus-core-2.1.2=telldus-core-2.1.2 -Wformat -Werror=format-security -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -I/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/include -I/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/include  " LDFLAGS="-L/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib -L/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/lib -L/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/usr/lib -L/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/lib -znow -zrelro -L/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib -Wl,-rpath-link=/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib -L/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/lib -Wl,-rpath-link=/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/lib " cmake -DCMAKE_SYSTEM_NAME=Linux -DCMAKE_SYSTEM_VERSION=1 -DCMAKE_SYSTEM_PROCESSOR=x86_64 -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_CXX_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_C_COMPILER="/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/bin/x86_64-openwrt-linux-musl-gcc" -DCMAKE_C_COMPILER_ARG1="" -DCMAKE_CXX_COMPILER="/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/bin/x86_64-openwrt-linux-musl-g++" -DCMAKE_CXX_COMPILER_ARG1="" -DCMAKE_ASM_COMPILER="/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/bin/x86_64-openwrt-linux-musl-gcc" -DCMAKE_ASM_COMPILER_ARG1="" -DCMAKE_EXE_LINKER_FLAGS:STRING="-L/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib -L/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/lib -L/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/usr/lib -L/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/lib -znow -zrelro -L/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib -Wl,-rpath-link=/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib -L/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/lib -Wl,-rpath-link=/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/lib" -DCMAKE_MODULE_LINKER_FLAGS:STRING="-L/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib -L/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/lib -L/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/usr/lib -L/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/lib -znow -zrelro -L/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib -Wl,-rpath-link=/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib -L/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/lib -Wl,-rpath-link=/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/lib -Wl,-Bsymbolic-functions" -DCMAKE_SHARED_LINKER_FLAGS:STRING="-L/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib -L/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/lib -L/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/usr/lib -L/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/lib -znow -zrelro -L/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib -Wl,-rpath-link=/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib -L/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/lib -Wl,-rpath-link=/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/lib -Wl,-Bsymbolic-functions" -DCMAKE_AR="/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/bin/x86_64-openwrt-linux-musl-gcc-ar" -DCMAKE_NM="/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/bin/x86_64-openwrt-linux-musl-gcc-nm" -DCMAKE_RANLIB="/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/bin/x86_64-openwrt-linux-musl-gcc-ranlib" -DCMAKE_FIND_ROOT_PATH="/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr;/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl" -DCMAKE_FIND_ROOT_PATH_MODE_PROGRAM=BOTH -DCMAKE_FIND_ROOT_PATH_MODE_LIBRARY=ONLY -DCMAKE_FIND_ROOT_PATH_MODE_INCLUDE=ONLY -DCMAKE_STRIP=: -DCMAKE_INSTALL_PREFIX=/usr -DDL_LIBRARY=/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl -DCMAKE_PREFIX_PATH=/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl -DCMAKE_SKIP_RPATH=TRUE -DBUILD_LIBTELLDUS-CORE=1 -DBUILD_TDTOOL=1 -DGENERATE_MAN=0 -DICONV_LIBRARY=-liconv /home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/telldus-core-2.1.2 )
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/bin/x86_64-openwrt-linux-musl-gcc
[... cut ...]
make[2]: *** [Makefile:72: /home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/telldus-core-2.1.2/.configured_a17fb5ef857664f03cd0ce37cc5ea591] Error 1
make[2]: Leaving directory '/home/stijn/Development/LEDE/openwrt-packages/utils/telldus-core'
time: package/feeds/packages/telldus-core/compile#0.98#0.60#1.53
make[1]: *** [package/Makefile:113: package/feeds/packages/telldus-core/compile] Error 2
make[1]: Leaving directory '/home/stijn/Development/LEDE/source'
make: *** [/home/stijn/Development/LEDE/source/include/toplevel.mk:227: package/telldus-core/compile] Error 2
```

neheb, 2019-10-25:
Does adding this to CMAKE_OPTIONS work?

```
-DDOXYGEN_EXECUTABLE:FILEPATH=DOXYGEN_EXECUTABLE-NOTFOUND
```

stintel, 2019-10-29:
> Does adding this to CMAKE_OPTIONS work?
> 
> ```
> -DDOXYGEN_EXECUTABLE:FILEPATH=DOXYGEN_EXECUTABLE-NOTFOUND
> ```

Unfortunately it does not.

closed 2019-10-29 by neheb