openwrt/packages #10357: minidlna: service stops after a file is added

craibuc, 2019-10-25:
Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (`arm_cortex-a9_vfpv3`, `Turris Omnia`
[minidlna.log](https://github.com/openwrt/packages/files/3773031/minidlna.log)
, `OpenWrt 18.06`/`TurrisOS 4.0.1`)

Description:

Upgraded to Turris OS 4.  Using minidlna version 1.2.1.

Adding a file (via SFTP) to one of the folders that are monitored for videos caused the service to stop with an error (in Luci) that reads `The miniDLNA service is not running.`

Restarting the service (`# service minidlna restart`) fixes the problem.

`/var/log/minidlna.log` attached.

References:
- https://gitlab.labs.nic.cz/turris/openwrt/issues/322

diizzyy, 2019-10-25:
I'd recommend you to test master branch of both OpenWrt and package repo but as you're running TurrisOS which isn't supported by OpenWrt you have to ask the maintainers of TurrisOS for support.

BKPepe, 2019-10-28:
@diizzyy I'm thinking that right now, the user is confused, where he should report an issue. I told him to report it here if you are willing to take a look at the cross-reference link as both issues are closed and they are not getting help nor fixed.

(OT: Turris OS 4.x is based on top of OpenWrt with our patches and our minimalistic feed with Turris specific things. You can take a look at [git-hash](https://repo.turris.cz/hbs/omnia/packages/git-hash) file from each branch to see from which commits it was created). The package comes from this feed without any further modifications nor patches.  It comes as it is from this repository. So, what he did is right. Also, as you can see you can ping me first as I'm contributing to both projects without you doing any premature conclusion. ;-) We should help users as much as possible. It should be investigated where the issue is and how we can fix it in the stable branch. On the other thing, test it on master branches might be a good idea and can save us a lot of time, but I suggested it as well.

I was able to find at OpenWrt forum that many users have similar or even the same problem as it was reported to us, but only @craibuc aware us on this problem here.  E.g.
https://forum.openwrt.org/t/resolved-dlna-not-serving-any-files/34111/7

Anyway, the maintainer of this package, which is @medaved  should be notified first about it.

craibuc, 2019-10-28:
> (OT: Turris OS 4.x is based on top of OpenWrt with our patches and our minimalistic feed with Turris specific things. You can take a look at git-hash file from each branch to see from which commits [...]
It would be useful to reference each package's source from the Turris OS repository.  It would help eliminate confusion.

> We should help users as much as possible. It should be investigated where the issue is and how we can fix it in the stable branch. On the other thing, test it on master branches might be a good idea and can save us a lot of time, but I suggested it as well.
If you are willing to explain how to do this, I'm happy to test it.  Relevant: [Opkg: install a package from a specified OpenWrt branch](https://forum.turris.cz/t/opkg-install-a-package-from-a-specified-openwrt-branch/11360)

diizzyy, 2019-10-28:
@BKPepe 
Either way I doubt anyone will spend time on 18.06

neheb, 2019-11-04:
@BKPepe as with most packages here, miniDLNA is unmaintained(actually, even upstream it's a similar story). @medaved 's last commit was from 2014.

diizzyy, 2019-11-04:
The current version of minidlna in the tree is much newer than the version in 18.X and having an by now rather old version of ffmpeg in most likely doesn't improve the user experience. Not to forget all other libraries used such as musl etc.

neheb, 2019-11-04:
Initial port of gerbera to OpenWrt gives a size of ~500KB in addition to having a lot more dependencies.

diizzyy, 2019-11-04:
It's not much worse and size isn't much of an issue as the devices you'd run it in have ample of space.
https://github.com/Entware/rtndev/blob/master/gerbera/Makefile#L27

craibuc, 2020-01-12:
So, `minidlna` isn't going to function properly on Turris' OS 4 because it is out-of-date?

If I wanted to install `gerbera`, what's the process of doing so on TOS4?  Do I need to add a custom feed to http://turris.lan/cgi-bin/luci/admin/system/packages/ipkg?

neheb, 2020-01-12:
It doesn't function properly because there's no maintainer for this package.

BKPepe, 2020-01-12:
> edit2: actually the Turris people can add it if they want.

I think Entware guys (@ryzhovau) can share it as they already have it packaged for OpenWrt. See the Makefile, which @diizzyy mentioned.

craibuc, 2020-01-12:
@BKPepe, how do I run the makefile?  Is it a matter of ssh'ing to the router, downloading the makefile, then running it in `make`?

neheb, 2020-01-12:
That Makefile will not work properly here.

craibuc, 2020-01-14:
> > edit2: actually the Turris people can add it if they want.
> 
> I think Entware guys (@ryzhovau) can share it as they already have it packaged for OpenWrt. See the Makefile, which @diizzyy mentioned.

Is this it?  https://github.com/samunders-core/gerbera-feed

neheb, 2020-01-14:
Looks like an outdated fees, but yes.

craibuc, 2020-01-20:
Still present as of `1.2.1-3.41`.

neheb, 2020-01-22:
Can you try to see if https://github.com/openwrt/packages/pull/11004 fixes your issue?

neheb, 2020-02-10:
I replicated this but have no idea how.

@craibuc can you run strace with minidlna?

type ps | grep minidlna to get the PID and then

strace -p XXXX to trace.

craibuc, 2020-02-12:
```bash
# ps | grep minidlna
```
Doesn't return anything.

Log file, if that's helpful.

```
# cat /var/log/minidlna.log 
[2020/02/11 20:09:55] minidlna.c:479: debug: Using locale dir '/usr/share/locale' and locale langauge C/C.UTF-8
[2020/02/11 20:09:55] minidlna.c:1048: warn: Starting MiniDLNA version 1.2.1.
[2020/02/11 20:09:55] minidlna.c:346: warn: Creating new database at /tmp/run/minidlna/files.db
[2020/02/11 20:09:55] scanner.c:731: info: Scanning /mnt/raid/Media/Movies
[2020/02/11 20:09:55] minidlna.c:133: error: bind(http): Address in use
[2020/02/11 20:09:55] minidlna.c:1088: fatal: Failed to open socket for HTTP. EXITING
[2020/02/11 20:09:55] metadata.c:816: debug: Container: 'matroska,webm' [X.mkv]
[2020/02/11 20:09:55] metadata.c:816: debug: Container: 'matroska,webm' [Y.mkv]
[2020/02/11 20:09:55] metadata.c:816: debug: Container: 'matroska,webm' [Z.mkv]
[2020/02/11 20:09:55] albumart.c:258: debug: Found new embedded album art in Z.mkv
[2020/02/11 20:09:55] metadata.c:816: debug: Container: 'mov,mp4,m4a,3gp,3g2,mj2' [A.mp4]
[2020/02/11 20:09:55] metadata.c:1243: debug: Stream 0 of A.mp4 is h.264
[2020/02/11 20:09:55] scanner.c:731: info: Scanning /mnt/raid/Media/Series
[2020/02/11 20:09:55] metadata.c:816: debug: Container: 'matroska,webm' NS01E08.mkv]
[2020/02/11 20:09:55] metadata.c:816: debug: Container: 'matroska,webm' NS01E09.mkv]
[2020/02/11 20:09:55] metadata.c:806: debug: Unhandled audio codec [0x15028]
[2020/02/11 20:09:55] metadata.c:816: debug: Container: 'matroska,webm' ST01E01.mkv]
[2020/02/11 20:09:56] playlist.c:135: warn: Parsing playlists...
[2020/02/11 20:09:56] playlist.c:269: warn: Finished parsing playlists.
[2020/02/11 20:09:56] scanner.c:939: debug: Initial file scan completed
[2020/02/11 20:10:11] minidlna.c:479: debug: Using locale dir '/usr/share/locale' and locale langauge C/C.UTF-8
[2020/02/11 20:10:11] minidlna.c:1048: warn: Starting MiniDLNA version 1.2.1.
[2020/02/11 20:10:11] minidlna.c:133: error: bind(http): Address in use
[2020/02/11 20:10:11] minidlna.c:1088: fatal: Failed to open socket for HTTP. EXITING
```
The number of videos listed in the UI doesn't match the number of actual files.

neheb, 2020-02-12:
If you're running TurrisOS or using coreutils ps, use ps aux | grep minidlna

That log is not terribly useful. strace should be more useful.

craibuc, 2020-02-12:
```bash
# ps aux | grep minidlna
root     11605  0.0  0.2  26252  2324 ?        Sl   20:06   0:00 /usr/bin/minidlna -f /tmp/minidlna.conf
root     16973  0.0  0.0   1104   508 pts/0    S+   20:42   0:00 grep minidlna

# strace -p 11605
strace: Process 11605 attached
_newselect(11, [5 6 7], [], NULL, {tv_sec=249, tv_usec=486718}
```
Now what?

craibuc, 2020-02-12:
Never mind...

```bash
clock_gettime(CLOCK_REALTIME, {tv_sec=1581475692, tv_nsec=301995192}) = 0
recvmsg(6, {msg_name={sa_family=AF_INET, sin_port=htons(53304), sin_addr=inet_addr("192.168.1.112")}, msg_namelen=16, msg_iov=[{iov_base="M-SEARCH * HTTP/1.1\r\nHOST: 239.2"..., iov_len=1499}], msg_iovlen=1, msg_control=[{cmsg_len=24, cmsg_level=SOL_IP, cmsg_type=IP_PKTINFO, cmsg_data={ipi_ifindex=if_nametoindex("br-lan"), ipi_spec_dst=inet_addr("192.168.1.1"), ipi_addr=inet_addr("239.255.255.250")}}], msg_controllen=24, msg_flags=0}, 0) = 174
clock_gettime(CLOCK_REALTIME, {tv_sec=1581475692, tv_nsec=302462039}) = 0
munmap(0xb6ffa000, 23)                  = 0
open("/etc/TZ", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 10
fstat64(10, {st_mode=S_IFREG|0644, st_size=23, ...}) = 0
mmap2(NULL, 23, PROT_READ, MAP_SHARED, 10, 0) = 0xb6ffa000
close(10)                               = 0
writev(3, [{iov_base="[2020/02/11 20:48:12] minissdp.c"..., iov_len=157}, {iov_base=NULL, iov_len=0}], 2) = 157
clock_gettime(CLOCK_REALTIME, {tv_sec=1581475692, tv_nsec=303195139}) = 0
_newselect(11, [5 6 7], [], NULL, {tv_sec=201, tv_usec=705077}) = 1 (in [6], left {tv_sec=200, tv_usec=702765})
clock_gettime(CLOCK_REALTIME, {tv_sec=1581475693, tv_nsec=305878519}) = 0
recvmsg(6, {msg_name={sa_family=AF_INET, sin_port=htons(53304), sin_addr=inet_addr("192.168.1.112")}, msg_namelen=16, msg_iov=[{iov_base="M-SEARCH * HTTP/1.1\r\nHOST: 239.2"..., iov_len=1499}], msg_iovlen=1, msg_control=[{cmsg_len=24, cmsg_level=SOL_IP, cmsg_type=IP_PKTINFO, cmsg_data={ipi_ifindex=if_nametoindex("br-lan"), ipi_spec_dst=inet_addr("192.168.1.1"), ipi_addr=inet_addr("239.255.255.250")}}], msg_controllen=24, msg_flags=0}, 0) = 174
clock_gettime(CLOCK_REALTIME, {tv_sec=1581475693, tv_nsec=306309265}) = 0
munmap(0xb6ffa000, 23)                  = 0
open("/etc/TZ", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 10
fstat64(10, {st_mode=S_IFREG|0644, st_size=23, ...}) = 0
mmap2(NULL, 23, PROT_READ, MAP_SHARED, 10, 0) = 0xb6ffa000
close(10)                               = 0
writev(3, [{iov_base="[2020/02/11 20:48:13] minissdp.c"..., iov_len=157}, {iov_base=NULL, iov_len=0}], 2) = 157
clock_gettime(CLOCK_REALTIME, {tv_sec=1581475693, tv_nsec=307038879}) = 0
_newselect(11, [5 6 7], [], NULL, {tv_sec=200, tv_usec=701234}) = 1 (in [6], left {tv_sec=199, tv_usec=705278})
clock_gettime(CLOCK_REALTIME, {tv_sec=1581475694, tv_nsec=303312601}) = 0
recvmsg(6, {msg_name={sa_family=AF_INET, sin_port=htons(53304), sin_addr=inet_addr("192.168.1.112")}, msg_namelen=16, msg_iov=[{iov_base="M-SEARCH * HTTP/1.1\r\nHOST: 239.2"..., iov_len=1499}], msg_iovlen=1, msg_control=[{cmsg_len=24, cmsg_level=SOL_IP, cmsg_type=IP_PKTINFO, cmsg_data={ipi_ifindex=if_nametoindex("br-lan"), ipi_spec_dst=inet_addr("192.168.1.1"), ipi_addr=inet_addr("239.255.255.250")}}], msg_controllen=24, msg_flags=0}, 0) = 174
clock_gettime(CLOCK_REALTIME, {tv_sec=1581475694, tv_nsec=303723074}) = 0
munmap(0xb6ffa000, 23)                  = 0
open("/etc/TZ", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 10
fstat64(10, {st_mode=S_IFREG|0644, st_size=23, ...}) = 0
mmap2(NULL, 23, PROT_READ, MAP_SHARED, 10, 0) = 0xb6ffa000
close(10)                               = 0
writev(3, [{iov_base="[2020/02/11 20:48:14] minissdp.c"..., iov_len=157}, {iov_base=NULL, iov_len=0}], 2) = 157
clock_gettime(CLOCK_REALTIME, {tv_sec=1581475694, tv_nsec=304438333}) = 0
_newselect(11, [5 6 7], [], NULL, {tv_sec=199, tv_usec=703834}) = 1 (in [6], left {tv_sec=198, tv_usec=701599})
clock_gettime(CLOCK_REALTIME, {tv_sec=1581475695, tv_nsec=307119627}) = 0
recvmsg(6, {msg_name={sa_family=AF_INET, sin_port=htons(53304), sin_addr=inet_addr("192.168.1.112")}, msg_namelen=16, msg_iov=[{iov_base="M-SEARCH * HTTP/1.1\r\nHOST: 239.2"..., iov_len=1499}], msg_iovlen=1, msg_control=[{cmsg_len=24, cmsg_level=SOL_IP, cmsg_type=IP_PKTINFO, cmsg_data={ipi_ifindex=if_nametoindex("br-lan"), ipi_spec_dst=inet_addr("192.168.1.1"), ipi_addr=inet_addr("239.255.255.250")}}], msg_controllen=24, msg_flags=0}, 0) = 174
clock_gettime(CLOCK_REALTIME, {tv_sec=1581475695, tv_nsec=307678001}) = 0
munmap(0xb6ffa000, 23)                  = 0
open("/etc/TZ", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 10
fstat64(10, {st_mode=S_IFREG|0644, st_size=23, ...}) = 0
mmap2(NULL, 23, PROT_READ, MAP_SHARED, 10, 0) = 0xb6ffa000
close(10)                               = 0
writev(3, [{iov_base="[2020/02/11 20:48:15] minissdp.c"..., iov_len=157}, {iov_base=NULL, iov_len=0}], 2) = 157
clock_gettime(CLOCK_REALTIME, {tv_sec=1581475695, tv_nsec=309140046}) = 0
_newselect(11, [5 6 7], [], NULL, {tv_sec=198, tv_usec=699132}) = 1 (in [6], left {tv_sec=194, tv_usec=935106})
clock_gettime(CLOCK_REALTIME, {tv_sec=1581475699, tv_nsec=73644173}) = 0
recvmsg(6, {msg_name={sa_family=AF_INET, sin_port=htons(58704), sin_addr=inet_addr("192.168.1.154")}, msg_namelen=16, msg_iov=[{iov_base="M-SEARCH * HTTP/1.1\r\nHOST: 239.2"..., iov_len=1499}], msg_iovlen=1, msg_control=[{cmsg_len=24, cmsg_level=SOL_IP, cmsg_type=IP_PKTINFO, cmsg_data={ipi_ifindex=if_nametoindex("br-lan"), ipi_spec_dst=inet_addr("192.168.1.1"), ipi_addr=inet_addr("239.255.255.250")}}], msg_controllen=24, msg_flags=0}, 0) = 174
clock_gettime(CLOCK_REALTIME, {tv_sec=1581475699, tv_nsec=74240505}) = 0
munmap(0xb6ffa000, 23)                  = 0
open("/etc/TZ", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 10
fstat64(10, {st_mode=S_IFREG|0644, st_size=23, ...}) = 0
mmap2(NULL, 23, PROT_READ, MAP_SHARED, 10, 0) = 0xb6ffa000
[... cut ...]
munmap(0xb6ffa000, 23)                  = 0
open("/etc/TZ", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 10
fstat64(10, {st_mode=S_IFREG|0644, st_size=23, ...}) = 0
mmap2(NULL, 23, PROT_READ, MAP_SHARED, 10, 0) = 0xb6ffa000
close(10)                               = 0
writev(3, [{iov_base="[2020/02/11 20:48:22] minissdp.c"..., iov_len=157}, {iov_base=NULL, iov_len=0}], 2) = 157
clock_gettime(CLOCK_REALTIME, {tv_sec=1581475702, tv_nsec=77880314}) = 0
_newselect(11, [5 6 7], [], NULL, {tv_sec=191, tv_usec=930392}
```

neheb, 2020-02-12:
That log tells me nothing. At the end, it's polling, just as before. Can you capture the crash?

I just checked dd-wrt's package. Seems there are useful patches there too...

craibuc, 2020-02-12:
start from the bottom:

```bash
read(4, "\0\0\0{\0\0\0\25\0\0\0\0\0\0\0\0", 16) = 16
stat64("/tmp/run/minidlna/files.db-wal", 0xbeffd198) = -1 ENOENT (No such file or directory)
fstat64(4, {st_mode=S_IFREG|0644, st_size=86016, ...}) = 0
fcntl64(4, F_SETLK64, {l_type=F_UNLCK, l_whence=SEEK_SET, l_start=0, l_len=0}) = 0
fcntl64(4, F_SETLK64, {l_type=F_RDLCK, l_whence=SEEK_SET, l_start=1073741824, l_len=1}) = 0
fcntl64(4, F_SETLK64, {l_type=F_RDLCK, l_whence=SEEK_SET, l_start=1073741826, l_len=510}) = 0
fcntl64(4, F_SETLK64, {l_type=F_UNLCK, l_whence=SEEK_SET, l_start=1073741824, l_len=1}) = 0
stat64("/tmp/run/minidlna/files.db-journal", 0xbeffd198) = -1 ENOENT (No such file or directory)
_llseek(4, 24, [24], SEEK_SET)          = 0
read(4, "\0\0\0{\0\0\0\25\0\0\0\0\0\0\0\0", 16) = 16
stat64("/tmp/run/minidlna/files.db-wal", 0xbeffd198) = -1 ENOENT (No such file or directory)
fstat64(4, {st_mode=S_IFREG|0644, st_size=86016, ...}) = 0
fcntl64(4, F_SETLK64, {l_type=F_UNLCK, l_whence=SEEK_SET, l_start=0, l_len=0}) = 0
fcntl64(4, F_SETLK64, {l_type=F_RDLCK, l_whence=SEEK_SET, l_start=1073741824, l_len=1}) = 0
fcntl64(4, F_SETLK64, {l_type=F_RDLCK, l_whence=SEEK_SET, l_start=1073741826, l_len=510}) = 0
fcntl64(4, F_SETLK64, {l_type=F_UNLCK, l_whence=SEEK_SET, l_start=1073741824, l_len=1}) = 0
stat64("/tmp/run/minidlna/files.db-journal", 0xbeffd198) = -1 ENOENT (No such file or directory)
_llseek(4, 24, [24], SEEK_SET)          = 0
read(4, "\0\0\0{\0\0\0\25\0\0\0\0\0\0\0\0", 16) = 16
stat64("/tmp/run/minidlna/files.db-wal", 0xbeffd198) = -1 ENOENT (No such file or directory)
fstat64(4, {st_mode=S_IFREG|0644, st_size=86016, ...}) = 0
fcntl64(4, F_SETLK64, {l_type=F_UNLCK, l_whence=SEEK_SET, l_start=0, l_len=0}) = 0
clock_gettime(CLOCK_REALTIME, {tv_sec=1581477478, tv_nsec=713183074}) = 0
sendto(10, "HTTP/1.1 200 OK\r\nContent-Type: t"..., 858, 0, NULL, 0) = 858
close(10)                               = 0
clock_gettime(CLOCK_REALTIME, {tv_sec=1581477478, tv_nsec=713982695}) = 0
_newselect(11, [5 6 7], [], NULL, {tv_sec=215, tv_usec=374672}) = 1 (in [7], left {tv_sec=210, tv_usec=374972})
clock_gettime(CLOCK_REALTIME, {tv_sec=1581477483, tv_nsec=714391152}) = 0
accept(7, {sa_family=AF_INET, sin_port=htons(52488), sin_addr=inet_addr("127.0.0.1")}, [16]) = 10
clock_gettime(CLOCK_REALTIME, {tv_sec=1581477483, tv_nsec=714779440}) = 0
munmap(0xb6ffa000, 23)                  = 0
open("/etc/TZ", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 11
fstat64(11, {st_mode=S_IFREG|0644, st_size=23, ...}) = 0
mmap2(NULL, 23, PROT_READ, MAP_SHARED, 11, 0) = 0xb6ffa000
close(11)                               = 0
writev(3, [{iov_base="[2020/02/11 21:18:03] minidlna.c"..., iov_len=83}, {iov_base=NULL, iov_len=0}], 2) = 83
clock_gettime(CLOCK_REALTIME, {tv_sec=1581477483, tv_nsec=716341590}) = 0
_newselect(11, [5 6 7 10], [], NULL, {tv_sec=210, tv_usec=372313}) = 1 (in [10], left {tv_sec=210, tv_usec=372140})
clock_gettime(CLOCK_REALTIME, {tv_sec=1581477483, tv_nsec=716876141}) = 0
recvfrom(10, "GET / HTTP/1.1\r\nUser-Agent: Wget"..., 2048, 0, NULL, NULL) = 141
clock_gettime(CLOCK_REALTIME, {tv_sec=1581477483, tv_nsec=717462030}) = 0
fcntl64(4, F_SETLK64, {l_type=F_RDLCK, l_whence=SEEK_SET, l_start=1073741824, l_len=1}) = 0
fcntl64(4, F_SETLK64, {l_type=F_RDLCK, l_whence=SEEK_SET, l_start=1073741826, l_len=510}) = 0
fcntl64(4, F_SETLK64, {l_type=F_UNLCK, l_whence=SEEK_SET, l_start=1073741824, l_len=1}) = 0
stat64("/tmp/run/minidlna/files.db-journal", 0xbeffd198) = -1 ENOENT (No such file or directory)
_llseek(4, 24, [24], SEEK_SET)          = 0
read(4, "\0\0\0{\0\0\0\25\0\0\0\0\0\0\0\0", 16) = 16
stat64("/tmp/run/minidlna/files.db-wal", 0xbeffd198) = -1 ENOENT (No such file or directory)
fstat64(4, {st_mode=S_IFREG|0644, st_size=86016, ...}) = 0
fcntl64(4, F_SETLK64, {l_type=F_UNLCK, l_whence=SEEK_SET, l_start=0, l_len=0}) = 0
fcntl64(4, F_SETLK64, {l_type=F_RDLCK, l_whence=SEEK_SET, l_start=1073741824, l_len=1}) = 0
fcntl64(4, F_SETLK64, {l_type=F_RDLCK, l_whence=SEEK_SET, l_start=1073741826, l_len=510}) = 0
fcntl64(4, F_SETLK64, {l_type=F_UNLCK, l_whence=SEEK_SET, l_start=1073741824, l_len=1}) = 0
stat64("/tmp/run/minidlna/files.db-journal", 0xbeffd198) = -1 ENOENT (No such file or directory)
_llseek(4, 24, [24], SEEK_SET)          = 0
read(4, "\0\0\0{\0\0\0\25\0\0\0\0\0\0\0\0", 16) = 16
stat64("/tmp/run/minidlna/files.db-wal", 0xbeffd198) = -1 ENOENT (No such file or directory)
fstat64(4, {st_mode=S_IFREG|0644, st_size=86016, ...}) = 0
fcntl64(4, F_SETLK64, {l_type=F_UNLCK, l_whence=SEEK_SET, l_start=0, l_len=0}) = 0
fcntl64(4, F_SETLK64, {l_type=F_RDLCK, l_whence=SEEK_SET, l_start=1073741824, l_len=1}) = 0
fcntl64(4, F_SETLK64, {l_type=F_RDLCK, l_whence=SEEK_SET, l_start=1073741826, l_len=510}) = 0
fcntl64(4, F_SETLK64, {l_type=F_UNLCK, l_whence=SEEK_SET, l_start=1073741824, l_len=1}) = 0
stat64("/tmp/run/minidlna/files.db-journal", 0xbeffd198) = -1 ENOENT (No such file or directory)
_llseek(4, 24, [24], SEEK_SET)          = 0
read(4, "\0\0\0{\0\0\0\25\0\0\0\0\0\0\0\0", 16) = 16
stat64("/tmp/run/minidlna/files.db-wal", 0xbeffd198) = -1 ENOENT (No such file or directory)
fstat64(4, {st_mode=S_IFREG|0644, st_size=86016, ...}) = 0
fcntl64(4, F_SETLK64, {l_type=F_UNLCK, l_whence=SEEK_SET, l_start=0, l_len=0}) = 0
clock_gettime(CLOCK_REALTIME, {tv_sec=1581477483, tv_nsec=726613228}) = 0
sendto(10, "HTTP/1.1 200 OK\r\nContent-Type: t"..., 858, 0, NULL, 0) = 858
close(10)                               = 0
clock_gettime(CLOCK_REALTIME, {tv_sec=1581477483, tv_nsec=727438316}) = 0
_newselect(11, [5 6 7], [], NULL, {tv_sec=210, tv_usec=361216} <unfinished ...>) = ?
+++ killed by SIGSEGV +++
```

neheb, 2020-02-12:
That's just weird. SIGSEGV means it crashed. But it did so during poll... I would expect after poll for it to crash.

```
recvfrom(10, "GET / HTTP/1.1\r\nUser-Agent: Wget"..., 2048, 0, NULL, NULL) = 141
```

is interesting... Where's Wget coming from?

This might be an SQLite3 issue. I'll look into that as well.

neheb, 2020-02-25:
From looking around, this seems to be a parsing bug with specific files. It's hard to tell without gdb whether the problem is in minidlna or ffmpeg...

neheb, 2020-02-28:
I'm tempted to close this as this is most likely an upstream bug and not OpenWrt specific. Can someone test this on Ubuntu or whatever?

craibuc, 2020-02-29:
I have Ubuntu on a LXC container that hosts Pi-hole.  Would that work?  Do the following:

- `apt-get update && apt-get upgrade -y`
- `apt install minidlna`
- modify the settings in `/etc/config/minidlna`
- restart the service: `service minidlna restart`
- add content
- open TV see if content is shared
- add new content
- see if service fails

Correct?

neheb, 2020-02-29:
Yeah.

craibuc, 2020-02-29:
The same process that makes `mnidlna` on Turris OS 4 works fine on Ubuntu Eoan.  The service indexes the new content and doesn't crash.

neheb, 2020-02-29:
OK so actual issue here. Next step would be to get a gdb backtrace.

craibuc, 2020-02-29:
OK.  Let me know how to do that.

neheb, 2020-02-29:
First I'd need to know what platform you're running on so I can compile some non-stripped binaries. copy those to OpenWrt and then just run "gdb minidlnad ...". When it crashes, run bt.

craibuc, 2020-03-01:
TurrisOS 4.0.5, Turris Omnia;  It's a fork of OpenWrt.  I think it's v18.  This is the project: https://gitlab.labs.nic.cz/turris.

Packages:
- Turris OS: `1.2.1-3.41`
- Ubuntu: `1.2.1+dfsg-1build1`

> copy those to OpenWrt and then just run "gdb minidlnad ...". When it crashes, run bt.

Here?
```bash
# which minidlna
/usr/bin/minidlna
```
Restart the service (`service minidlna restart`)?

What is `bt`?  I'm not seeing it listed as a package:
```bash
~# opkg find *bt*
btrfs-progs - 4.15.1-1.37
ebtables - 2018-06-27-48cff25d-1.37
kmod-ebtables - 4.14.162-1-0a66bb0316b4402bf65555c64ceed313.16
kmod-fs-btrfs - 4.14.162-1-0a66bb0316b4402bf65555c64ceed313.0
libtirpc - 1.0.3-2.27
rmbt-client - 1.0-1.0
```

neheb, 2020-03-01:
bt is a command ran in gdb.

I'll compile some binaries when I can.

neheb, 2020-03-03:
https://mega.nz/#!jIAWjSQQ!LeVd9XHwKG2yjGMzIO0WePwYDiUZRkB8sqaH2jzrl5Y

replace it on your router

Run

```
gdb minidlnad
run -S -f /var/etc/minidlna.conf
(crash the program)
bt
```

More info: https://sourceware.org/gdb/onlinedocs/gdb/Backtrace.html

I have a feeling this is related to fortify-headers. I don't have those installed locally...

craibuc, 2020-03-04:
Actual process:

OS X
```bash
scp  minidlnad root@192.168.1.1:/usr/bin/minidlnad
```
Router
In Openwrt, `minidlna` is located in `/usr/bin` and the configuration is in `/etc/config/minidlna`.

```bash
cd /usr/bin

# save current binary
mv minidlna minidlna.bak

# rename new binary
mv minidlnad minidlna

# make executable
chmod +x minidlna

# start debugger
gdb minidlna

# run
(gdb) run -S -f /etc/config/minidlna
Starting program: /usr/bin/minidlna -S -f /etc/config/minidlna
warning: Unable to find dynamic linker breakpoint function.
GDB will be unable to debug shared library initializers
and track explicitly loaded dynamic code.
Error loading shared library libavformat.so.57: No such file or directory (needed by /usr/bin/minidlna)
Error loading shared library libavutil.so.55: No such file or directory (needed by /usr/bin/minidlna)
Error relocating /usr/bin/minidlna: avformat_close_input: symbol not found
Error relocating /usr/bin/minidlna: avformat_find_stream_info: symbol not found
Error relocating /usr/bin/minidlna: av_dict_get: symbol not found
Error relocating /usr/bin/minidlna: avformat_open_input: symbol not found
Error relocating /usr/bin/minidlna: av_strerror: symbol not found
Error relocating /usr/bin/minidlna: av_log_set_level: symbol not found
Error relocating /usr/bin/minidlna: av_register_all: symbol not found
Error relocating /usr/bin/minidlna: av_reduce: symbol not found
[Inferior 1 (process 18609) exited with code 0177]
```

neheb, 2020-03-04:
That's weird. Are you lacking ffmpeg on the router?

craibuc, 2020-03-04:
I see `libffmpeg` `v4.0.5-1.16`.

neheb, 2020-03-04:
Hrm. Those No such file or directory errors are strange then. Try to find those in the system.

Does minidlnad work without gdb?

craibuc, 2020-03-04:
That was the `opkg` package, by the way.

```bash
# which minidlna
/usr/bin/minidlna

# service minidlna start
Error loading shared library libavformat.so.57: No such file or directory (needed by /usr/bin/minidlna)
Error loading shared library libavutil.so.55: No such file or directory (needed by /usr/bin/minidlna)
Error relocating /usr/bin/minidlna: avformat_close_input: symbol not found
Error relocating /usr/bin/minidlna: avformat_find_stream_info: symbol not found
Error relocating /usr/bin/minidlna: av_dict_get: symbol not found
Error relocating /usr/bin/minidlna: avformat_open_input: symbol not found
Error relocating /usr/bin/minidlna: av_strerror: symbol not found
Error relocating /usr/bin/minidlna: av_log_set_level: symbol not found
Error relocating /usr/bin/minidlna: av_register_all: symbol not found
Error relocating /usr/bin/minidlna: av_reduce: symbol not found
```

```bash
/# find .  -name *ffmpeg*
./usr/lib/opkg/info/libffmpeg.files-sha256
./usr/lib/opkg/info/libffmpeg.prerm
./usr/lib/opkg/info/libffmpeg.list
./usr/lib/opkg/info/libffmpeg.postinst
./usr/lib/opkg/info/libffmpeg.control
```

neheb, 2020-03-04:
Maybe ffmpeg 4 changed the library names. Anyway, I will try to compile a static build.

neheb, 2024-07-04:
this has long been fixed.

closed 2024-07-04 by neheb