openwrt/packages #10360: qemu: qemu-system-x86_64: Spice and VNC currently disabled

alex9434, 2019-10-26:
Maintainer: @yousong @mcbridematt @plntyk @ja-pa @champtar 
Environment: OpenWRT 18.06.4 on x64

Description:
qemu-system-x86_64 is compiled without support for SPICE and VNC. This makes it unusable on a (headless) router, because there is no way to display the boot screen (e.g. from Windows)

`qemu-system-x86_64: -spice port=5900,password=mypass: spice support is disabled`
and
`qemu-system-x86_64: -vnc 0.0.0.0:1: VNC support is disabled`

How to enable support for VNC and/or SPICE?

yousong, 2019-10-26:
Makes sense.

I plan to make basic VNC ui support enabled by default.  SPICE and other VNC features, etc. can be accessible as kconfig options and available only for custom build.

alex9434, 2019-10-26:
Sounds great, thank you so much! Is this achieved by simply changing 
`--disable-vnc`
to
`--enable-vnc`
in the Makefile?

yousong, 2019-10-26:
> @yousong
> Sounds great, thank you so much!
> Is basic VNC ui support achieved by simply changing
> `--disable-vnc`
> to
> `--enable-vnc`
> in the Makefile?

For the time being, yes!

alex9434, 2019-10-26:
Hi @yousong,
I can confirm that the package successfully compiled with --enable-vnc. I have not enabled vnc-png and/or vnc-jpeg because they have additional dependencies.

However, when I start qemu-system-x86_64 I get the following error:
```
qemu-system-x86_64: warning: hub 0 with no nics
qemu-system-x86_64: -vnc 0.0.0.0:1: could not read keymap file: 'en-us'
```
The first warning is probably related to an incorrect network setup. However the keyboard error could not be solved by specifying specific layouts (-k en-us or de)

One more question: The package compiles the softmmu version. Is there any way to compile the TCG based linux-user emulation, that is faster?

alex9434, 2019-10-26:
Quick update: I manually copied the keyboard definitions from the debian package "qemu-system-data" to /usr/share/qemu/keymaps and I am now able to start the VNC connection (hurray)!

Now I need to solve the networking issue.

yousong, 2019-10-26:
Hi @alex9434 , I just opened a pull request for this.  You may want to give it a try.  Note that it's unlikely these will be backported to 18.06 release.

PR #10363 (2019-10-28): qemu: vnc and spice ui support
Maintainer: me
Compile tested: x86_64
Run tested: x86_64

Description:

- basic vnc enabled by default
- advanced vnc features and spice support are off by default

Resolves #10360
closed 2019-10-28 by yousong