openwrt/packages #10364: vim: vim-fuller package should install vimdiff symlink

eklitzke, 2019-10-26:
# Issue template (remove lines from top till here)

Maintainer: @neheb or @ratkaj 

Description:
Vim will automatically go into a diff mode when invoked as `vimdiff`. There are no build options that need to be changed, instead there just needs to be a symlink with source `/usr/bin/vim` dest `/usr/bin/vimdiff`. I verified this works with the `vim-fuller` package, I'm not sure exactly which other vim builds this functionality applies to (I think it might be disabled for the tiny builds?).

You can test this works by invoking a command like `vimdiff /etc/config/luci /etc/config/luci-okg` and checking that you get the diff mode of vim.

neheb, 2020-01-28:
https://github.com/openwrt/packages/commit/53a77352338cf08a89157a35a9475ee03724c48f

closed 2020-01-28 by neheb

Massimo-B, 2020-02-25:
It this already released? I installed vim-fuller, added a symlink vimdiff -> vim .
Calling
`vimdiff /etc/config/luci  /etc/config/luci-opkg`
fails with
`E97: Cannot create diffs`
then shows a split with the 2 buffers but no diffs highlighted.

duxco, 2021-02-15:
@Massimo-B You need to install `diffutils`.