openwrt/packages #10429: keepalived ->  track_process not available - is CONFIG_PROC_EVENTS enabled in kernel config?

camelcamro, 2019-11-02:
Hello,
current trunk not working with keepalived ?
trunk: LuCI Master (git-19.304.35036-4798b3f) / OpenWrt SNAPSHOT r11372-2d00cf7515 

was working fine before, and now tried to install newest trunk, and now again something changed or whats the matter ?

when i start, i see in logs:

`Sat Nov  2 03:38:10 2019 daemon.info Keepalived[18437]: Starting Keepalived v2.0.19 (10/19,2019)
Sat Nov  2 03:38:10 2019 daemon.info Keepalived[18437]: Running on Linux 4.14.151 #0 Fri Nov 1 20:19:40 2019 (built for Linux 4.14.151)
Sat Nov  2 03:38:10 2019 daemon.info Keepalived[18437]: Command line: '/usr/sbin/keepalived' '-n' '-f' '/tmp/keepalived.conf'
Sat Nov  2 03:38:10 2019 daemon.info Keepalived[18437]: Opening file '/tmp/keepalived.conf'.
Sat Nov  2 03:38:10 2019 daemon.info Keepalived[18437]: track_process not available - is CONFIG_PROC_EVENTS enabled in kernel config?
Sat Nov  2 03:38:10 2019 daemon.info Keepalived[18437]: pidfile_write : Cannot open /run/keepalived.pid pidfile
Sat Nov  2 03:38:10 2019 daemon.info Keepalived[18437]: Stopped Keepalived v2.0.19 (10/19,2019)
Sat Nov  2 03:38:15 2019 daemon.info Keepalived[18497]: Starting Keepalived v2.0.19 (10/19,2019)
`

config file:
/etc/config/keepalived
`config global_defs
	option alt_config_file		"/etc/keepalived/keepalived.conf"`
and file:
/etc/keepalived/keepalived.conf
`vrrp_instance VI_1 {
#    state MASTER
    interface br-lan
    virtual_router_id 10
    priority 220
    advert_int 1
    authentication {
        auth_type PASS
        auth_pass 1111
    }
    virtual_ipaddress {
        192.168.1.10
    }
}
`

dagmoller, 2019-12-08:
Same problem here...

camelcamro, 2019-12-08:
see: https://github.com/openwrt/packages/pull/10317
the workaround was for me to make a link to the var/run ...

pls test it again, and before ... create a folder in root:
mkdir /run or do:
ln -s /var/run /run

then test again, i guess as i have seen that "CONFIG_PROC_EVENTS" stuff seems to be only a cosmetic message in logs ....
the real problem was to create the PID files ....
seems to be working for me,

pls test

this is only a workaround, why the PID file refers to /run/.... and not anymore to /var/run ... don't know

dagmoller, 2019-12-08:
> see: #10317
> the workaround was for me to make a link to the var/run ...
>
> pls test it again, and before ... create a folder in root:
> mkdir /run or do:
> ln -s /var/run /run
>
> then test again, i guess as [...]

Working... Thanks...

feckert, 2019-12-08:
This was already fixed in master branch https://github.com/openwrt/packages/commit/bc98aaa3f711f1cd6f7fa3a65da43411c2db98aa#diff-8ada53a0982d851e47876a53b4cdd033

closed 2019-12-08 by feckert

pgaufillet, 2021-02-03:
This bug is still present in 19.07.6. It would be nice to port it to branch 19.07.

feckert, 2021-02-03:
I do not have this problem for the openwrt-19.07 branch. I have now staged a branch for this problem in my [repository](https://github.com/TDT-AG/packages/tree/pr/20210203-19.07-keepalived). I have to say that I did not made a run test. Can you try this and see if it fixes your problem. If so, I have to try it again on my system to see if there is a regression on my setup

pgaufillet, 2021-02-03:
On my side I have rebuilt OpenWRT 19.07.6 from sources, and for sure the issue exists in this release. The fix is not visible in keepalived/Makefile file. If you diff this file between master and 19.07, you get:

```
@@ -86,9 +86,7 @@ endef
 
 CONFIGURE_ARGS+= \
        --with-init=SYSV \
-       --disable-nftables \
-       --disable-track-process \
-       --with-run-dir="/var/run"
+       --disable-nftables
```