openwrt/packages #10435: glib2: out-of-date due to transition to Meson

flyn-org, 2019-11-02:
Maintainer: @tripolar 
Environment: master

Description:
The OpenWrt glib2 package has lagged behind upstream due to the absence of Meson in our build system. The merge of https://github.com/openwrt/packages/pull/10043 added Meson to OpenWrt. Here are some sample OpenWrt packages that are moving to Meson:

  * mpd: https://github.com/openwrt/packages/pull/10052
  * grilo: https://github.com/openwrt/packages/pull/10433
  * libsoup: https://github.com/openwrt/packages/pull/10434

neheb, 2019-11-03:
My view is that we should hold off until it becomes necessary.

glib2 is huge. It's almost a megabyte on mips24kc and over that on other platforms. I imagine newer versions to be even bigger.

neheb, 2020-02-28:
Can you look at https://github.com/openwrt/packages/pull/11442 ?

neheb, 2020-07-28:
Updated