openwrt/packages #10438: adblock - dependency on logger

ratsputin, 2019-11-03:
Maintainer: @dibdot 
Environment: OpenWrt SNAPSHOT r11372-2d00cf7515 / LuCI Master git-19.305.42300-af5dc26

Description:

An (apparently) relatively recent update to `adblock.sh` leverages `logger` command in the `f_log()` routine.  Unfortunately, in deployments where a standard syslog has been implemented, `logger` is unavailable and adblock fails to start.

```
root@orion:/usr/var/log# adblock.sh
/usr/bin/adblock.sh: line 919: logger: not found
```

ratsputin, 2019-11-03:
It appears I've misinterpreted the nature of the problem.  Adblock does have a dependency on `logger`; however, the build system apparently doesn't reflect this dependency.  I am adding the `logger` package to my build to see if it resolves the issue.

ratsputin, 2019-11-03:
This is definitely the issue.  Please add a dependency on `logger` for `adblock` in the build system.

dibdot, 2019-11-03:
Will be fixed with #10446 (made logger optional)

closed 2019-11-03 by dibdot