openwrt/packages #10447: <lighttpd-mod-webdav> seg faults

nerdhardest, 2019-11-03:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version) 

GL-MT300A
MediaTek MT7620A ver:2 eco:6
OpenWrt SNAPSHOT r11254-54d9ac8ba8 / LuCI Master (git-19.304.35036-4798b3f)
Kernel Version
4.14.149

(But I don't think the architecture matters, since this bug has been reported & acknowledged as generic bug #2958 in all versions of mod_webdav (see below).)

Description:
https://redmine.lighttpd.net/issues/2958
lighttpd 1.4.54 segfaults on start in mod_webdav
2019-06-05

This bug has already been fixed; the fix just needs to migrate into OpenWrt.
```
Formating code blocks by wrapping them with pairs of ```
```

neheb, 2020-01-14:
ping @BKPepe your guys use this.

BKPepe, 2020-02-22:
Should be fixed in lighttpd 1.4.55, which was released 10 days ago. I will try to update it and if everything goes well, I will send pull request today or tomorrow.

PR #11409 (2020-02-23): lighttpd: update to version 1.4.55
Maintainer: @flyn-org 
Compile tested: Turris Omnia, mvebu (cortex-a9), OpenWrt master
Run tested: Turris Omnia, mvebu (cortex-a9), OpenWrt master

Description:
- Update to version [1.4.55](https://www.lighttpd.net/2020/1/31/1.4.55/)

Fixes: #10447
closed 2020-02-23 by neheb