openwrt/packages #10462: mariadb not compiling with uClibc++

neheb, 2019-11-04:
Maintainer: @micmac1 
Environment: master

Can you look into this? uClibc++ seems to be a good idea as libstdcpp sizes have gone crazy with recent GCC versions.

micmac1, 2019-11-04:
Nope. Not interested at all personally. mariadb-server is so big it doesn't matter one bit whether you can save a few bytes with uclibc++. If somebody else wants to assign this issue and gets a patch upstream go for it. Ping me to get the depends sorted for the package Makefile.

micmac1, 2019-11-04:
lol, crazy package size of libstdcpp?

libstdcpp6_8.3.0-2_mips_24kc.ipk | 375.4 KB

That's nothing at all compared to mariadb-server. I think you won't find anybody who'll self-assign this issue, seriously. It'll just stay here forever. ;-)

Regards,
Seb

neheb, 2019-11-04:
Point taken.

It seems uClibc++ first needs C++11 math to compile mariadb.

closed 2019-11-07 by neheb