openwrt/packages #10490: [1907]nfs-kernel-server: recursive dependency detected!

manjuprajna, 2019-11-07:
Maintainer: @tripolar 
Environment: OpenWrt 1907 rc1, build on ubuntu 1804

Description:
tmp/.config-package.in:104620:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:104620:  symbol PACKAGE_nfs-kernel-server depends on NFS_KERNEL_SERVER_V4
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
feeds/packages/net/nfs-kernel-server/Config.in:4:       symbol NFS_KERNEL_SERVER_V4 depends on PACKAGE_nfs-kernel-server
```
I already posted on July when 1907 came out, but several monthes past, still same issue, seems no one care about it. now 1907 already in release candidate, I don't want every time when I run make menuconfig, still same annoying error message pops up!
```

mans0n, 2019-12-06:
The dependency graph is `nfs-kernel-server -> libdevmapper -> kmod-dm -> kmod-dax -> LINUX_4_9`
The error message was first introduced by openwrt/openwrt@000d400baa. It seems removing all 4.9 targets causes this, but I don't know why.
Removing `!LINUX_4_9` dependencies on kmod-dm and kmod-dax will also fix it, but I'm not sure this is a fundamental solution.

mans0n, 2019-12-23:
The fix patch is merged so this can be closed now.

closed 2019-12-23 by neheb