openwrt/packages #10492: luci-app-sqm: Failed to execute cbi dispatcher target for entry

mikk150, 2019-11-07:
Maintainer: @tohojo
Environment: OpenWrt 19.07.0-rc1 Asus RT-N14U

Description:

When installing `luci-app-sqm` to clean router, I get this error when going to `Network > SQM QoS`

```
Failed to execute cbi dispatcher target for entry '/admin/network/sqm'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:938: module 'luci.cbi' not found:
	no field package.preload['luci.cbi']
	no file './luci/cbi.lua'
	no file '/usr/share/lua/luci/cbi.lua'
	no file '/usr/share/lua/luci/cbi/init.lua'
	no file '/usr/lib/lua/luci/cbi.lua'
	no file '/usr/lib/lua/luci/cbi/init.lua'
	no file './luci/cbi.so'
	no file '/usr/lib/lua/luci/cbi.so'
	no file '/usr/lib/lua/loadall.so'
	no file './luci.so'
	no file '/usr/lib/lua/luci.so'
	no file '/usr/lib/lua/loadall.so'
stack traceback:
	[C]: in function 'require'
	/usr/lib/lua/luci/dispatcher.lua:938: in function </usr/lib/lua/luci/dispatcher.lua:937>
```

jow-, 2019-11-07:
You can solve the issue in the meanwhile by installing `luci-compat`.

BKPepe, 2019-11-09:
Fixed by this commit: https://github.com/openwrt/packages/commit/349d73b2c39df1dcf00d5561b164556b89f38c31

closed 2019-11-09 by BKPepe