openwrt/packages #10493: MWAN3 Uptime not readabe in 19.07 RC1

Snotmann, 2019-11-07:
Hey guys,

with the new RC1 from 19.07 I installed mwan3 on my two WRT1043N V2 and V5 and see this in the overview of the multiwan tool:

MWAN Interfaces
Interface: wan
Status: Online
Uptime: NaNh:NaNm:NaNs
Interface: wanbackup
Status: Online
Uptime: NaNh:NaNm:NaNs

feckert, 2019-11-08:
What is the output of ```ubus call mwan3 status```?

Snotmann, 2019-11-08:
Here it is ... looks good so far, but not in luci

{
        "interfaces": {
                "wan": {
                        "age": 1,
                        "uptime": 63275,
                        "downtime": 0,
                        "score": 6,
                        "lost": 0,
                        "turn": 12218,
                        "status": "online",
                        "running": true,
                        "track_ip": [
                                {
                                        "ip": "194.25.0.60",
                                        "status": "up",
                                        "latency": 0,
                                        "packetloss": 0
                                },
                                {
                                        "ip": "212.100.35.2",
                                        "status": "skipped",
                                        "latency": 0,
                                        "packetloss": 0
                                },
                                {
                                        "ip": "8.8.8.8",
                                        "status": "up",
                                        "latency": 0,
                                        "packetloss": 0
                                }
                        ]
                },
                "wanbackup": {
                        "age": 4,
                        "uptime": 63267,
                        "downtime": 0,
                        "score": 6,
                        "lost": 0,
                        "turn": 12328,
                        "status": "online",
                        "running": true,
                        "track_ip": [
                                {
                                        "ip": "194.25.0.60",
                                        "status": "up",
                                        "latency": 0,
                                        "packetloss": 0
                                },
                                {
                                        "ip": "212.100.35.2",
                                        "status": "skipped",
                                        "latency": 0,
                                        "packetloss": 0
                                },
                                {
                                        "ip": "8.8.8.8",
                                        "status": "up",
                                        "latency": 0,
                                        "packetloss": 0
                                }
                        ]
                }
        },
        "connected": {
                "ipv4": [
                        "192.168.2.1",
                        "172.16.2.0/24",
                        "192.168.2.0",
                        "172.16.2.255",
                        "10.40.1.255",
                        "172.16.1.0/24",
                        "172.16.2.0",
                        "192.168.0.255",
                        "192.168.0.1",
                        "127.0.0.1",
                        "192.168.0.0/24",
                        "10.40.1.0",
                        "192.168.2.255",
                        "127.0.0.0/8",
                        "172.16.1.0",
                        "10.40.1.1",
                        "127.255.255.255",
                        "224.0.0.0/3",
                        "172.16.1.254",
                        "172.16.1.255",
                        "192.168.0.0",
                        "192.168.2.0/24",
                        "172.16.2.2",
                        "10.40.1.0/24",
                        "127.0.0.0"
                ],
                "ipv6": [
                        "fe80::/64"
                ]
        },
        "policies": {
                "ipv4": {
                        "balanced": [
                                {
                                        "interface": "wan",
                                        "percent": 100
                                }
                        ],
                        "wan_only": [
                                {
                                        "interface": "wan",
                                        "percent": 100
                                }
                        ],
                        "wan_wanb": [
                                {
                                        "interface": "wan",
                                        "percent": 100
                                }
                        ],
                        "wanb_only": [

                        ],
                        "wanb_wan": [

                        ]
                },
                "ipv6": {
                        "balanced": [

                        ],
                        "wan_only": [

                        ],
                        "wan_wanb": [

                        ],
                        "wanb_only": [

                        ],
                        "wanb_wan": [

                        ]
                }
        }
}

feckert, 2019-11-08:
Ok so the failure could be in this https://github.com/openwrt/luci/blob/master/applications/luci-app-mwan3/luasrc/view/mwan/overview_status_interface.htm#L10 on this line.
Maybe we have to convert the string to a number. Because NaN means not ```not a number```
Could you check this is Javascript? I do not have this fault.

Snotmann, 2019-11-08:
... ok i understand ... but where i find it on the router ?

feckert, 2019-11-09:
This is under `/usr/lib/lua/luci/view/mwan3`

Snotmann, 2019-11-11:
Dont get it what i need to change to keep the uptime showing correctly -.-

feckert, 2019-11-11:
Please give me the value **time**
Add the line ``` console.log(time)``` to the source before ```parseInt(time, 10)```
You can see the value in firefox in the ```Web Console```

feckert, 2019-11-12:
I think this is a luci problem. So we should move this issue to the luci repo

feckert, 2019-11-15:
There was a pullrequest that addresses the same issue see comment.
https://github.com/openwrt/luci/pull/3305#issuecomment-554282764

I think this should get fixed in the upcoming openwrt-19.07 snapshot.
Please try this again.
If you think it does not fix your problem please feel free to reopen this issue again.

closed 2019-11-15 by feckert