openwrt/packages #10496: p910nd: bonjour printers show up on the printer list with the hostname instead of the printer name

jpc, 2019-11-08:
Maintainer: @pkerling (find it by checking history of the package Makefile)
Environment: OpenWRT master, Archer C5 v1 MIPS, macOS

Description:

p910nd advertises printers over mdns but the service instance name is hardcoded to be the hostname of the router instead of a more descriptive name. This instance name shows up when adding printers on macOS and everyone else puts the printer model there (or "model @ hostname" for printers shared via another computer).

[umdns has support for setting instance names since 2017](https://git.openwrt.org/?p=project/mdnsd.git;a=blobdiff;f=service.c;h=0a9e25dbf77d356b528587ac355a67a0fc98a2db;hp=67e8f40a53b8c50417ba419e042863a226845011;hb=0e8b948ff8dd1cb1763180c99b522390529c7841;hpb=49fdb9f7a5454f7e11e8bcd621bff17cbf9da5ca) but this support was is not exposed in the procd helpers. I managed to hack around this issue but unfortunately there appears to be no easy way to extend the procd mdns shell API properly.

Here are my hacks that are enough to make this work:

```
--- a/rom/etc/init.d/p910nd
+++ b/overlay/upper/etc/init.d/p910nd
@@ -52,7 +52,7 @@ start_p910nd() {
 		config_get mdns_mfg "$section" mdns_mfg
 		config_get mdns_mdl "$section" mdns_mdl
 		config_get mdns_cmd "$section" mdns_cmd
-		[ "$mdns" -gt 0 ] && procd_add_mdns "pdl-datastream" "tcp" "$((port+9100))" "note=$mdns_note" "ty=$mdns_ty" "product=$mdns_product" "usb_MFG=$mdns_mfg" "usb_MDL=$mdns_mdl" "usb_CMD=$mdns_cmd"
+		[ "$mdns" -gt 0 ] && MDNS_INSTANCE_NAME="$mdns_ty" procd_add_mdns "pdl-datastream" "tcp" "$((port+9100))" "note=$mdns_note" "ty=$mdns_ty" "product=$mdns_product" "usb_MFG=$mdns_mfg" "usb_MDL=$mdns_mdl" "usb_CMD=$mdns_cmd"

 		procd_close_instance
 	fi
```

```
diff --git a/rom/lib/functions/procd.sh b/overlay/upper/lib/functions/procd.sh
index b8f9a21..1f47ba9 100644
--- a/rom/lib/functions/procd.sh
+++ b/overlay/upper/lib/functions/procd.sh
@@ -499,6 +499,7 @@ procd_add_mdns_service() {
 	json_add_object "${service}_$port"
 	json_add_string "service" "_$service._$proto.local"
 	json_add_int port "$port"
+	[ -n "$MDNS_INSTANCE_NAME" ] && json_add_string "instance" "$MDNS_INSTANCE_NAME"
 	[ -n "$1" ] && {
 		json_add_array txt
 		for txt in "$@"; do json_add_string "" "$txt"; done
```

Btw. this is probably not the ideal place to discuss the procd API extension problem. Should I file another bug at bugs.openwrt.org against umdns or procd?

neheb, 2020-01-15:
Yes.

If this fixes an actual issue, consider making a pull request.

meteozond, 2022-09-14:
The problem is that provided workaround requires some non  common openwrt `procd.sh` patching. Passing data through environment variable instead of command line argument doesn't look convenient. Сhanging `procd_add_mdns_service` interface would be better but it makes this fix backwards incompatible.

But it definitely works
[image]

systemcrash, 2024-02-16:
See openwrt/openwrt#10715