openwrt/packages #10509: ntpd: compilation fails due to missing libevent_pthread dependency [19.07 branch]

Borromini, 2019-11-09:
This is the same issue as https://github.com/openwrt/packages/issues/10307, which was fixed by PR https://github.com/openwrt/packages/pull/10308 for the master branch, as indicated by karlp. 

I have applied #10308 to my local 19.07 branch and together with a {clean,compile} for ntpd and libevent2, things seem to be fixed just like in master, so please apply this fix to the 19.07 branch as well.

Maintainer: @tripolar 
Environment: 19.07 HEAD, x86/64, ath79/generic.

Description: the same issue as referenced in bug report #10307 .

feckert, 2019-11-12:
I have pushed the fix see https://github.com/openwrt/packages/commit/e91e3fd6ad879db8d5a35374bb1fe9db3672739d

closed 2019-11-12 by feckert

karlp, 2019-11-12:
well, it fixes the build at least, but it wasn't (and isn't) the right fix.  https://github.com/openwrt/packages/pull/10491 does it properly.