openwrt/packages #10510: [meta] Freeze for openwrt-19.07?

jefferyto, 2019-11-09:
19.07.0-rc1 was released today: https://lists.infradead.org/pipermail/openwrt-devel/2019-November/019957.html

Should we freeze the openwrt-19.07 branch now or have a schedule for when to do so?

I think it would be nice to inform package maintainers and give them a chance to update their packages in 19.07, but I have no idea how to contact _everyone_.

champtar, 2019-11-09:
Hi @jefferyto 
I'm personally not a fan of any hard freeze, but you can just spam everyone via email to get the bulk of the changes asap

jefferyto, 2019-11-09:
@champtar I've spammed too many people here already :joy:

I don't have a strong opinion on whether to freeze or not (I think in either case bug/security fixes should continue to be offered), but it may be easier to offer stable packages for users if we freeze and bugfix.

diizzyy, 2019-11-09:
It's so far behind master at this point that most wont bother testing and backporting so it's probably not much of an issue. I personally still think it makes much more sense to have quarterly releases of packages rather than syncing it with OpenWrt's longwinded release schedule.

jefferyto, 2019-11-24:
There doesn't appear to be interest in continuing to discuss this, closing.

closed 2019-11-24 by jefferyto