openwrt/packages #10536: <collectd>:RRA files stop updating after system power reset due to year set forward to year 2033 instead of backward 19xx

darrentinghc, 2019-11-13:
After install luci-statistics and collectd packages it work fine untill system power recycle.  

once rebooted, all RRA files update stop due to OpenWRT default system time on boot to year 2033 on Linksys WRT1900ACv2 and collectd start sequence is way before ntp.

When this happen, RRA updated with year 2033 records and stop updating after ntp set system time corrected [consider old records since 2019 way before 2033].

This issues always reproduceable by stop collectd, remove all RRA and start collectd.  update will work again untill next reboot

feckert, 2019-11-13:
There was recently a fix that should address this issue https://github.com/openwrt/packages/commit/b295c9cff27749558f96527fe91377509aba44ef#diff-b4951230485a220e5550871ca7a86001
have you tried this?

darrentinghc, 2019-11-13:
> There was recently a fix that should address this issue [b295c9c#diff-b4951230485a220e5550871ca7a86001](https://github.com/openwrt/packages/commit/b295c9cff27749558f96527fe91377509aba44ef#diff-b4951230485a220e5550871ca7a86001)
> have you tried this?

I had not tried this fix yet, but above issues I believe need more that collectd fixes alone.
My exact issues started from boot time set to year 2033 cause collectd to update RRA last record as 2033, once ntp set year back to 2019, all new record consider outdated comparing to year 2033.

Once way to fix this we can do following both:-
1. collectd will discards old records if time backward vs last RRA record. [not sure this is already in-place].
2. OpenWRT kernel need to preset date before 2019 to make sure date is always backward if ntp not sync yet. 

This way, collectd would not update RRA untill ntp sync done.

feckert, 2019-11-13:
>     2. OpenWRT kernel need to preset date before 2019 to make sure date is always backward if ntp not sync yet.

One of my system does not have a RTC as well I get the time by an ntp sync after reboot. So I have the same situation. I make my on builds and the time is the time of the last file in time that was touched by the system (as I can remember). So the question is why do you have the year 2033 during boot?

darrentinghc, 2019-11-13:
> One of my system does not have a RTC as well I get the time by an ntp sync after reboot. So I have the same situation. I make my on builds and the time is the time of the last file in time that was [...]

It seeam kernel default time to 2033 during new boot? after ntp synced time normal

My /etc/urandom.seed is having year 2033 timestamps upon reboot
-rw-------    1 root     root           512 Oct 17  2033 urandom.seed

feckert, 2019-11-13:
This code is responsible for the time during boot if no rtc is present https://github.com/openwrt/openwrt/blob/34e22653ac18b6ac7fd368ca47625f665808067f/package/base-files/files/etc/init.d/sysfixtime#L27

The question is now why is the time set to 2033. Does the  Linksys WRT1900ACv2 have a RTC yes or no? If yes maby the rtc is wrong configured? If you do have a fresh build then the first time if no rtc is present and the system boots the first time then the build time is used and thats is in the past.
Maybe the build server does have a wrong time?

darrentinghc, 2019-11-13:
It look like WRT1900ACv2 doesn't have rtc

root@LEDE:/dev# /sbin/hwclock 
hwclock: can't open '/dev/misc/rtc': No such file or directory
root@LEDE:/dev# find . -name "*rtc*"  [found nothing]

darrentinghc, 2019-11-13:
I found where the 2033 come from.
Since WRT1900 have not rtc, mearning sysfixtime will not exit 0
boot() {
	start && exit 0 ==>This exit 0 would not happen and continue following code!!

	local maxtime="$(maxtime)"
	local curtime="$(date +%s)"
	[ $curtime -lt $maxtime ] && date -s @$maxtime
}

And $(maxtime) is based on newest file timestamp in /etc which is 2033 in my case

karlp, 2019-11-13:
I'm not sure this should be closed yet.  Those files shouldn't have such a forward timestamp.

hnyman, 2023-01-14:
Closing this.
Invalid file time stamps in the system are not related to collectd RRD databases.

closed 2023-01-14 by hnyman