openwrt/packages #10541: MPD missing dependency

neheb, 2019-11-13:
Maintainer: @thess
Environment: master

https://downloads.openwrt.org/snapshots/faillogs/mipsel_mips32/packages/mpd/mini/compile.txt

ping @dhewg

dhewg, 2019-11-14:
Error is:
```
Package mpd-mini is missing dependencies for the following libraries:
libasound.so.2
```

mpd package has:
- `+AUDIO_SUPPORT:alsa-lib` in `DEPENDS`
- `-Dalsa=$(if CONFIG_AUDIO_SUPPORT,enabled,disabled)` as the meson configure option

So that should work I guess?
What I'm not seeing from the log is if `CONFIG_AUDIO_SUPPORT` is enabled (e.g. if mpd is buggy in that regard and detects/uses alsa even if passing `-Dalsa=disabled`)

dhewg, 2019-11-14:
Doesn't seem to be a bug in mpd's build system:
```
meson ~/src/mpd -Dalsa=disabled
...
Dependency alsa skipped: feature alsa disabled
...
```

dhewg, 2019-11-14:
Didn't even see the issue while quoting the bug...
Sorry, silly mistake

commit 556ba181 (2019-11-14): mpd: fix compilation without AUDIO_SUPPORT

Fixes #10541
closed 2019-11-14 by neheb