openwrt/packages #10542: afalg_engine not compiling

neheb, 2019-11-13:
Maintainer: @cotequeiroz 
Environment: master

https://downloads.openwrt.org/snapshots/faillogs/mipsel_mips32/packages/afalg_engine/compile.txt

No idea what's going on here. Maybe missing kernel?

cotequeiroz, 2019-11-13:
It may be that:
 - afalg_engine is being built before the kernel headers are installed--unlikely;
 - kernel headers were not installed--bu then other packages would be failing.
 - this particular header is not being installed--kernel config option?  I can't find one that would make sense.  Also, this is the first `<linux/>` header being included, so it may be the first of many.
 - toolchain can't find where they're installed--CFLAGS error is possible here--I've seen an empty include shadow a valid one: `-I -I/toolchain/include`.  The package is not touching CFLAGS though.

I see that my hack to find the engines dir is not working with the bots either.  While it should not be the cause here--file not found will not eval to anything--it does not look good.
I'll take it out for the next version.

I'm working on some speedups, and need to fix zero-copy compilation as well (missing `CONFIG_` inside an `$(if` statement will always evaluate to false).

neheb, 2019-11-15:
Is this relevant? https://www.phoronix.com/scan.php?page=news_item&px=Linux-5.5-Finishes-SKCIPHER

cotequeiroz, 2019-11-18:
This should not have touched any AF_ALG stuff.  AF_ALG is a "consumer" of the crypto API, which "have been converted long ago".  Besides, this was just merged into linux devcrypto master, but not yet into the main Linux tree.
This does impact the qce driver I'm investigating--in fact I had already [backported the mentioned changes](https://github.com/openwrt/openwrt/pull/2518/commits/940179ee0f63b36b9351eb29f23ba4632b937eb6), which fixed a missing piece of code (the skcipher-added walksize field, to be precise) to bring gcm(aes) closer to working.

cotequeiroz, 2020-01-17:
The build problems I mentioned are fixed by #11044.

I still don't know what's happening here, but AFAIK there are no mipsel_mips32 devices with hardware-crypto capabilities.  If I'm right, this should not be a big deal.

Perhaps @jow- can shine some light into it: what SDK can I use to try to figure out what is going on here?

jow-, 2020-01-17:
Try https://downloads.openwrt.org/snapshots/targets/ar7/generic/openwrt-sdk-ar7-generic_gcc-8.3.0_musl.Linux-x86_64.tar.xz

cotequeiroz, 2020-01-17:
Thanks @jow- 
It was indeed very helpful.  It appears that linux 4.9 does not export the `linux/cryptouser.h` header.  I will have `cmake` detect that and disable the driver info display accordingly.  I will make that feature optional, which will avoid a warning when querying the drivers in the 19.07 package, since 19.07 does not build the `crypto_user.ko` module.

neheb, 2020-01-19:
Fixed

closed 2020-01-19 by neheb