openwrt/packages #10552: Python 3 failing to compile on powerpc

neheb, 2019-11-15:
Maintainer: @commodo @jefferyto 
Environment: master

https://downloads.openwrt.org/snapshots/faillogs/powerpc_464fp/packages/python3/compile.txt

jefferyto, 2019-11-16:
I looked into this... During `configure`, it does a test compile using a list of known compiler symbols, and sets a "platform triplet" based on this test. For powerpc_464fp it chooses `powerpc-linux-gnu` (the `gnu` part is just hard-coded).

`configure` also calls `$CC --print-multiarch` to get the multiarch value. If it has both a platform triplet and the multiarch value, it compares the two, and if they don't match it throws the error listed in the build log (`internal configure error for the platform triplet`).

I've tried `$CC --print-multiarch` for armvirt-64, malta-be, x86-64, and mpc85xx-generic (which is powerpc_8540), and they all return a blank value. `configure` doesn't have a multiarch value, so it doesn't try to compare it to the platform triplet.

For apm821xx-nand (powerpc_464fp), it returns `powerpc-linux-musl` (which doesn't match `powerpc-linux-gnu`).

So I'm guessing something changed for gcc for powerpc_464fp, that it now returns a multiarch value? Not sure if it is intentional or not - I will keep looking.

commodo, 2019-11-18:
there may be a generic solution to fix this, by removing some patches I added to Python3 for the `platform triplet` ; not sure if `014-remove-platform-so-suffix.patch` or `016-adjust-config-paths.patch` or both

I don't recall all the details about the patches and why I added it; it's probable that it was something related to how library files were generated, and this seemed simpler at the time;

i've been wanting to remove these patches for a while now;
maybe it's a good idea now

jefferyto, 2019-11-18:
I don't think those patches are related to this issue... A generic solution would be to patch out the platform triplet == multiarch test in `configure`, but I think there may be a deeper issue here.

There is a difference in the toolchain gcc for apm821xx compared to other architectures. When I run `./aarch64-openwrt-linux-musl-gcc -print-search-dirs` (for armvirt-64) I get this output:

```
install: /media/jeff/Jekyll/Files/openwrt/testing/armvirt-64/staging_dir/toolchain-aarch64_generic_gcc-8.3.0_musl/bin/../lib/gcc/aarch64-openwrt-linux-musl/8.3.0/
programs: =/media/jeff/Jekyll/Files/openwrt/testing/armvirt-64/staging_dir/toolchain-aarch64_generic_gcc-8.3.0_musl/bin/../libexec/gcc/aarch64-openwrt-linux-musl/8.3.0/:/media/jeff/Jekyll/Files/openwrt/testing/armvirt-64/staging_dir/toolchain-aarch64_generic_gcc-8.3.0_musl/bin/../libexec/gcc/:/media/jeff/Jekyll/Files/openwrt/testing/armvirt-64/staging_dir/toolchain-aarch64_generic_gcc-8.3.0_musl/bin/../lib/gcc/aarch64-openwrt-linux-musl/8.3.0/../../../../aarch64-openwrt-linux-musl/bin/aarch64-openwrt-linux-musl/8.3.0/:/media/jeff/Jekyll/Files/openwrt/testing/armvirt-64/staging_dir/toolchain-aarch64_generic_gcc-8.3.0_musl/bin/../lib/gcc/aarch64-openwrt-linux-musl/8.3.0/../../../../aarch64-openwrt-linux-musl/bin/
libraries: =/media/jeff/Jekyll/Files/openwrt/testing/armvirt-64/staging_dir/toolchain-aarch64_generic_gcc-8.3.0_musl/bin/../lib/gcc/aarch64-openwrt-linux-musl/8.3.0/:/media/jeff/Jekyll/Files/openwrt/testing/armvirt-64/staging_dir/toolchain-aarch64_generic_gcc-8.3.0_musl/bin/../lib/gcc/:/media/jeff/Jekyll/Files/openwrt/testing/armvirt-64/staging_dir/toolchain-aarch64_generic_gcc-8.3.0_musl/bin/../lib/gcc/aarch64-openwrt-linux-musl/8.3.0/../../../../aarch64-openwrt-linux-musl/lib/aarch64-openwrt-linux-musl/8.3.0/:/media/jeff/Jekyll/Files/openwrt/testing/armvirt-64/staging_dir/toolchain-aarch64_generic_gcc-8.3.0_musl/bin/../lib/gcc/aarch64-openwrt-linux-musl/8.3.0/../../../../aarch64-openwrt-linux-musl/lib/
```

When I run `./powerpc-openwrt-linux-musl-gcc -print-search-dirs` (for apm821xx-nand) I get this output:

```
install: /media/jeff/Jekyll/Files/openwrt/testing/apm821xx-nand/staging_dir/toolchain-powerpc_464fp_gcc-8.3.0_musl/bin/../lib/gcc/powerpc-openwrt-linux-musl/8.3.0/
programs: =/media/jeff/Jekyll/Files/openwrt/testing/apm821xx-nand/staging_dir/toolchain-powerpc_464fp_gcc-8.3.0_musl/bin/../libexec/gcc/powerpc-openwrt-linux-musl/8.3.0/:/media/jeff/Jekyll/Files/openwrt/testing/apm821xx-nand/staging_dir/toolchain-powerpc_464fp_gcc-8.3.0_musl/bin/../libexec/gcc/powerpc-linux-musl/:/media/jeff/Jekyll/Files/openwrt/testing/apm821xx-nand/staging_dir/toolchain-powerpc_464fp_gcc-8.3.0_musl/bin/../libexec/gcc/:/media/jeff/Jekyll/Files/openwrt/testing/apm821xx-nand/staging_dir/toolchain-powerpc_464fp_gcc-8.3.0_musl/bin/../lib/gcc/powerpc-openwrt-linux-musl/8.3.0/../../../../powerpc-openwrt-linux-musl/bin/powerpc-openwrt-linux-musl/8.3.0/:/media/jeff/Jekyll/Files/openwrt/testing/apm821xx-nand/staging_dir/toolchain-powerpc_464fp_gcc-8.3.0_musl/bin/../lib/gcc/powerpc-openwrt-linux-musl/8.3.0/../../../../powerpc-openwrt-linux-musl/bin/powerpc-linux-musl/:/media/jeff/Jekyll/Files/openwrt/testing/apm821xx-nand/staging_dir/toolchain-powerpc_464fp_gcc-8.3.0_musl/bin/../lib/gcc/powerpc-openwrt-linux-musl/8.3.0/../../../../powerpc-openwrt-linux-musl/bin/
libraries: =/media/jeff/Jekyll/Files/openwrt/testing/apm821xx-nand/staging_dir/toolchain-powerpc_464fp_gcc-8.3.0_musl/bin/../lib/gcc/powerpc-openwrt-linux-musl/8.3.0/:/media/jeff/Jekyll/Files/openwrt/testing/apm821xx-nand/staging_dir/toolchain-powerpc_464fp_gcc-8.3.0_musl/bin/../lib/gcc/powerpc-linux-musl/:/media/jeff/Jekyll/Files/openwrt/testing/apm821xx-nand/staging_dir/toolchain-powerpc_464fp_gcc-8.3.0_musl/bin/../lib/gcc/:/media/jeff/Jekyll/Files/openwrt/testing/apm821xx-nand/staging_dir/toolchain-powerpc_464fp_gcc-8.3.0_musl/bin/../lib/gcc/powerpc-openwrt-linux-musl/8.3.0/../../../../powerpc-openwrt-linux-musl/lib/powerpc-openwrt-linux-musl/8.3.0/:/media/jeff/Jekyll/Files/openwrt/testing/apm821xx-nand/staging_dir/toolchain-powerpc_464fp_gcc-8.3.0_musl/bin/../lib/gcc/powerpc-openwrt-linux-musl/8.3.0/../../../../powerpc-openwrt-linux-musl/lib/powerpc-linux-musl/:/media/jeff/Jekyll/Files/openwrt/testing/apm821xx-nand/staging_dir/toolchain-powerpc_464fp_gcc-8.3.0_musl/bin/../lib/gcc/powerpc-openwrt-linux-musl/8.3.0/../../../../powerpc-openwrt-linux-musl/lib/
```

It's hard to see, but there are extra directories in this output:

```
/media/jeff/Jekyll/Files/openwrt/testing/apm821xx-nand/staging_dir/toolchain-powerpc_464fp_gcc-8.3.0_musl/bin/../libexec/gcc/powerpc-linux-musl/
/media/jeff/Jekyll/Files/openwrt/testing/apm821xx-nand/staging_dir/toolchain-powerpc_464fp_gcc-8.3.0_musl/bin/../lib/gcc/powerpc-openwrt-linux-musl/8.3.0/../../../../powerpc-openwrt-linux-musl/bin/powerpc-linux-musl/
/media/jeff/Jekyll/Files/openwrt/testing/apm821xx-nand/staging_dir/toolchain-powerpc_464fp_gcc-8.3.0_musl/bin/../lib/gcc/powerpc-linux-musl/
/media/jeff/Jekyll/Files/openwrt/testing/apm821xx-nand/staging_dir/toolchain-powerpc_464fp_gcc-8.3.0_musl/bin/../lib/gcc/powerpc-openwrt-linux-musl/8.3.0/../../../../powerpc-openwrt-linux-musl/lib/powerpc-linux-musl/
```

None of these are included in the SDK archive, but gcc still looks for them. This in way confirms why `$CC --print-multiarch` returns `powerpc-linux-musl`, it thinks `powerpc-linux-musl` is the path for its selected multiarch.

jefferyto, 2019-11-18:
I've submitted a patch for gcc that should fix this:
https://patchwork.ozlabs.org/patch/1197041/
https://lists.infradead.org/pipermail/openwrt-devel/2019-November/020177.html

Original issue:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=89587
https://gcc.gnu.org/ml/gcc-patches/2019-03/msg00188.html

jefferyto, 2019-12-25:
This should be fixed - the Python 3 packages appear to be building correctly now.

closed 2019-12-25 by neheb