openwrt/packages #10565: Modemmanager: EM7455 not coming out of LPM

wicadmin, 2019-11-17:
Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (x86_64, x86, OpenWrt version: snapshot)

Description:
Using a previously compiled (modemmanager 1.8.0-1) from a year ago with 18.06.1, works with the EM7455. Now, trying to build a snapshot with the included modemmanager 1.12.0-1 is keeping the EM744 in low-power mode. If I switch back to the previous (my 18.06.1 image), everything works so the modem appears to be setup ok.

I didn't see anything standing out in dmesg or at boot log.

wicadmin, 2019-11-17:
current fw version:      02.33.03.00
 current carrier name:    GENERIC
 current config name:     GENERIC_002.072_000
 current subpri index:    000'

wicadmin, 2019-11-17:
Further investigation appears that 'Mobile Data' does not show up as a Protocol for the interface.

wicadmin, 2019-11-17:
[image]
[image]

wicadmin, 2019-11-17:
This is how it looks in the old working image:
[image]

predators46, 2019-11-18:
`WARNING: Makefile 'package/feeds/luci/luci-app-cifsd/Makefile' has a dependency on 'cifsd-tools', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-proto-modemmanager/Makefile' has a dependency on 'modemmanager', which does not exist
tmp/.config-package.in:104620:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:104620:symbol PACKAGE_nfs-kernel-server depends on NFS_KERNEL_SERVER_V4
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
feeds/packages/net/nfs-kernel-server/Config.in:4:symbol NFS_KERNEL_SERVER_V4 depends on PACKAGE_nfs-kernel-server`

micmac1, 2019-11-18:
@nickberry17 @aleksander0m OP forgot to add maintainer ;)

aleksander0m, 2019-11-22:
@wicadmin what do you mean it's kept in low-power mode?

Could you run MM with --debug (i.e. edit /etc/init/modemmanager to add --debug) and get the system logs while reproducing the issue?

wicadmin, 2019-11-23:
```
 --------------------------------
  General  |            dbus path: /org/freedesktop/ModemManager1/Modem/0
           |            device id: <redact>
  --------------------------------
  Hardware |         manufacturer: Sierra Wireless, Incorporated
           |                model: EM7455
           |    firmware revision: SWI9X30C_02.33.03.00 r8209 CARMD-EV-FRMWR2 2019/08/28 20:59:30
           |       carrier config: default
           |         h/w revision: 1.0
           |            supported: gsm-umts, lte
           |              current: gsm-umts, lte
           |         equipment id: <redact>
  --------------------------------
  System   |               device: /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4
           |              drivers: qcserial, qmi_wwan
           |               plugin: Sierra
           |         primary port: cdc-wdm0
           |                ports: ttyUSB0 (qcdm), ttyUSB1 (gps), ttyUSB2 (at), 
           |                       cdc-wdm0 (qmi), wwan0 (net)
  --------------------------------
  Numbers  |                  own: <redact>
  --------------------------------
  Status   |                 lock: sim-pin2
           |       unlock retries: sim-pin (3), sim-pin2 (3), sim-puk (10), sim-puk2 (10)
           |                state: disabled
           |          power state: low
           |       signal quality: 0% (cached)
  --------------------------------
  Modes    |            supported: allowed: 3g; preferred: none
           |                       allowed: 4g; preferred: none
           |                       allowed: 3g, 4g; preferred: 3g
           |                       allowed: 3g, 4g; preferred: 4g
           |              current: allowed: 4g; preferred: none
  --------------------------------
  Bands    |            supported: utran-1, utran-3, utran-4, utran-5, utran-8, utran-2, 
           |                       eutran-1, eutran-2, eutran-3, eutran-4, eutran-5, eutran-7, eutran-8, 
           |                       eutran-12, eutran-13, eutran-20, eutran-25, eutran-26, eutran-41
           |              current: eutran-25
  --------------------------------
  IP       |            supported: ipv4, ipv6, ipv4v6
  --------------------------------
  3GPP     |                 imei: <redact>
  --------------------------------
  3GPP EPS | ue mode of operation: csps-2
  --------------------------------
  SIM      |            dbus path: /org/freedesktop/ModemManager1/SIM/0
```

wicadmin, 2019-11-23:
@aleksander0m  Let me know where you want me to upload the log (16 MB),

wicadmin, 2019-11-23:
end of debug log. Let me know where you want me to upload the full log:

```
<<<<<< RAW:
<<<<<<   length = 20
<<<<<<   data   = 01:13:00:80:02:02:02:0F:00:40:00:07:00:02:04:00:01:00:47:00
[/dev/cdc-wdm0] received generic response (translated)...
<<<<<< QMUX:
<<<<<<   length  = 19
<<<<<<   flags   = 0x80
<<<<<<   service = "dms"
<<<<<<   client  = 2
<<<<<< QMI:
<<<<<<   flags       = "response"
<<<<<<   transaction = 15
<<<<<<   tlv_length  = 7
<<<<<<   message     = "UIM Get CK Status" (0x0040)
<<<<<< TLV:
<<<<<<   type       = "Result" (0x02)
<<<<<<   length     = 4
<<<<<<   value      = 01:00:47:00
<<<<<<   translated = FAILURE: InvalidQmiCommand
<debug> [1574489734.692440] Couldn't query facility 'corp-pers' status, assuming disabled
<debug> [1574489734.692458] Retrieving PIN status to check for enabled PIN
[/dev/cdc-wdm0] sent message...
<<<<<< RAW:
<<<<<<   length = 13
<<<<<<   data   = 01:0C:00:00:02:02:00:10:00:2B:00:00:00
[/dev/cdc-wdm0] sent generic request (translated)...
<<<<<< QMUX:
<<<<<<   length  = 12
<<<<<<   flags   = 0x00
<<<<<<   service = "dms"
<<<<<<   client  = 2
<<<<<< QMI:
<<<<<<   flags       = "none"
<<<<<<   transaction = 16
<<<<<<   tlv_length  = 0
<<<<<<   message     = "UIM Get PIN Status" (0x002B)
[/dev/cdc-wdm0] received message...
<<<<<< RAW:
<<<<<<   length = 20
<<<<<<   data   = 01:13:00:80:02:02:02:10:00:2B:00:07:00:02:04:00:01:00:47:00
[/dev/cdc-wdm0] received generic response (translated)...
<<<<<< QMUX:
<<<<<<   length  = 19
<<<<<<   flags   = 0x80
<<<<<<   service = "dms"
<<<<<<   client  = 2
<<<<<< QMI:
<<<<<<   flags       = "response"
<<<<<<   transaction = 16
<<<<<<   tlv_length  = 7
<<<<<<   message     = "UIM Get PIN Status" (0x002B)
<<<<<< TLV:
<<<<<<   type       = "Result" (0x02)
<<<<<<   length     = 4
<<<<<<   value      = 01:00:47:00
<<<<<<   translated = FAILURE: InvalidQmiCommand
<debug> [1574489734.756443] Couldn't query PIN status, assuming SIM PIN is disabled
<debug> [1574489734.756495] loading UE mode of operation for EPS...
<debug> [1574489734.756521] (ttyUSB2) device open count is 2 (open)
<debug> [1574489734.756577] (ttyUSB2): --> 'AT+CEMODE?<CR>'
<debug> [1574489734.773502] (ttyUSB2): <-- '<CR><LF>+CEMODE: 2<CR><LF><CR><LF>OK<CR><LF>'
<debug> [1574489734.773992] (ttyUSB2) device open count is 1 (close)
<debug> [1574489734.774047] Couldn't initialize interface: 'USSD not supported'
[/dev/cdc-wdm0] sent message...
<<<<<< RAW:
<<<<<<   length = 24
<<<<<<   data   = 01:17:00:00:10:01:00:01:00:43:00:0B:00:10:01:00:05:01:04:00:03:00:00:00
[... 142 lines cut ...]
<debug> [1574489735.042689] (ttyUSB2): <-- '<CR><LF>OK<CR><LF>'
<debug> [1574489735.042823] (ttyUSB2): --> 'AT+CCLK?<CR>'
<debug> [1574489735.055294] (ttyUSB2): <-- '<CR><LF>+CCLK: "80/01/06,00:00:07-20"<CR><LF><CR><LF>OK<CR><LF>'
<debug> [1574489735.055631] Extended signal capabilities supported
<debug> [1574489735.055665] (ttyUSB2) device open count is 1 (close)
<debug> [1574489735.055966] OMA capabilities not supported
<debug> [1574489735.056011] Couldn't initialize interface: 'OMA not supported'
<debug> [1574489735.056310] (ttyUSB2) device open count is 2 (open)
<debug> [1574489735.056371] (ttyUSB2): --> 'ATH<CR>'
<debug> [1574489735.064454] (ttyUSB2): <-- '<CR><LF>OK<CR><LF>'
<debug> [1574489735.064559] (ttyUSB2) device open count is 3 (open)
<debug> [1574489735.064599] (ttyUSB2) device open count is 2 (close)
<debug> [1574489735.064645] (ttyUSB2): --> 'AT+CLCC=?<CR>'
<debug> [1574489735.077808] (ttyUSB2): <-- '<CR><LF>OK<CR><LF>'
<debug> [1574489735.078025] periodic call list polling will be used if supported
<debug> [1574489735.078251] (ttyUSB2) device open count is 1 (close)
<info>  [1574489735.078441] Modem: state changed (unknown -> disabled)
<debug> [1574489735.078543] modem properly initialized
<debug> [1574489735.079940] [device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4] exported modem at path '/org/freedesktop/ModemManager1/Modem/0'
<debug> [1574489735.079963] [device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4]    plugin:  Sierra
<debug> [1574489735.079978] [device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4]    vid:pid: 0x1199:0x9071
<debug> [1574489735.080011] (ttyUSB2) device open count is 0 (close)
<debug> [1574489735.080029] (ttyUSB2) closing serial port...
<debug> [1574489735.080454] (ttyUSB2) serial port closed
```

wicadmin, 2019-11-26:
Hi @aleksander0m

It seems in the current working one (details below 18.06.1), the simple connect is called whereas the new one nothing else happens:

```
<info>  [1574804604.614955] Modem: state changed (unknown -> disabled)
<debug> [1574804604.615046] modem properly initialized
<debug> [1574804604.616274] [device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4] exported modem at path '/org/freedesktop/ModemManager1/Modem/0'
<debug> [1574804604.616305] [device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4]    plugin:  Sierra
<debug> [1574804604.616328] [device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4]    vid:pid: 0x1199:0x9071
<debug> [1574804604.616366] (ttyUSB2) device open count is 0 (close)
<debug> [1574804604.616393] (ttyUSB2) closing serial port...
<debug> [1574804604.616779] (ttyUSB2) serial port closed
<info>  [1574804605.001727] Simple connect started...
<debug> [1574804605.001744]    PIN: unspecified
<debug> [1574804605.001752]    Operator ID: unspecified
<debug> [1574804605.001759]    Allowed roaming: yes
<debug> [1574804605.001766]    APN: r.ispsn
<debug> [1574804605.001774]    IP family: ipv4
<debug> [1574804605.001781]    Allowed authentication: unspecified
<debug> [1574804605.001788]    User: unspecified
<debug> [1574804605.001795]    Password: unspecified
<debug> [1574804605.001802]    Number: unspecified
<info>  [1574804605.001808] Simple connect state (3/8): Enable
<info>  [1574804605.001839] Modem /org/freedesktop/ModemManager1/Modem/0: state changed (disabled -> enabling)
<debug> [1574804605.002025] Skipping modem initialization: device hotplugged
<debug> [1574804605.002042] (ttyUSB2) opening serial port...
<debug> [1574804605.002306] (ttyUSB2): setting up baudrate: 57600
<debug> [1574804605.002328] (ttyUSB2): port attributes not fully set
<debug> [1574804605.002353] (ttyUSB2) device open count is 1 (open)
<debug> [1574804605.002363] (ttyUSB2): running init sequence...
<debug> [1574804605.002387] (ttyUSB0) opening serial port...
<debug> [1574804605.002467] (ttyUSB0) device open count is 1 (open)
<debug> [1574804605.002477] Flashing primary AT port before enabling...
<debug> [1574804605.002491] (ttyUSB2): port attributes not fully set
<debug> [1574804605.002520] (ttyUSB2): --> 'ATE0<CR>'
<debug> [1574804605.011386] (ttyUSB2): <-- '<CR><LF>OK<CR><LF>'
<debug> [1574804605.011450] (ttyUSB2): --> 'ATV1<CR>'
<debug> [1574804605.017443] (ttyUSB2): <-- '<CR><LF>OK<CR><LF>'
<debug> [1574804605.017537] (ttyUSB2): --> 'AT+CMEE=1<CR>'
<debug> [1574804605.029203] (ttyUSB2): <-- '<CR><LF>OK<CR><LF>'
<debug> [1574804605.029280] (ttyUSB2): --> 'ATX4<CR>'
<debug> [1574804605.036727] (ttyUSB2): <-- '<CR><LF>OK<CR><LF>'
<debug> [1574804605.036843] (ttyUSB2): --> 'AT&C1<CR>'
<debug> [1574804605.044801] (ttyUSB2): <-- '<CR><LF>OK<CR><LF>'
<debug> [1574804605.103165] Getting device operating mode...
```

aleksander0m, 2019-11-27:
@wicadmin could you test with the following branch? https://github.com/openwrt/packages/pull/10597

wicadmin, 2019-11-27:
@aleksander0m - I was waiting for that PR, actively looking at it up to yesterday....sure, I can try and make a build but not sure if I have to wait for imagebuilder to get updated?

wicadmin, 2019-11-27:
UPDATE: Yes, snapshot still on: modemmanager_1.12.0-3_x86_64.ipk. I believe the latest is 1.12.0-8.

wicadmin, 2019-11-27:
BTW - I thought 19.07 was still open for new packages? Doesn't see MM is in there: `/ releases / 19.07-SNAPSHOT / packages `

aleksander0m, 2019-11-27:
@wicadmin feel free to suggest it for 19.07 :D

wicadmin, 2019-11-27:
@aleksander0m - I believe since it is actually at rc1 it may not accept anything new which is a shame not to get this in. Perhaps @feckert may be able to help.

wicadmin, 2019-11-27:
BTW - is luci-proto-modemmanager broken or working with this?

wicadmin, 2019-11-28:
OK, it is now in Snapshot. I'll go ahead and make a build.

wicadmin, 2019-11-28:
Well, no dice. It did not work. Modem still in LPM:

```
mmcli -m 0 --command='!gstatus?'
response: '!GSTATUS: 
Current Time:  25		Temperature: 39
Reset Counter: 8		Mode:        LOW POWER MODE '
```

feckert, 2019-11-29:
> @aleksander0m - I believe since it is actually at rc1 it may not accept anything new which is a shame not to get this in. Perhaps @feckert may be able to help.

I think it is not only done with porting the modemmanger also the dependencies and the toolchain must be cherrypicked.
So I think as I know openwrt-19.07 is feature complete.
Only bugfixes should get backported.
Modemmager is a new package.

wicadmin, 2019-11-29:
These are the packages pertaining to this which are in snapshot which are not in 19.07.0-rc1:

- mbim-utils.
- qmi-utils.
- libmbim.
- libqmi.
- modemmanager.
- luci-proto-modemmanager.

I don't know if the snapshot ipks are standalone without any necessary dependencies.

wicadmin, 2019-12-01:
@alexsander0m - I built 18.06.5 and same issue.

nickberry17, 2019-12-02:
@wicadmin  Hey, Are you using the LuCI master branch?  https://github.com/openwrt/luci/blob/master/protocols/luci-proto-modemmanager/htdocs/luci-static/resources/protocol/modemmanager.js  luci-proto-modemmanager is only officially in master.

If you are on 18.xx you would be using the older LuCI branch.  luci-proto-modemmanager didn't make it into 18, but you can compile it and install it yourself using this: https://github.com/nickberry17/luci-proto-modemmanager

wicadmin, 2019-12-02:
Hi @nickberry17 - yes, I complied it from there to build a 18.05.6 image. But got the INVALID_PROTO unknown error.

nickberry17, 2019-12-02:
Can you list the contents of  `/usr/lib/lua/luci/model/network/`

wicadmin, 2019-12-02:
```
ls -l /mnt/openwrtimg/usr/lib/lua/luci/model/network
total 32K
-rw-r--r-- 1 root root 1.7K Nov 29 04:01 proto_4x6.lua
-rw-r--r-- 1 root root 1.1K Nov 29 04:01 proto_6x4.lua
-rw-r--r-- 1 root root  401 Nov 29 04:01 proto_dhcpv6.lua
-rw-r--r-- 1 root root  410 Nov 29 04:01 proto_hnet.lua
-rwxrwxr-x 1 root root 1.6K Nov  5 09:12 proto_modemmanager.lua
-rw-r--r-- 1 root root 2.2K Nov 29 04:01 proto_ppp.lua
-rw-r--r-- 1 root root 1.5K Nov 29 04:01 proto_qmi.lua
-rw-r--r-- 1 root root  870 Nov 29 04:01 proto_wireguard.lua
```

nickberry17, 2019-12-02:
Looks like it is installed but LuCI is not able to use it.  Can you please verify you are using LuCI 18, and not some other branch? Run `cat /usr/lib/lua/luci/version.lua`

wicadmin, 2019-12-02:
```
luciname    = "LuCI openwrt-18.06 branch"
luciversion = "git-19.332.23140-f01d93b"
```

feckert, 2019-12-02:
I think the problem is than in master we moved to client side javascripting and in openwrt-18.06 this part is missing. So you should update to openwrt-19.06 or master to have all needed dependencies.

aleksander0m, 2019-12-02:
@wicadmin the FCC auth step is not even being tried, because there is no request to put the modem in online mode, as far as I can tell. That request should come from the MM protocol handler in netifd, as part of the simple connect procedure, and as you saw, looks like that is not being done.

I'm wondering whether you could try to test the setup without luci-proto-modemmanager. I've been too lazy and I have never tried that myself... I should! Anyway, if you can configure the modem settings for netifd in /etc/config/network manually, using `/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4` as `device` option (see https://github.com/openwrt/packages/blob/master/net/modemmanager/README.md), we could rule out whether it's some issue in the luci package or in the real protocol handler. While doing that test, try to gather syslog to check the netifd logs.

wicadmin, 2019-12-02:
@feckert - openwrt-19.06? You mean 19.07?

What I did was to use 18.06 and compiled these from their source with 18.06 SDK and imagebuilder. I also used the previous lua luci-proto-modemmanager as indicated by @nickberry17 

-     mbim-utils
-     qmi-utils
-     libmbim
-     libqmi
-     modemmanager
-     luci-proto-modemmanager

Shouldn't the above work?

nickberry17, 2019-12-02:
@wicadmin It should, it's working for me; but maybe there is some difference we haven't spotted yet.  I'll do some more investigating today.

wicadmin, 2019-12-03:
What is the point to have luci-proto-modemmanager in 19.07, mark it as @BROKEN instead of adding modemmanager into 19.07?

jow-, 2019-12-03:
`luci-proto-modemmanager` slipped into 19.07 because LuCI got synced with master. It was subsequently marked broken because modemmanger is unavailable there. I can drop the LuCI package entirely if this is preferable.

nickberry17, 2019-12-03:
> `luci-proto-modemmanager` slipped into 19.07 because LuCI got synced with master. It was subsequently marked broken because modemmanger is unavailable there. I can drop the LuCI package entirely if this is preferable.

No reason to have it there IMO if MM is not there with it.

wicadmin, 2019-12-04:
Well, be nice and preferable to slip modemmanager into 19.07 as for all intents and purposes it is stable to be in there and badly needed.

mans0n, 2019-12-06:
Please just drop it. I wish there's no dependency warning on stable point release.

poranje, 2019-12-07:
> 
> @wicadmin feel free to suggest it for 19.07 :D

Indeed !
+1

wicadmin, 2019-12-08:
@mans0n - I have already listed the dependencies which all work with 19.07. There is not further dependencies and nothing to backport as they all compile on 19.07. I do not understand the incompetence here as more connectivity and iot uses are requiring LTE modem and OpenWrt is refusing to adopt a proven package. We have the fronend part already in there, what is wrong with adding the backend?

mans0n, 2019-12-08:
@wicadmin Sorry for the annoyance. I haven't read all the previous comments.
I don't really have objection to adding modemmanager to 19.07. But modemmanager needs libmbim and libqmi, and these packages also should be backported. I thought that removing just one luci package would be much easier, but if there's much demand on modemmanager, then it would be preferable to backport it. All I want is just to clear the dependency warnings before the 19.07.0 release.

wicadmin, 2019-12-08:
@mans0n, first, apologies for my time earlier. It has just been fustrating having this capability excluded again. Lots of efforts were done by the authors and maintainers to have this excluded at this point. I hope someone can add it in as all the dependencies should have a clean build.

Thank you.

aleksander0m, 2019-12-10:
@wicadmin a good way forward to keep on discussing on the backport of MM to 19.07 would be to try to do the port yourself and suggest it as a PR to the 19.07 branch, and keep on discussing that in the PR. I have no idea what the process is to backport packages when the list has already been closed, but that would be a good and positive start. Complaining about this in an unrelated bug report is definitely not useful. I'm not an openwrt maintainer, but I'd suggest to keep a good and positive tone in all these discussions; using strong words and a bad tone doesn't help your cause ;)

aleksander0m, 2019-12-10:
> I'm wondering whether you could try to test the setup without luci-proto-modemmanager. I've been too lazy and I have never tried that myself... I should! Anyway, if you can configure the modem [...]

@wicadmin  were you able to try the setup without using the luci package, i.e. manually configuring /etc/config/network yourself?

nickberry17, 2019-12-12:
@wicadmin are you by chance selecting the kernel modules for your modem?  In the newer package I don't believe we are selecting all kernel modules by default.  You can reference the help text in menuconfig for more information, but the gist is that you may need to go to Kernel Modules > USB > and select the appropriate kmods for your modem, such as kmod-usb-net-*

wicadmin, 2019-12-15:
@aleksander0m - I do have the settings already in ` /etc/config/network`

@nickberry17  - I do have those modules too

wicadmin, 2019-12-15:
```
<<<<<< QMUX:
<<<<<<   length  = 19
<<<<<<   flags   = 0x80
<<<<<<   service = "loc"
<<<<<<   client  = 1
<<<<<< QMI:
<<<<<<   flags       = "indication"
<<<<<<   transaction = 2
<<<<<<   tlv_length  = 7
<<<<<<   message     = "Get Predicted Orbits Data Source" (0x0036)
<<<<<< TLV:
<<<<<<   type       = "Indication Status" (0x01)
<<<<<<   length     = 4
<<<<<<   value      = 01:00:00:00
<<<<<<   translated = general-failure
<warn>  [1576391153.059066] couldn't load supported assistance data types: 'LOC service: general failure'
<debug> [1576391153.059387] Messaging capabilities supported
<debug> [1576391153.059518] Supported storages loaded:
<debug> [1576391153.059532]   mem1 (list/read/delete) storages: 'sm, me'
<debug> [1576391153.059545]   mem2 (write/send) storages:       'sm, me'
<debug> [1576391153.059558]   mem3 (reception) storages:        'sm, me'
<debug> [1576391153.059608] (ttyUSB2) device open count is 2 (open)
<debug> [1576391153.059654] (ttyUSB2): --> 'AT+CPMS?<CR>'
<debug> [1576391153.073183] (ttyUSB2): <-- '<CR><LF>+CPMS: "SM",0,0,"SM",0,0,"SM",0,0<CR><LF><CR><LF>OK<CR><LF>'
<debug> [1576391153.073343] Current storages initialized:
<debug> [1576391153.073364]   mem1 (list/read/delete) storages: 'sm'
<debug> [1576391153.073381]   mem2 (write/send) storages:       'sm'
<debug> [1576391153.073398] Current storages initialized
<debug> [1576391153.073586] (ttyUSB2) device open count is 3 (open)
<debug> [1576391153.073632] (ttyUSB2) device open count is 2 (close)
<debug> [1576391153.073679] (ttyUSB2): --> 'AT+CTZU=1<CR>'
<debug> [1576391153.088485] (ttyUSB2): <-- '<CR><LF>OK<CR><LF>'
<debug> [1576391153.088623] (ttyUSB2): --> 'AT+CCLK?<CR>'
<debug> [1576391153.101078] (ttyUSB2): <-- '<CR><LF>+CCLK: "80/01/06,00:00:07-20"<CR><LF><CR><LF>OK<CR><LF>'
<debug> [1576391153.101409] Extended signal capabilities supported
<debug> [1576391153.101443] (ttyUSB2) device open count is 1 (close)
<debug> [1576391153.101728] OMA capabilities not supported
<debug> [1576391153.101771] Couldn't initialize interface: 'OMA not supported'
<debug> [1576391153.102079] (ttyUSB2) device open count is 2 (open)
<debug> [1576391153.102139] (ttyUSB2): --> 'ATH<CR>'
<debug> [1576391153.110150] (ttyUSB2): <-- '<CR><LF>OK<CR><LF>'
<debug> [1576391153.110265] (ttyUSB2) device open count is 3 (open)
<debug> [1576391153.110304] (ttyUSB2) device open count is 2 (close)
<debug> [1576391153.110351] (ttyUSB2): --> 'AT+CLCC=?<CR>'
<debug> [1576391153.123343] (ttyUSB2): <-- '<CR><LF>OK<CR><LF>'
<debug> [1576391153.123544] periodic call list polling will be used if supported
<debug> [1576391153.123796] (ttyUSB2) device open count is 1 (close)
<info>  [1576391153.123973] Modem: state changed (unknown -> disabled)
<debug> [1576391153.124074] modem properly initialized
<debug> [1576391153.125460] [device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4] exported modem at path '/org/freedesktop/ModemManager1/Modem/0'
<debug> [1576391153.125482] [device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4]    plugin:  Sierra
<debug> [1576391153.125496] [device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4]    vid:pid: 0x1199:0x9071
<debug> [1576391153.125522] (ttyUSB2) device open count is 0 (close)
<debug> [1576391153.125540] (ttyUSB2) closing serial port...
<debug> [1576391153.125947] (ttyUSB2) serial port closed
```
Only other warning of note (but just could be unsupported feature:
```
<warn>  [1576391151.459450] couldn't load carrier config: 'Cannot send message: QMI service 'pdc' version '1.15' required, got version '1.0''
```

aleksander0m, 2019-12-16:
@wicadmin can you please share your /etc/config/network and the output of "mmcli -m 0"?

aleksander0m, 2019-12-16:
> Only other warning of note (but just could be unsupported feature:
> 
> ```
> <warn>  [1576391151.459450] couldn't load carrier config: 'Cannot send message: QMI service 'pdc' version '1.15' required, got version '1.0''
> ```

Yes, that is unrelated, and that error will actually go away in libqmi 1.26: https://gitlab.freedesktop.org/mobile-broadband/libqmi/merge_requests/80

wicadmin, 2019-12-17:
```
  --------------------------------
  System   |               device: /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4
           |              drivers: qcserial, qmi_wwan
           |               plugin: Sierra
           |         primary port: cdc-wdm0
           |                ports: ttyUSB0 (qcdm), ttyUSB1 (gps), ttyUSB2 (at), 
           |                       cdc-wdm0 (qmi), wwan0 (net)
  --------------------------------

```

```
config interface 'wan1'
	option proto 'modemmanager'
	option device '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4'
	option apn 'r.ispsn'
	option auth 'none'
	option delegate '0'
	option iptype 'ipv4'
        option lowpower '1'
```

aleksander0m, 2019-12-17:
@wicadmin ok, this is good. And what happens if you manually run `ifup wan1`?

wicadmin, 2019-12-18:
Absolutely nothing seems to happen.

If I issue: `mmcli -m 0 --set-power-state-on` the modem moves out of 'low power' to 'on'

Then it seems to connect to the carrier network:
```
mmcli -m 0 --command='!gstatus?'
response: '!GSTATUS: 
Current Time:  356		Temperature: 40
Reset Counter: 56		Mode:        ONLINE         
System mode:   LTE        	PS state:    Attached     
LTE band:      B25    		LTE bw:      15 MHz  
LTE Rx chan:   8615		LTE Tx chan: 26615
LTE CA state:  NOT ASSIGNED
EMM state:     Registered     	Normal Service 
RRC state:     RRC Connected  
IMS reg state: No Srv  		

PCC RxM RSSI:  -48		RSRP (dBm):  -78
PCC RxD RSSI:  -64		RSRP (dBm):  -94
```
Here is all the modules installed:
```
adblock - 3.8.14-1
base-files - 204.1-r10775-db8345d8e4
bash - 5.0-3
bnx2-firmware - 20190416-1
busybox - 1.30.1-5
ca-bundle - 20190110-1
ca-certificates - 20190110-1
cgi-io - 14
curl - 7.66.0-1
dbus - 1.12.12-1
diffutils - 3.7-2
dnsmasq-full - 2.80-14
dropbear - 2019.78-2
e2fsprogs - 1.44.5-2
fdisk - 2.34-1
firewall - 2019-11-22-8174814a-1
fstools - 2019-08-28-6a61b9ac-1
fwtool - 2
getrandom - 2019-06-16-4df34a4d-3
glib2 - 2.58.3-3
ip-tiny - 5.0.0-2.1
ip6tables - 1.8.3-1
iperf3 - 3.7-1
ipset - 7.3-1
iptables - 1.8.3-1
jq - 1.6-1
jshn - 2019-06-16-ecf56174-1
jsonfilter - 2018-02-04-c7e938d6-1
kernel - 4.14.156-1-e1dd7676581672f6f0bdb1363506dee1
kmod-bnx2 - 4.14.156-1
kmod-button-hotplug - 4.14.156-3
kmod-hid - 4.14.156-1
kmod-hid-generic - 4.14.156-1
kmod-hwmon-core - 4.14.156-1
kmod-i2c-algo-bit - 4.14.156-1
kmod-i2c-core - 4.14.156-1
kmod-igb - 4.14.156-1
kmod-input-core - 4.14.156-1
kmod-input-evdev - 4.14.156-1
kmod-ip6tables - 4.14.156-1
kmod-ipt-conntrack - 4.14.156-1
kmod-ipt-core - 4.14.156-1
kmod-ipt-ipset - 4.14.156-1
kmod-ipt-nat - 4.14.156-1
kmod-ipt-offload - 4.14.156-1
kmod-lib-crc-itu-t - 4.14.156-1
kmod-libphy - 4.14.156-1
[... 269 lines cut ...]
[    7.213794] usbserial: USB Serial support registered for Qualcomm USB modem
[    7.222058] kmodloader: done loading kernel modules from /etc/modules.d/*
[    8.772401] r8169 0000:01:00.0 eth0: link down
[    8.772404] r8169 0000:01:00.0 eth0: link down
[    8.772552] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    8.774648] IPv6: ADDRCONF(NETDEV_UP): eth0.300: link is not ready
[    8.776348] IPv6: ADDRCONF(NETDEV_UP): eth0.400: link is not ready
[    8.778035] IPv6: ADDRCONF(NETDEV_UP): eth0.500: link is not ready
[    8.779655] IPv6: ADDRCONF(NETDEV_UP): eth0.900: link is not ready
[    8.780783] IPv6: ADDRCONF(NETDEV_UP): eth0.10: link is not ready
[    8.781484] IPv6: ADDRCONF(NETDEV_UP): eth0.2: link is not ready
[    8.810142] usb 1-4: new high-speed USB device number 4 using xhci_hcd
[    8.990671] usb 1-4: config 1 has an invalid interface number: 8 but max is 3
[    8.990674] usb 1-4: config 1 has no interface number 1
[    8.992002] qcserial 1-4:1.0: Qualcomm USB modem converter detected
[    8.992077] usb 1-4: Qualcomm USB modem converter now attached to ttyUSB0
[    8.992492] qcserial 1-4:1.2: Qualcomm USB modem converter detected
[    8.992554] usb 1-4: Qualcomm USB modem converter now attached to ttyUSB1
[    8.992967] qcserial 1-4:1.3: Qualcomm USB modem converter detected
[    8.993032] usb 1-4: Qualcomm USB modem converter now attached to ttyUSB2
[    8.993466] qmi_wwan 1-4:1.8: cdc-wdm0: USB WDM device
[    8.993752] qmi_wwan 1-4:1.8 wwan0: register 'qmi_wwan' at usb-0000:00:14.0-4, WWAN/QMI device, 06:03:70:ec:f4:8d
[   11.653413] r8169 0000:01:00.0 eth0: link up

```

aleksander0m, 2019-12-18:
Absolutely no idea what may be happening I'm afraid. You shouldn't need the manual mmcli command to set it to power-on because the netfid protocol handler is the one responsible for "enabling" the modem, and the power-on state is part of the enabling phase.

Would it be possible to give me ssh access to that machine so that I can try to debug it further? Email me details if possible.

wicadmin, 2019-12-19:
yeah, this one is not easy to figure out.

Do you have one of these modems? If so, I can get you the image. You can run it in qemu or some other VM. e.g.:

```
qemu-system-x86_64 -name OpenWRT -cpu Nehalem -m 1024 -boot d -hda openwrt-19.07.0-rc2-x86-64-combined-ext4.img
```

micmac1, 2019-12-19:
Hi there,

I'd just give git-bisect a try. Because you know it worked in the past.

In the beginning you wrote that MM 1.8.0-1 works on OpenWrt 18.06.1 with
EM7455. So I'd install 18.06.1 on my router. Then I would fetch the
18.06.1 SDL and replace MM 1.8.0-1 in it with the current MM. Then I'd
compile MM, load it onto the router and give it a try.

If it works the issue shouldn't be with MM. If it doesn't use git-bisect
on MM to find the commit where things went south.

The issue could also be with the freedesktop libs. So the same procedure
could be applied to them.

If you can exclude the libs and MM then the problem is likely in the
kernel. You could git bisect between OpenWrt 18.06.1 and master, in this
case.

This may boil down to some work for you. But in the past whenever I used
git-bisect it got me where I needed to be. :)

Just my 2 cents.

Regards,
Seb

aleksander0m, 2019-12-19:
@micmac1 in this very specific case, though, the issue is neither in MM nor in the freedesktop libs. The modem is correctly detected by MM, but there is no "enable" request arriving to MM, so the modem is never enabled (set to power state on).

I still don't know exactly why this is happening, but I suspect more on the MM protocol handler than on any other thing. Speaking of which....

@wicadmin could you please install the **pppd** package as well? I don't see it in the list you provided, and it's a *requirement* of the protocol handler. Your issue could be as simple as that... We should make the MM package depend on pppd, but I didn't get a chance to do that myself (patches welcome). See this bug report for context: https://github.com/openwrt/packages/issues/10802

wicadmin, 2019-12-19:
@aleksander0m - it could be that simple. Not sure how it is there is 18.06.1 and now gone using the same build scripts. I will add and try that.

aleksander0m, 2019-12-19:
@wicadmin my original packaging rules, the ones you probably used in 18.06.1, had an explicit dependency on ppp: https://gitlab.freedesktop.org/mobile-broadband/mobile-broadband-openwrt/blob/master/modemmanager/Makefile

aleksander0m, 2019-12-19:
(also please reboot your setup once you have installed the ppp package)

wicadmin, 2019-12-19:
well, that was it. PPP was the missing link.

Now I can go try moving to the em12-g and the dw5821e /T77W968 should be arriving soon. Just need to know how to lock these to Sprint band 25. Any help on that before I close out this issue.

[6 more replies]

closed 2023-09-13 by feckert