openwrt/packages #10569: libperl.so: undefined reference to `__stack_chk_guard'

fuqiang03, 2019-11-17:
ubuntu 19.10
gcc 8.2 
glibc 2.27
[image]

neheb, 2020-01-14:
Is this still an issue?

fuqiang03, 2020-01-14:
@neheb 
Excuse me, do you have any idea

neheb, 2020-01-14:
That issue is caused by using LD instead of CC for linking. Maybe the issue is specific to glibc.

fuqiang03, 2020-01-14:
Other programs have made the same mistake, not just glibc

fuqiang03, 2020-01-14:
I guess it's the tool chain

mewsf, 2020-01-28:
I have the same issue when compiling openwrt x86 target with glibc. I edited compiler and linker flags to force perl not to compile with stack guard. There's also similar issue that says undefined xxx when compiling other packages, I don't remember the details because it was a few days ago. Using musl is ok so it has to be glibc which caused the problem.

fuqiang03, 2020-01-28:
@mewsf 
Can you share specific changes
I changed the glibc version. The problem is still the same
Question the tool chain

mewsf, 2020-01-29:
I edited `openwrt/feeds/packages/lang/perl/Makefile` and in the `define Build/Configure` section, I changed a line from

`	(cd $(PKG_BUILD_DIR) && ./Configure -S)`
to make it like this
`	(cd $(PKG_BUILD_DIR) && sed -i "s/-fstack-protector-strong/-fno-stack-protector/g" $(PKG_BUILD_DIR)/config.sh && sed -i "s/-fstack-protector/-fno-stack-protector/g" $(PKG_BUILD_DIR)/config.sh && ./Configure -S)`

It's a dirty way to change compiler flags. But I still hope if someone can really fix the problem with glibc. I'm not good at this.

neheb, 2020-01-29:
proper solution is to change the linker from ld to gcc.

mewsf, 2020-01-31:
> 
> 
> proper solution is to change the linker from ld to gcc.

That's much beyond my ability, I can only wait for someone who can submit relevant code for that.

mewsf, 2020-02-19:
> 
> 
> proper solution is to change the linker from ld to gcc.

I'm rebuilding openwrt now, and I checked config.sh of perl, there's a line says `ld='x86_64-openwrt-linux-gnu-gcc'`, I think it means that it has already used gcc as linker?

I added '-lssp' to LDFLAGS and it compiled successfully. It turns out that when using glibc to compile, some packages needs special linker flags like '-ldl' or '-lssp' to solve the library problem.

fuqiang03, 2020-02-21:
@mewsf 
Thanks for your help

fuqiang03, 2020-02-23:
@mewsf 
My build passes,They don't seem interested in glibc,
Because some programs cannot be built in glibc
For example, ksmbd tools

mewsf, 2020-02-23:
> 
> 
> @mewsf
> My build passes,They don't seem interested in glibc,
> Because some programs cannot be built in glibc
> For example, ksmbd tools

I have never built that package. but maybe I can help with that if you paste build log here.

fuqiang03, 2020-02-23:
It has been modified and passed
You can verify it for me
[image]

mewsf, 2020-02-23:
> 
> 
> It has been modified and passed
> You can verify it for me
> [image]

...原来是国人 你这是说编译通过了吗还是没有

fuqiang03, 2020-02-23:
编译通过了
这沟通省事多了

mewsf, 2020-02-23:
那我觉得可以关了这个issue了  openwrt本身主要还是侧重于嵌入式设备 所以官方也不太会有太多精力关注glibc的问题

fuqiang03, 2020-02-23:
哎 只不过glibc用习惯了.官方不关注自己修改很麻烦只能有时间自己改改.
这个issue暂时留着,做个记录,问题都解决了,我自然就关闭了.
[image]

mewsf, 2020-02-23:
搜了一下 这个估计是Makefile本身加了类似`-Werror`的编译flag导致的

fuqiang03, 2020-02-23:
看错误提示是这样的
makefile文件在编译的时候是生成的 源码里面没有这个文件
得要仔细的找找了
先一步一步的试试了

fuqiang03, 2020-02-24:
实在受不了,需要改动的程序太多了.

closed 2020-02-24 by fuqiang03

LitingLin, 2020-04-07:
in 'feeds/packages/lang/perl/files/libc.config' add ldflags variable in line 3 with '-lssp'