openwrt/packages #10579: igmpproxy: not setting firewall rules

finasfinas, 2019-11-17:
Maintainer: @\<github-user> (find it by checking history of the package Makefile)

Environment: ipq80xx, latest trunk

Description:
igmpproxy does not set the two firewall rules that allow multicast traffic to work.
actually, it does set the igmp one but it is probably not ordered correctly as I still need to add an equal rule manually to make it work. 
The forward rule ( 214.xxx.xxx.xxx ) is not inserted and has to be put manually.

`
root@gateway:~# iptables --list |grep igmp
ACCEPT     igmp --  anywhere             anywhere             /* !fw3: ubus:igmpproxy[instance1] rule 0 */
ACCEPT     igmp --  anywhere             anywhere             /* !fw3: Allow-IGMP */

`
the first rule is inserted by igmpproxy but it does not work. I have to insert the second one, so I assume it is because of it's order in the cain.
`
root@gateway:~# iptables --list |grep mcast
zone_iptv_dest_ACCEPT  udp  --  anywhere             base-address.mcast.net/4  /* !fw3: IPTV */
root@gateway:~# iptables --list |grep 224
root@gateway:~# 

`
the mcast rule is not inserted. I had to manually add it.

@dedeckeh @Mushoz @ldir-EDB0

diizzyy, 2019-11-20:
Wrong repo, it's in the OpenWrt repo which should be submitted here https://bugs.openwrt.org/

closed 2019-11-20 by diizzyy

finasfinas, 2019-11-20:
how come? 
Isn't igmpproxy a package?