openwrt/packages #10589: ddns-scripts gandi provider

ptsneves, 2019-11-19:
Maintainer: @chris5560 
Environment: Non arch specific, new provider

I would like to ask if it is possible to add support for gandi live dns service. I could try to do it but they do not use a user:password system but an API.

The query is pretty simple, and can be found [`here`](https://nextnet.top/content/using-gandi-livedns-dynamic-dns-server)
```
# retreive current external IP address from Akamai server
MY_IP=$(curl -s http://whatismyip.akamai.com/)  

# Gandi livedn API KEY
APIKEY="................."

# Static Domain hosted at Gandi
DOMAIN="nextnet.top"

# Dynamic Subdomain
SUBDOMAIN="pi3gre"

#Get the current Zone for the provided domain
CURRENT_ZONE_HREF=$(curl -s -H "X-Api-Key: $APIKEY" https://dns.api.gandi.net/api/v5/domains/$DOMAIN | jq -r '.zone_records_href')

# Update the A reccord of the Dynamic Subdomain by PUTing on the current zone
curl -D- -X PUT -H "Content-Type: application/json" \
        -H "X-Api-Key: $APIKEY" \
        -d "{\"rrset_name\": \"$SUBDOMAIN\",
             \"rrset_type\": \"A\",
             \"rrset_ttl\": 1200,
             \"rrset_values\": [\"$MY_IP\"]}" \
        $CURRENT_ZONE_HREF/$SUBDOMAIN/A
```

If nobody wants to do this, could somebody point me out on how to accomplish this? Also I am available to test it if someone requires.

Thanks a lot
Paulo Neves

diizzyy, 2019-11-20:
There's no maintainer for this package so someone(tm) needs to do a pull request. If you want to use Gandi's service prompty there are both python and shell scripts available on GitHub.

jefferyto, 2019-11-20:
The ddns-scripts package supports custom scripts placed in `/usr/lib/ddns` and several are available as separate packages (all prefixed with `ddns-scripts-`) for different services. You can take a look at those scripts (`update-*.sh` in the [files](https://github.com/openwrt/packages/tree/master/net/ddns-scripts/files) directory) to see how they are written, and there is also a [sample script](https://github.com/openwrt/packages/blob/master/net/ddns-scripts/samples/update_sample.sh) you can use as a starting point.

Once your script is ready, you can [add it as a service](https://github.com/openwrt/packages/blob/8022b0de7c88ac1581fd7c7ae5f9126a8f77f2f1/net/ddns-scripts/files/services#L20-L23) to the `services` file in `/etc/ddns`. You can also open a pull request to add your script and package it as an additional service.

ptsneves, 2019-11-21:
@jefferyto  I think i can do it then. I had looked at the package, but the scripts that seemed similar were very dense bash scripts. I had not seen the sample script, and now i did i am confident i can do that. Just one question: In the pull request, will you allow that I force curl for the transfer? That seems the easiest for me to deliver.

jefferyto, 2019-11-21:
I can't say whether forcing curl will be accepted, but based on #6150 I would say it's probably okay.

Personally I would try to make it compatible with both curl and wget since other people may prefer wget, but that's just my opinion.

ajgrf, 2020-11-09:
I wrote a script for just this purpose a while back. See [here](https://ajgrf.com/post/openwrt-ddns-gandi/). If you want to use it, I'm fine with licensing it as GPLv2.

PR #14038 (2020-12-07): ddns-scripts: add gandi.net provider
Maintainer: me / @Ansuel 
Compile tested: NO
Run tested: NO

Description:
Add ddns provider gandi.net to the ddns-scripts package.
Thanks goes to @ajgrf who provides the script.

@ptsneves @jefferyto  I cannot test this because I do not use this ddns provider.

This fixes #10589
closed 2020-12-07 by neheb