openwrt/packages #10590: Eror jool lede 17.01.7 #kernel 4.4.182

predators46, 2019-11-19:
`/home/ubuntu/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/jool-66a791c90751d7941b08c142babe1deec73d0996/mod/stateful/fragment_db.c:96:14: error: 'INETFRAGS_HASHSZ' undeclared (first use in this function) return c & (INETFRAGS_HASHSZ - 1); ^ /home/ubuntu/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/jool-66a791c90751d7941b08c142babe1deec73d0996/mod/stateful/fragment_db.c:96:14: note: each undeclared identifier is reported only once for each function it appears in /home/ubuntu/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/jool-66a791c90751d7941b08c142babe1deec73d0996/mod/stateful/fragment_db.c:97:1: warning: control reaches end of non-void function [-Wreturn-type] } ^ scripts/Makefile.build:277: recipe for target '/home/ubuntu/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/jool-66a791c90751d7941b08c142babe1deec73d0996/mod/stateful/fragment_db.o' failed make[5]: *** [/home/ubuntu/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/jool-66a791c90751d7941b08c142babe1deec73d0996/mod/stateful/fragment_db.o] Error 1 Makefile:1436: recipe for target '_module_/home/ubuntu/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/jool-66a791c90751d7941b08c142babe1deec73d0996/mod/stateful' failed make[4]: *** [_module_/home/ubuntu/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/jool-66a791c90751d7941b08c142babe1deec73d0996/mod/stateful] Error 2 make[4]: Leaving directory '/home/ubuntu/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/linux-4.4.182' Makefile:112: recipe for target '/home/ubuntu/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/jool-66a791c90751d7941b08c142babe1deec73d0996/.built' failedmake[3]: *** [/home/ubuntu/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/jool-66a791c90751d7941b08c142babe1deec73d0996/.built] Error 2 make[3]: Leaving directory '/home/ubuntu/openwrt/feeds/packages/net/jool' package/Makefile:105: recipe for target 'package/feeds/packages/jool/compile' failed make[2]: *** [package/feeds/packages/jool/compile] Error 2 make[2]: Leaving directory '/home/ubuntu/openwrt' package/Makefile:101: recipe for target '/home/ubuntu/openwrt/staging_dir/target-mips_24kc_musl-1.1.16/stamp/.package_compile' failed make[1]: *** [/home/ubuntu/openwrt/staging_dir/target-mips_24kc_musl-1.1.16/stamp/.package_compile] Error 2 make[1]: Leaving directory '/home/ubuntu/openwrt' /home/ubuntu/openwrt/include/toplevel.mk:205: recipe for target 'world' failed make: *** [world] Error 2`

neheb, 2019-11-19:
Is this really 17.01? This failure is a problem for newer kernels. It should not be an issue for 17.01. See https://github.com/openwrt/packages/issues/7095

predators46, 2019-11-19:
> Is this really 17.01? This failure is a problem for newer kernels. It should not be an issue for 17.01. See #7095

yes, lede 17.07

neheb, 2019-11-20:
I reproduced. That's bizare. Even the commit message mentions kernel 4.17. Maybe it has to do with wireless-backports or something...

Anyway, use this patch for now: https://raw.githubusercontent.com/openwrt/packages/4f28139a3cf897cdf7d8bb9a48fdd5c0f490b0e0/net/jool/patches/010-Add-support-for-kernel-4.17.patch

put it in net/jool/patches (create the directory).

neheb, 2019-11-20:
Never mind. I pushed the fix.

closed 2019-11-20 by neheb