openwrt/packages #10594: stubby error message logs

th0ma7, 2019-11-20:
# Issue template (remove lines from top till here)

Maintainer: @salim-b 
Environment: TP-Link WDR4300, OpenWRT 19.07-rc1
```
Linux th0ma7-router 4.14.151 #0 Tue Nov 5 23:26:40 2019 mips GNU/Linux
```

Description:
Configured following your really well made documentation (seriously).  Using stubby over dnsmasq without DNSSEC and working really well AFAIK.  As I have setup centralized loggin I decided to activate loggin on stubby when I stunbled on this:  when setting to any log level below 2 I receive tons of error messages.  For some I'm not even sure if `.err` is the right channel... 

Adjusted using the following:
```
# uci set stubby.global.log_level='6'
# uci commit
# /etc/init.d/stubby restart
```
Here is the output right after startup:
```
$ ssh root@192.168.80.1

BusyBox v1.30.1 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 19.07.0-rc1, r10649-c4fdb377a2
 -----------------------------------------------------
root@th0ma7-router:~# logread -f
Tue Nov 19 21:25:46 2019 daemon.err stubby[12623]: [02:25:46.336163] STUBBY: Read config from file /var/etc/stubby/stubby.yml
Tue Nov 19 21:25:46 2019 daemon.err stubby[12623]: [02:25:46.353908] STUBBY: DNSSEC Validation is OFF
Tue Nov 19 21:25:46 2019 daemon.err stubby[12623]: [02:25:46.356893] STUBBY: Transport list is:
Tue Nov 19 21:25:46 2019 daemon.err stubby[12623]: [02:25:46.359121] STUBBY:   - TLS
Tue Nov 19 21:25:46 2019 daemon.err stubby[12623]: [02:25:46.361393] STUBBY: Privacy Usage Profile is Strict (Authentication required)
Tue Nov 19 21:25:46 2019 daemon.err stubby[12623]: [02:25:46.363650] STUBBY: (NOTE a Strict Profile only applies when TLS is the ONLY transport!!)
Tue Nov 19 21:25:46 2019 daemon.err stubby[12623]: [02:25:46.365923] STUBBY: Starting DAEMON....
Tue Nov 19 21:25:48 2019 daemon.err stubby[12623]: [02:25:48.890572] STUBBY: 2606:4700:4700::1111                     : Upstream   : TLS - Resps=     0, Timeouts  =     0, Best_auth =   None
Tue Nov 19 21:25:48 2019 daemon.err stubby[12623]: [02:25:48.892967] STUBBY: 2606:4700:4700::1111                     : Upstream   : TLS - Conns=     0, Conn_fails=     1, Conn_shuts=      0, Backoffs     =     0
Tue Nov 19 21:25:48 2019 daemon.err stubby[12623]: [02:25:48.895583] STUBBY: 2606:4700:4700::1001                     : Upstream   : TLS - Resps=     0, Timeouts  =     0, Best_auth =   None
Tue Nov 19 21:25:48 2019 daemon.err stubby[12623]: [02:25:48.898012] STUBBY: 2606:4700:4700::1001                     : Upstream   : TLS - Conns=     0, Conn_fails=     1, Conn_shuts=      0, Backoffs     =     0
Tue Nov 19 21:25:50 2019 daemon.err stubby[12623]: [02:25:50.251825] STUBBY: 2606:4700:4700::1111                     : Upstream   : TLS - Resps=     0, Timeouts  =     0, Best_auth =   None
Tue Nov 19 21:25:50 2019 daemon.err stubby[12623]: [02:25:50.254222] STUBBY: 2606:4700:4700::1111                     : Upstream   : TLS - Conns=     0, Conn_fails=     2, Conn_shuts=      0, Backoffs     =     0
Tue Nov 19 21:25:50 2019 daemon.err stubby[12623]: [02:25:50.256707] STUBBY: 2606:4700:4700::1111                     : Upstream   : !Backing off TLS on this upstream    - Will retry again in 2s at Wed Nov 20 02:25:52 2019
Tue Nov 19 21:25:50 2019 daemon.err stubby[12623]: [02:25:50.259310] STUBBY: 2606:4700:4700::1001                     : Upstream   : TLS - Resps=     0, Timeouts  =     0, Best_auth =   None
Tue Nov 19 21:25:50 2019 daemon.err stubby[12623]: [02:25:50.261787] STUBBY: 2606:4700:4700::1001                     : Upstream   : TLS - Conns=     0, Conn_fails=     2, Conn_shuts=      0, Backoffs     =     0
Tue Nov 19 21:25:50 2019 daemon.err stubby[12623]: [02:25:50.264267] STUBBY: 2606:4700:4700::1001                     : Upstream   : !Backing off TLS on this upstream    - Will retry again in 2s at Wed Nov 20 02:25:52 2019
Tue Nov 19 21:25:53 2019 daemon.err stubby[12623]: [02:25:53.924569] STUBBY: 2606:4700:4700::1001                     : Upstream   : TLS - Resps=     0, Timeouts  =     0, Best_auth =   None
Tue Nov 19 21:25:53 2019 daemon.err stubby[12623]: [02:25:53.924689] STUBBY: 2606:4700:4700::1001                     : Upstream   : TLS - Conns=     0, Conn_fails=     1, Conn_shuts=      0, Backoffs     =     1
Tue Nov 19 21:25:54 2019 daemon.err stubby[12623]: [02:25:54.419500] STUBBY: 2606:4700:4700::1111                     : Upstream   : TLS - Resps=     0, Timeouts  =     0, Best_auth =   None
Tue Nov 19 21:25:54 2019 daemon.err stubby[12623]: [02:25:54.423180] STUBBY: 2606:4700:4700::1111                     : Upstream   : TLS - Conns=     0, Conn_fails=     1, Conn_shuts=      0, Backoffs     =     1
Tue Nov 19 21:25:54 2019 daemon.err stubby[12623]: [02:25:54.425842] STUBBY: 2606:4700:4700::1001                     : Upstream   : TLS - Resps=     0, Timeouts  =     0, Best_auth =   None
Tue Nov 19 21:25:54 2019 daemon.err stubby[12623]: [02:25:54.428538] STUBBY: 2606:4700:4700::1001                     : Upstream   : TLS - Conns=     0, Conn_fails=     2, Conn_shuts=      0, Backoffs     =     1
Tue Nov 19 21:25:54 2019 daemon.err stubby[12623]: [02:25:54.432243] STUBBY: 2606:4700:4700::1001                     : Upstream   : !Backing off TLS on this upstream    - Will retry again in 4s at Wed Nov 20 02:25:58 2019
Tue Nov 19 21:25:54 2019 daemon.err stubby[12623]: [02:25:54.524054] STUBBY: 2606:4700:4700::1111                     : Upstream   : TLS - Resps=     0, Timeouts  =     0, Best_auth =   None
Tue Nov 19 21:25:54 2019 daemon.err stubby[12623]: [02:25:54.526790] STUBBY: 2606:4700:4700::1111                     : Upstream   : TLS - Conns=     0, Conn_fails=     2, Conn_shuts=      0, Backoffs     =     1
Tue Nov 19 21:25:54 2019 daemon.err stubby[12623]: [02:25:54.529802] STUBBY: 2606:4700:4700::1111                     : Upstream   : !Backing off TLS on this upstream    - Will retry again in 4s at Wed Nov 20 02:25:58 2019
Tue Nov 19 21:26:00 2019 daemon.err stubby[12623]: [02:26:00.830511] STUBBY: 2606:4700:4700::1111                     : Upstream   : TLS - Resps=     0, Timeouts  =     0, Best_auth =   None
Tue Nov 19 21:26:00 2019 daemon.err stubby[12623]: [02:26:00.830631] STUBBY: 2606:4700:4700::1111                     : Upstream   : TLS - Conns=     0, Conn_fails=     1, Conn_shuts=      0, Backoffs     =     2
Tue Nov 19 21:26:00 2019 daemon.err stubby[12623]: [02:26:00.830818] STUBBY: 2606:4700:4700::1001                     : Upstream   : TLS - Resps=     0, Timeouts  =     0, Best_auth =   None
Tue Nov 19 21:26:00 2019 daemon.err stubby[12623]: [02:26:00.830901] STUBBY: 2606:4700:4700::1001                     : Upstream   : TLS - Conns=     0, Conn_fails=     1, Conn_shuts=      0, Backoffs     =     2
Tue Nov 19 21:26:01 2019 daemon.err stubby[12623]: [02:26:01.632708] STUBBY: 2606:4700:4700::1111                     : Upstream   : TLS - Resps=     0, Timeouts  =     0, Best_auth =   None
Tue Nov 19 21:26:01 2019 daemon.err stubby[12623]: [02:26:01.635085] STUBBY: 2606:4700:4700::1111                     : Upstream   : TLS - Conns=     0, Conn_fails=     2, Conn_shuts=      0, Backoffs     =     2
Tue Nov 19 21:26:01 2019 daemon.err stubby[12623]: [02:26:01.637564] STUBBY: 2606:4700:4700::1111                     : Upstream   : !Backing off TLS on this upstream    - Will retry again in 8s at Wed Nov 20 02:26:09 2019
Tue Nov 19 21:26:01 2019 daemon.err stubby[12623]: [02:26:01.640268] STUBBY: 2606:4700:4700::1001                     : Upstream   : TLS - Resps=     0, Timeouts  =     0, Best_auth =   None
Tue Nov 19 21:26:01 2019 daemon.err stubby[12623]: [02:26:01.642695] STUBBY: 2606:4700:4700::1001                     : Upstream   : TLS - Conns=     0, Conn_fails=     2, Conn_shuts=      0, Backoffs     =     2
Tue Nov 19 21:26:01 2019 daemon.err stubby[12623]: [02:26:01.645193] STUBBY: 2606:4700:4700::1001                     : Upstream   : !Backing off TLS on this upstream    - Will retry again in 8s at Wed Nov 20 02:26:09 2019
Tue Nov 19 21:26:11 2019 daemon.err stubby[12623]: [02:26:11.719088] STUBBY: 1.0.0.1                                  : Upstream   : TLS - Resps=     8, Timeouts  =     0, Best_auth =Success
[... cut ...] Upstream   : TLS - Conns=     1, Conn_fails=     0, Conn_shuts=      1, Backoffs     =     0
Tue Nov 19 21:26:13 2019 daemon.err stubby[12623]: [02:26:13.427906] STUBBY: 1.1.1.1                                  : Upstream   : TLS - Resps=    14, Timeouts  =     0, Best_auth =Success
Tue Nov 19 21:26:13 2019 daemon.err stubby[12623]: [02:26:13.428027] STUBBY: 1.1.1.1                                  : Upstream   : TLS - Conns=     1, Conn_fails=     0, Conn_shuts=      1, Backoffs     =     0
```

salim-b, 2019-11-20:
Hi @th0ma7 

I'm not a maintainer (I just fixed one line in the README some time ago). I think @jonathanunderwood or @iamperson347 are the right persons... :)

jonathanunderwood, 2019-11-21:
I am not sure I understand the problem. You've set log level to 6, so you're getting very verbose logging, which is expected.

I am sure I must be missing something here - can you help me understand your problem better?

th0ma7, 2019-11-22:
So I guess my explanation wasn't sufficiently clear.. Let me try to explain differently.

Setting the `log_level` to 6 does:
```
'6': INFO   - Informational message
```
But I did tried evey other levels and only level 2 makes sense in terms of amount of logging being shown.  Level 2 is only for critical which sort of surprises me that every other messages from stubby get on the error channel.

Question is: are the logs above indicating a problem or not?

As such I'm guessing there may be two issues here:
1. I might actually have a configuration issue leading to error messages above
2. I also think that "normal" log messages are not presented at the right syslogging channel therefore adjust stubby for proper messaging isn't fully available.  Perhaps some of the message above should be in *.debug channel instead or elsewhere?

My current configuration is the following:
```
# uci show stubby
stubby.global=stubby
stubby.global.manual='0'
stubby.global.trigger='wan'
stubby.global.dns_transport='GETDNS_TRANSPORT_TLS'
stubby.global.tls_authentication='1'
stubby.global.tls_query_padding_blocksize='128'
stubby.global.appdata_dir='/var/lib/stubby'
stubby.global.edns_client_subnet_private='1'
stubby.global.idle_timeout='10000'
stubby.global.round_robin_upstreams='1'
stubby.global.listen_address='127.0.0.1@5453' '0::1@5453' '192.168.80.1@5453'
stubby.global.log_level='2'
stubby.@resolver[0]=resolver
stubby.@resolver[0].address='2606:4700:4700::1111'
stubby.@resolver[0].tls_auth_name='cloudflare-dns.com'
stubby.@resolver[1]=resolver
stubby.@resolver[1].address='2606:4700:4700::1001'
stubby.@resolver[1].tls_auth_name='cloudflare-dns.com'
stubby.@resolver[2]=resolver
stubby.@resolver[2].address='1.1.1.1'
stubby.@resolver[2].tls_auth_name='cloudflare-dns.com'
stubby.@resolver[3]=resolver
stubby.@resolver[3].address='1.0.0.1'
stubby.@resolver[3].tls_auth_name='cloudflare-dns.com'
```

jonathanunderwood, 2019-11-23:
This really seems like an issue you should report upstream to the stubby developers - it doesn't seem like an openwrt specific problem. Stubby on openwrt seems to be functioning as it would on any other system, right?

jonathanunderwood, 2019-12-10:
It seems like there's no action to take here from the perspective of the OpenWRT Stubby package, right? That being the case, can you close this?

th0ma7, 2019-12-10:
Agreed, not much to do from this side.  Thnx for replying.

closed 2019-12-10 by th0ma7