openwrt/packages #10620: building image and "/lib/functions/network.sh: No such file or directory"

mistepien, 2019-11-22:
I got the error during building image for ramips with these options:

`make image PROFILE=ubiquiti_edgerouterx PACKAGES="luci-ssl -wpad-mini -hostapd-common -hostapd dnsmasq haveged -urandom-seed avahi-nodbus-daemon luci-proto-ppp  screen pv htop  avahi-daemon-service-ssh tcpdump iperf iperf3 luci-app-watchcat irqbalance ca-bundle  luci-app-ddns -odhcpd  -odhcpd-ipv6only -odhcp6c nmap-ssl openssh-server openssh-sftp-server  drill -urngd luci-proto-wireguard snmpd lsof iptables-mod-ipopt kmod-ipt-ipopt ethtool ipset  stubby prometheus-node-exporter-lua-netstat prometheus-node-exporter-lua-openwrt"
`

This is the error:

> Finalizing root filesystem...
/root/openwrt-imagebuilder-ramips-mt7621.Linux-x86_64/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/prometheus-node-exporter-lua: line 7: /lib/functions/network.sh: No such file or directory
Enabling avahi-daemon
Enabling boot
Enabling bootcount
Enabling cron
Enabling ddns
Enabling dnsmasq
Enabling done
Enabling dropbear
Enabling firewall
Enabling gpio_switch
Enabling haveged
Enabling irqbalance
Enabling led
Enabling log
Enabling network
./etc/init.d/prometheus-node-exporter-lua: line 7: /lib/functions/network.sh: No such file or directory
Enabling prometheus-node-exporter-lua
Enabling rpcd
Enabling set-irq-affinity
Enabling snmpd
Enabling sshd
Enabling stubby
Enabling sysctl
Enabling sysfixtime
Enabling sysntpd
Enabling system
Enabling ucitrack
Enabling uhttpd
Enabling umount
Enabling watchcat

mistepien, 2019-12-15:
I got the same error trying build 19.07-rc2. 
It seems to be problem of prometheus, not openwrt as such:
https://bugs.openwrt.org/index.php?do=details&task_id=2633

feckert, 2019-12-16:
The problem is this line https://github.com/openwrt/packages/blob/23460f34e56ac0b4270707c103b36341e563cfaa/utils/prometheus-node-exporter-lua/files/etc/init.d/prometheus-node-exporter-lua#L7

If you use the imagebuilder this line could not get evaluated.
We have two options
* move the import to the function start_service
https://github.com/openwrt/packages/blob/23460f34e56ac0b4270707c103b36341e563cfaa/utils/prometheus-node-exporter-lua/files/etc/init.d/prometheus-node-exporter-lua#L13

or

* add `${IPKG_INSTROOT}` so we have `. ${IPKG_INSTROOT}/lib/functions/network.sh`

mistepien, 2019-12-16:
On 16/12/2019 14:42, Florian Eckert wrote:
> The problem is this line https://github.com/openwrt/packages/blob/23460f34e56ac0b4270707c103b36341e563cfaa/utils/prometheus-node-exporter-lua/files/etc/init.d/prometheus-node-exporter-lua#L7
>
> If you [...]
The second options is less invasive.

neheb, 2020-01-28:
I pushed a fix: https://github.com/openwrt/packages/commit/3aa009478aa6a7562688a413e7e54318969873a3

closed 2020-01-28 by neheb