openwrt/packages #10623: collectd: rrd_update_r failed

dengqf6, 2019-11-23:
Maintainer: @hnyman @jow- 
Environment: IPQ40xx, OpenWrt master r11565-20fdfe37d2, collectd 5.10.0

Description:
collectd's sensors plugin does not work properly

Floods system log with errors:
```
Sat Nov 23 09:49:22 2019 daemon.err collectd[3469]: rrdtool plugin: rrd_update_r failed: /tmp/rrd/OpenWrt/sensors-ath10k_hwmon-isa-0000/temperature-temp1.rrd: illegal attempt to update using time 1574473763 when last update time is 1574473763 (minimum one second step)
Sat Nov 23 09:49:52 2019 daemon.err collectd[3469]: rrdtool plugin: rrd_update_r failed: /tmp/rrd/OpenWrt/sensors-ath10k_hwmon-isa-0000/temperature-temp1.rrd: illegal attempt to update using time 1574473793 when last update time is 1574473793 (minimum one second step)
Sat Nov 23 09:50:22 2019 daemon.err collectd[3469]: rrdtool plugin: rrd_update_r failed: /tmp/rrd/OpenWrt/sensors-ath10k_hwmon-isa-0000/temperature-temp1.rrd: illegal attempt to update using time 1574473823 when last update time is 1574473823 (minimum one second step)
Sat Nov 23 09:50:52 2019 daemon.err collectd[3469]: rrdtool plugin: rrd_update_r failed: /tmp/rrd/OpenWrt/sensors-ath10k_hwmon-isa-0000/temperature-temp1.rrd: illegal attempt to update using time 1574473853 when last update time is 1574473853 (minimum one second step)
```
Intermittent graph
[image]

hnyman, 2019-11-23:
Works quite ok for mvebu WRT3200ACM.
[image]

That error, submitting two data points from the same moment combined to gaps in data series, may indicate some problem either with your system clock, lm-sensors library getting stuck or collectd sensors module getting stuck. 

But there is nothing Openwrt specific for sensors module in OpenWrt. You may need to debug it further by yourself.

dengqf6, 2019-11-24:
`sensors` detects two sensors with same name. I think that is the cause.
```
root@OpenWrt:~# sensors 
ath10k_hwmon-isa-0000
Adapter: ISA adapter
temp1:        +77.0°C  

ath10k_hwmon-isa-0000
Adapter: ISA adapter
temp1:        +76.0°C  
```

hnyman, 2019-11-24:
> detects two sensors with same name

Sounds like you ran into something similar as upstream issue https://github.com/collectd/collectd/issues/2905

That is likely nothing really OpenWrt specific.

Likely you need to debug by yourself, and hopefully author a suitable patch and get that into upstream, collectd and/or lm-sensors sources.

You might start by enabling the "debug" mode when compiling your own collectd, and then see from logs what happens.
https://github.com/openwrt/packages/blob/master/utils/collectd/Makefile#L233

(Hint: drop all other plugins while you are testing debug. Otherwise you get swamped with collectd logging. When I have been debugging collectd, I have added additional fprintf statements to get more exact failure points. )

7217043955, 2022-05-27:
I encountered the same problem! The ipq4019 that had two wifi 2.4g and 5g, one sensor for each frequency, unfortunately those two sensors had same name, both are ```ath10k_hwmon-isa-0000 - temp1```, if I enabled both frequency, it will be conflicted and the rrd will shows infinite ```illegal attempt to update using time....```error in system log and the graphic will show nothing, if I disable one of them, the error goes away and the graphic displays with no problem. 
The solutions could be distinguishing two sensors, make it
```ath10k_hwmon-isa-0000 - temp1```
```ath10k_hwmon-isa-0000 - temp2```
or
```ath10k_hwmon-isa-0001 - temp1```
```ath10k_hwmon-isa-0002 - temp1```
some thing like that, or distinguish them in the plug-in.
However, I don't know where to start, as long as I had a clue, I'll see what my limited programing knowledge can do.

7217043955, 2022-09-05:
I may have some ideas now! although it's merely a guess.
Look at the two sensor chip names `ath10k_hwmon-isa-0000`, they both end up with `-0000` which is incorrect, this should be the sensor's chip address, and look at the code from lm-sensors: 
https://github.com/lm-sensors/lm-sensors/blob/1667b850a1ce38151dae17156276f981be6fb557/lib/sysfs.c#L663-L669
```
	if ((!subsys || !strcmp(subsys, "platform") ||
			!strcmp(subsys, "of_platform"))) {
		/* must be new ISA (platform driver) */
		if (sscanf(dev_name, "%*[a-z0-9_].%d", &entry->chip.addr) != 1)
			entry->chip.addr = 0;
		entry->chip.bus.type = SENSORS_BUS_TYPE_ISA;
		entry->chip.bus.nr = 0;
```
The chip address is acquired through the device name formatted as %*[a-z0-9_].%d, if this fails, 0 is used instead.
And look at an example device name that ipq4019 uses:
`a000000.wifi `
As you can see, the format doesn't match.
This could be solved to add an else if to support the format:
```
		if (sscanf(dev_name, "%*[a-z0-9_].%d", &entry->chip.addr) != 1)
		else if (sscanf(dev_name, "%x.%*[a-z0-9]", &entry->chip.addr) ! = 1)
		else entry->chip.addr = 0;
```
The ideal result should be chip names like this:
```
ath10k_hwmon-isa-a000000
ath10k_hwmon-isa-a800000
```
This solves the conflict and should stops the collectd error and the luci-app-statistics should display two graphics now.
But I don't know how to compile and load it to openwrt to test my theory, so any help on that?

7217043955, 2022-09-05:
@hnyman

7217043955, 2022-09-05:
It worked! Although I used some sneaky way to modify the code, just run `make menuconfig` , select right target and package, then run `make package/lm-sensors/compile`, then go to the directory` ./openwrt-21.02.3/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lm-sensors-3-6-0/lib/sysfs.c` manually edit the code then run `make package/lm-sensors/compile` again, then find the libsensors5 and lm-sensors ipk at `./openwrt-21.02.3/bin/packages/arm_cortex-a7_neon-vfpv4/packages/` and upload and install it on the openwrt. The result:
[image]
[image]

hnyman, 2022-09-06:
As it works, you should file a patch/PR  for the upstream lm-sensors at their site.

7217043955, 2022-09-06:
I have made a pull request on lm-sensors' master, but I saw their last activity was back to Feb 24, 2021, so I don't know how long it will be reviewed, so I post my patch here in case someone need it, the patch is modified to fit lm-sensors 3.6.0 that openwrt 21.02.3 uses.
```
From 6da486d7415e426ec983baa5f8edafe0ff4c4171 Mon Sep 17 00:00:00 2001
From: 7217043955 <85880133+7217043955@users.noreply.github.com>
Date: Tue, 6 Sep 2022 15:48:11 +0800
Subject: [PATCH] Add support for dev_name formatted like "a000000.wifi"

---
 lib/sysfs.c | 5 +++--
 1 file changed, 3 insertions(+), 2 deletions(-)

diff --git a/lib/sysfs.c b/lib/sysfs.c
index 9516cec4..c1d43465 100644
--- a/lib/sysfs.c
+++ b/lib/sysfs.c
@@ -663,8 +663,9 @@ static int classify_device(const char *dev_name,
 	if ((!subsys || !strcmp(subsys, "platform") ||
 			!strcmp(subsys, "of_platform"))) {
 		/* must be new ISA (platform driver) */
-		if (sscanf(dev_name, "%*[a-z0-9_].%d", &entry->chip.addr) != 1)
-			entry->chip.addr = 0;
+		if (sscanf(dev_name, "%*[a-zA-Z0-9_]%*1[.:]%d", &entry->chip.addr) == 1);
+		else if (sscanf(dev_name, "%x.%*s", &entry->chip.addr) == 1);
+		else entry->chip.addr = 0;
 		entry->chip.bus.type = SENSORS_BUS_TYPE_ISA;
 		entry->chip.bus.nr = 0;
 	} else if (subsys && !strcmp(subsys, "acpi")) {

```

closed 2022-11-18 by 1715173329