openwrt/packages #10634: [logrotate] version bump 3.15.1

n8v8R, 2019-11-25:
Maintainer: @bk138 
Contributer: @neheb 
Environment: Master & 19.07

Description:
upstream release https://github.com/logrotate/logrotate/releases/tag/3.15.1

n8v8R, 2019-11-25:
Apparently the repo has DCO requiring to sign off on a PR which however however Github's web UI does not cater for

https://github.community/t5/How-to-use-Git-and-GitHub/Feature-request-DCO-signoff-through-Github-PR-s/td-p/24784

>This defeats the point of allowing edits through Github's web interface, which is that people should be able to make fast, easy edits without having to download the whole repo and make changes with a local editor.

neheb, 2019-11-25:
You could just add

n8v8R, 2019-11-25:
Look, I do not mind trying to contribute a PR but I do mind wasting my time by having to jump through hoops and having to dig through documentation of DCO just for making a simple PR.

Nor am I interested to exhibit my email address for grabbing.

closed 2019-11-25 by n8v8R