openwrt/packages #10635: [tor] support sandboxing for torrc

n8v8R, 2019-11-25:
Maintainer: @hauke @tripolar 
Environment: Master & 19.07

Description:
https://2019.www.torproject.org/docs/tor-manual.html.en#Sandbox

Without sandbox support it prints in the log
> This version of Tor was built without support for sandboxing. To build with support for sandboxing on Linux, you must have libseccomp and its necessary header files (e.g. seccomp.h).

tripolar, 2019-11-25:
feel free to submit a patch

n8v8R, 2019-11-25:
Someone with the capability to generate a patch would certainly have done so instead of opening an issue here.

Feel at liberty to close if you are not inclined to look at the feature and how to commit it.

n8v8R, 2019-11-25:
Been considering to submit a patch but learned that it a signing off is required which though is not available through Github's web UI.  Thus no point in asking for a PR.

ja-pa, 2019-11-29:
@n8v8R Hi, I can create patch for that. But I'm not sure in which builds is SECCOMP activated so it won't be compiled with this option in every target.

n8v8R, 2019-12-03:
@ja-pa appreciate the offer and would be obliged if you could create the patch. SECCOMP is already installed on my node.