openwrt/packages #10641: nut: configure: error: libgd not found, required for CGI build

stintel, 2019-11-25:
Maintainer: @cshoredaniel 
Environment: OpenWrt master r11591-c5497ebb1c on x86/64

Description:
Unable to build nut because it cannot find libgd although I have `CONFIG_PACKAGE_libgd=y` in my .config.

```
make[2]: Entering directory '/home/stijn/Development/LEDE/openwrt-packages/net/nut'
touch /home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/nut-2.7.4/.prepared_21d8468662afaeea785acea868290ba9_6664517399ebbbc92a37c5bb081b5c53_check
. /home/stijn/Development/LEDE/source/include/shell.sh; gzip -dc /home/stijn/Development/LEDE/source/dl/nut-2.7.4.tar.gz | tar -C /home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/nut-2.7.4/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/nut-2.7.4

Applying ./patches/0001-Add-compatibility-with-openssl-1.1.0.patch using plaintext: 
patching file clients/upsclient.c
patching file m4/nut_check_libopenssl.m4
patching file server/netssl.c
Hunk #1 succeeded at 386 (offset -1 lines).

Applying ./patches/0002-Fix-check-for-empty-string.patch using plaintext: 
patching file clients/upssched.c
touch /home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/nut-2.7.4/.prepared_21d8468662afaeea785acea868290ba9_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/nut-2.7.4/.configured_*
rm -f /home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/stamp/.nut_installed
(cd /home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/nut-2.7.4; rm -f aclocal.m4; if [ -f ./configure.ac ] || [ -f ./configure.in ]; then [ -d ./autom4te.cache ] && rm -rf autom4te.cache; [ -e ./config.rpath ] || ln -s /home/stijn/Development/LEDE/source/scripts/config.rpath ./config.rpath; touch NEWS AUTHORS COPYING ABOUT-NLS ChangeLog; AUTOM4TE=/home/stijn/Development/LEDE/source/staging_dir/host/bin/autom4te AUTOCONF=/home/stijn/Development/LEDE/source/staging_dir/host/bin/autoconf AUTOMAKE=/home/stijn/Development/LEDE/source/staging_dir/host/bin/automake ACLOCAL=/home/stijn/Development/LEDE/source/staging_dir/host/bin/aclocal AUTOHEADER=/home/stijn/Development/LEDE/source/staging_dir/host/bin/autoheader LIBTOOLIZE=/home/stijn/Development/LEDE/source/staging_dir/host/bin/libtoolize LIBTOOL=/home/stijn/Development/LEDE/source/staging_dir/host/bin/libtool M4=/home/stijn/Development/LEDE/source/staging_dir/host/bin/m4 AUTOPOINT=true LIBTOOLIZE='/home/stijn/Development/LEDE/source/staging_dir/host/bin/libtoolize --install' /home/stijn/Development/LEDE/source/staging_dir/host/bin/autoreconf -v -f -i -s -B /home/stijn/Development/LEDE/source/staging_dir/host/share/aclocal -I /home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/host/share/aclocal -I /home/stijn/Development/LEDE/source/staging_dir/hostpkg/share/aclocal -I /home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/share/aclocal -I m4 -I . . || true; fi; );
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: /home/stijn/Development/LEDE/source/staging_dir/host/bin/aclocal -I /home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/host/share/aclocal -I /home/stijn/Development/LEDE/source/staging_dir/hostpkg/share/aclocal -I /home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/share/aclocal -I m4 -I . --force -I m4
autoreconf: configure.ac: tracing
autoreconf: running: /home/stijn/Development/LEDE/source/staging_dir/host/bin/libtoolize --install --force
OpenWrt-libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, `.'.
OpenWrt-libtoolize: linking file `./config.guess'
OpenWrt-libtoolize: linking file `./config.sub'
OpenWrt-libtoolize: linking file `./install-sh'
OpenWrt-libtoolize: linking file `./ltmain.sh'
OpenWrt-libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
OpenWrt-libtoolize: linking file `m4/libtool.m4'
OpenWrt-libtoolize: linking file `m4/ltoptions.m4'
OpenWrt-libtoolize: linking file `m4/ltsugar.m4'
OpenWrt-libtoolize: linking file `m4/ltversion.m4'
OpenWrt-libtoolize: linking file `m4/lt~obsolete.m4'
autoreconf: running: /home/stijn/Development/LEDE/source/staging_dir/host/bin/autoconf --include=/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/host/share/aclocal --include=/home/stijn/Development/LEDE/source/staging_dir/hostpkg/share/aclocal --include=/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/share/aclocal --include=m4 --include=. --prepend-include=/home/stijn/Development/LEDE/source/staging_dir/host/share/aclocal --force
autoreconf: running: /home/stijn/Development/LEDE/source/staging_dir/host/bin/autoheader --include=/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/host/share/aclocal --include=/home/stijn/Development/LEDE/source/staging_dir/hostpkg/share/aclocal --include=/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/share/aclocal --include=m4 --include=. --prepend-include=/home/stijn/Development/LEDE/source/staging_dir/host/share/aclocal --force
autoreconf: running: /home/stijn/Development/LEDE/source/staging_dir/host/bin/automake --add-missing --force-missing
autoreconf: Leaving directory `.'
(cd /home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/nut-2.7.4/./; if [ -x ./configure ]; then find /home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/nut-2.7.4/ -name config.guess | xargs -r chmod u+w; find /home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/nut-2.7.4/ -name config.guess | xargs -r -n1 cp --remove-destination /home/stijn/Development/LEDE/source/scripts/config.guess; find /home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/nut-2.7.4/ -name config.sub | xargs -r chmod u+w; find /home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/nut-2.7.4/ -name config.sub | xargs -r -n1 cp --remove-destination /home/stijn/Development/LEDE/source/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -O2 -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/nut-2.7.4=nut-2.7.4 -Wformat -Werror=format-security -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-O2 -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/nut-2.7.4=nut-2.7.4 -Wformat -Werror=format-security -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-O2 -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/nut-2.7.4=nut-2.7.4 -Wformat -Werror=format-security -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/include -I/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/usr/include -I/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/include/fortify -I/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/include " LDFLAGS="-L/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib -L/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/lib -L/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/usr/lib -L/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/lib -znow -zrelro "   ./configure --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  --sysconfdir=/etc/nut --datadir=/usr [... cut ...]
checking for gd library flags... -Wl,-O1 -Wl,--as-needed -lgd -lm -lz -lpng16 -lz -lfreetype -lfontconfig -lfreetype -ljpeg -ltiff -lwebp
checking for gd.h... no
checking for gdfontmb.h... no
checking for library containing gdImagePng... no
configure: error: libgd not found, required for CGI build
make[2]: *** [Makefile:554: /home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/nut-2.7.4/.configured_dcb1d064782d4867bdfd8957d6342e06] Error 1
make[2]: Leaving directory '/home/stijn/Development/LEDE/openwrt-packages/net/nut'
time: package/feeds/packages/nut/compile#14.92#5.36#20.24
```

stintel, 2019-11-25:
@neheb mind having a look since you made the pkgconfig changes that are likely the cause of this.

neheb, 2019-11-26:
Looking at it closer, that commit is already applied on your end. pkg-config is not finding the proper file. Neither with pkg-config nor with pkgconf.

Speaking of which, I posted that patch on the mailing list. Could give that a try.

cshoredaniel, 2019-11-27:
@stintel Hmmm....odd, master has been working for me since the pkgconfig changes.  Which commit/branch of packages are you building from?

stintel, 2019-11-27:
Building from master 7d6cde6327239d0a1fc54cc5f8dd9794345eec1e. After reverting all libgd commits up to and including cf5e16ac70dd02869f561611b5da07b6e3bab4f6, I can build nut again.

I suspect it is related to building x86/64 target on x86/64 host, and host include leakage. I will test https://github.com/openwrt/packages/pull/10648 now and report back.

cshoredaniel, 2019-12-17:
Trying out #10648 tonight or tomorrow.

cshoredaniel, 2019-12-25:
Can you show other packages in your build?  I noticed that the buildbots are not failing on NUT, so this might be one of those cases where it depends on what configure finds in staging.  I'm building now, so I'll see if an only base + nut build works, but if so, then it's more of a challenge to find the culprit.

cshoredaniel, 2019-12-25:
So far I cannot reproduce this.  I have just installed package libgd-dev on my host and am trying again (to see if it is a host leakage issue).

cshoredaniel, 2019-12-25:
@stintel I see checking via gdlib-config... 2.2.5 found in your logs -- what distro are you using and what version?  My distro has the same version of libgd as OpenWrt (3.0.5) which is why there isn't an obvious problem.  Even with libgd-dev I cannot reproduce this error.

cshoredaniel, 2019-12-25:
@stintel @micmac1  I tried to build with Debian jessie (which has a 2.x libgd) but with current snapshot SDK I can't build with Jessie due not a new enough version of Python3 (wants 3.5, Jessie only has 3.4), so rather than try and find distro with a sufficiently old libgd and new enough python, I'll wait until I have more info about what an environment I can reproduce in is.

Also, was this a clean build with master?

micmac1, 2019-12-27:
It's not easy to repro I'd say. And it's complicated (a bit anyway). In @stintel 's log it's visible that libgd pkg-config wasn't found. That I can repro easily, even with a current master checkout:

`checking for gd version via pkg-config... checking via gdlib-config... none found`

Why is that? libgd installs a pkg-config file after all. Well, the test in m4/nut_check_libgd.m4 is wrong:

```
GD_VERSION=`pkg-config --silence-errors --modversion gdlib 2>/dev/null`
if test "$?" != "0" -a -n "${GD_VERSION}"; then
```

It only continues when pkg-config returns something that is not 0, which it should only do when pkg-config fails.

Continuing with @stinel 's log we see that gdlib-config is found. But that is not OpenWrt's gdlib-config, because our libgd package doesn't install such a file in staging. So it must've been picked up from @stintel 's build host. Which in the end results in build failure:

`configure: error: libgd not found, required for CGI build`

I don't think there's any need for you to try to repro this. The patch for m4/nut_check_libgd.m4 currently included in nut is obviously not correct. The test described above is wrong, but also the block following the test contains a mistake:

`CFLAGS"`pkg-config --silence-errors --cflags gdllib 2>/dev/null`"`

Note the missing equal sign after CFLAGS (there are some backticks you can't see here because of github's markup).

The pull request #10648 does away with the faulty patch and provides gd libs and includes via configure args to nut, and that will just work fine.

cshoredaniel, 2019-12-27:
@micmac1 Ok, you've convinced me.

commit cdd660a4 (2019-12-27): nut: fix CGI setup

This commit makes

 - the libgd dependency (as well as specifying libs and includes)
 - the configure argument "--with-cgi"

dependant on whether the package nut-web-cgi is selected.

nut-web-cgi is also added to PKG_CONFIG_DEPENDS.

Resolves: #10641
commit 154da8aa (2020-01-20): nut: fix CGI setup

This commit makes

 - the libgd dependency (as well as specifying libs and includes)
 - the configure argument "--with-cgi"

dependant on whether the package nut-web-cgi is selected.

nut-web-cgi is also added to PKG_CONFIG_DEPENDS.

Resolves: #10641

(cherry-picked from cdd660a41dbfb87c9a253a7ebf856d7b733b752f)
closed 2019-12-27 by hnyman