openwrt/packages #10642: keepalived: build_dir/target-x86_64_musl/keepalived-2.0.19/keepalived/vrrp/vrrp_ipset.c:324: undefined reference to `keepalived_modprobe'

stintel, 2019-11-25:
Maintainer: @feckert 
 Environment: OpenWrt master r11591-c5497ebb1c on x86/64

Description:
keepalived no longer builds since e2689219be11418902f6a366017dadc80f6cb60e
```
...
make[6]: Entering directory '/home/build/openwrt/build_dir/target-x86_64_musl/keepalived-2.0.19/keepalived'
x86_64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../lib  -D_GNU_SOURCE -I/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/include -I/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/usr/include -I/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/include/fortify -I/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/include -DNETSNMP_NO_INLINE -D_FORTIFY_SOURCE=2 -DNETSNMP_ENABLE_IPV6 -DNETSNMP_REMOVE_U64 -D_FORTIFY_SOURCE=2 -Ulinux -Dlinux=linux -I/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/include -I/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/usr/include -I/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/include/fortify -I/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/include -I/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/include/libnl-tiny -I/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/include  -I./include -I ../lib -I/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/include -I/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/usr/include -I/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/include/fortify -I/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/include  -D_GNU_SOURCE -g -O2 -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/keepalived-2.0.19=keepalived-2.0.19 -Wformat -Werror=format-security -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -I/home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.85 -Wall -Wextra -Wunused -Wstrict-prototypes -Walloca -Walloc-zero -Warray-bounds=2 -Wattribute-alias -Wbad-function-cast -Wcast-align -Wcast-qual -Wchkp -Wdate-time -Wdisabled-optimization -Wdouble-promotion -Wduplicated-branches -Wduplicated-cond -Wfloat-conversion -Wfloat-equal -Wformat-overflow -Wformat-security -Wformat-signedness -Wformat-truncation -Wframe-larger-than=5120 -Wimplicit-fallthrough=3 -Winit-self -Winline -Wjump-misses-init -Wlogical-op -Wmissing-declarations -Wmissing-field-initializers -Wmissing-prototypes -Wnested-externs -Wnormalized -Wnull-dereference -Wold-style-definition -Woverlength-strings -Wpointer-arith -Wredundant-decls -Wshadow -Wshift-overflow=2 -Wstack-protector -Wstrict-overflow=4 -Wstrict-prototypes -Wstringop-overflow=2 -Wsuggest-attribute=cold -Wsuggest-attribute=const -Wsuggest-attribute=format -Wsuggest-attribute=malloc -Wsuggest-attribute=noreturn -Wsuggest-attribute=pure -Wsync-nand -Wtrampolines -Wundef -Wuninitialized -Wunknown-pragmas -Wunsuffixed-float-constants -Wunused-const-variable=2 -Wunused-macros -Wvariadic-macros -Wwrite-strings -fPIE -Wformat -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -O2 -O2 -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/net-snmp-5.8=net-snmp-5.8 -Wformat -Werror=format-security -fstack-protector-strong -Wl,-z,now -Wl,-z,relro -fpic -fno-strict-aliasing -O2 -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/net-snmp-5.8=net-snmp-5.8 -Wformat -fstack-protector-strong -Wl,-z,now -Wl,-z,relro -fpic  -O2 -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/keepalived-2.0.19=keepalived-2.0.19 -Wformat -Werror=format-security -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -I/home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.85  -MT main.o -MD -MP -MF .deps/main.Tpo -c -o main.o main.c
mv -f .deps/main.Tpo .deps/main.Po
x86_64-openwrt-linux-musl-gcc -g -O2 -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/keepalived-2.0.19=keepalived-2.0.19 -Wformat -Werror=format-security -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -I/home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.85 -Wall -Wextra -Wunused -Wstrict-prototypes -Walloca -Walloc-zero -Warray-bounds=2 -Wattribute-alias -Wbad-function-cast -Wcast-align -Wcast-qual -Wchkp -Wdate-time -Wdisabled-optimization -Wdouble-promotion -Wduplicated-branches -Wduplicated-cond -Wfloat-conversion -Wfloat-equal -Wformat-overflow -Wformat-security -Wformat-signedness -Wformat-truncation -Wframe-larger-than=5120 -Wimplicit-fallthrough=3 -Winit-self -Winline -Wjump-misses-init -Wlogical-op -Wmissing-declarations -Wmissing-field-initializers -Wmissing-prototypes -Wnested-externs -Wnormalized -Wnull-dereference -Wold-style-definition -Woverlength-strings -Wpointer-arith -Wredundant-decls -Wshadow -Wshift-overflow=2 -Wstack-protector -Wstrict-overflow=4 -Wstrict-prototypes -Wstringop-overflow=2 -Wsuggest-attribute=cold -Wsuggest-attribute=const -Wsuggest-attribute=format -Wsuggest-attribute=malloc -Wsuggest-attribute=noreturn -Wsuggest-attribute=pure -Wsync-nand -Wtrampolines -Wundef -Wuninitialized -Wunknown-pragmas -Wunsuffixed-float-constants -Wunused-const-variable=2 -Wunused-macros -Wvariadic-macros -Wwrite-strings -fPIE -Wformat -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -O2 -O2 -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/net-snmp-5.8=net-snmp-5.8 -Wformat -Werror=format-security -fstack-protector-strong -Wl,-z,now -Wl,-z,relro -fpic -fno-strict-aliasing -O2 -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/net-snmp-5.8=net-snmp-5.8 -Wformat -fstack-protector-strong -Wl,-z,now -Wl,-z,relro -fpic  -O2 -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/keepalived-2.0.19=keepalived-2.0.19 -Wformat -Werror=format-security -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -I/home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.85  -L/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib -L/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/lib -L/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/usr/lib -L/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/lib -znow -zrelro -pie -Wl,-z,relro -Wl,-z,now -L/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/lib -L/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib -L/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/lib -L/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/usr/lib -znow -zrelro  -L/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/usr/lib -L/home/stijn/Development/LEDE/source/staging_dir/target-x86_64_musl/lib -L/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/usr/lib -L/home/stijn/Development/LEDE/source/staging_dir/toolchain-x86_64_gcc-8.3.0_musl/lib -znow -zrelro  -o keepalived main.o core/libcore.a  vrrp/libvrrp.a bfd/libbfd.a core/libcore.a ../lib/libl [... cut ...]
make[5]: Leaving directory '/home/build/openwrt/build_dir/target-x86_64_musl/keepalived-2.0.19/keepalived'
make[4]: *** [Makefile:430: all-recursive] Error 1
make[4]: Leaving directory '/home/build/openwrt/build_dir/target-x86_64_musl/keepalived-2.0.19'
make[3]: *** [Makefile:271: /home/stijn/Development/LEDE/source/build_dir/target-x86_64_musl/keepalived-2.0.19/.built] Error 2
make[3]: Leaving directory '/home/stijn/Development/LEDE/openwrt-packages/net/keepalived'
time: package/feeds/packages/keepalived/compile#34.52#12.19#47.28```

feckert, 2019-11-26:
@stintel Thanks for reporting.
But I could not reproduce this on my own builds.
Is there an output from https://buildbot.openwrt.org/master/images/builders/x86%2F64
The builds look fine x86_64 the target is shown as green?

stintel, 2019-11-27:
It seems to compile on the build slaves indeed. The ipk is available for the x86/64 target. Must be another host related problem then.

feckert, 2019-11-29:
Is this then still a problem?

stintel, 2019-11-29:
It is, as it reliably doesn't build for me on my host (Gentoo x86/64). I'll try to do some more digging this weekend.

stintel, 2019-12-01:
I am also seeing it when building for octeon/mips64.

stintel, 2020-03-04:
This is still a problem after the 2.0.20 bump, and happens both on my x86/64 as ppc64le build hosts. The builds on my ppc64le host are automated and start from scratch every time, so it's not some leftover in my local git clone.

Maybe you can reproduce it with my [diffconfig](https://gist.githubusercontent.com/stintel/6e7f05fac09298e5e949143d5461e5ce/raw/ca6f627382f3d031ba3e80d1057016a539bd73f1/OpenWrt-APU2.config)?

stintel, 2020-03-04:
I suspect it is due to my config resulting in these two configure options being passed:
`--disable-lvs --disable-libipset-dynamic`. If you look at this commit, keepalived_modprobe is not being included with those options:

https://github.com/acassen/keepalived/commit/72c4e54e9

feckert, 2020-03-04:
Let's just ping @pqarmitage maybe he has a clue where the problem might be?
@pqarmitage and hint from your side was the problem could be?

stintel, 2020-03-04:
Reported it upstream at https://github.com/acassen/keepalived/issues/1516

pqarmitage, 2020-03-04:
The problem was caused by --disable-libipset-dynamic not checking the correct define generated by configure. keepalived commit [42c020a](https://github.com/pqarmitage/keepalived/commit/42c020aab6e51d09f22a2e4a33ce6fd73009e2dc) resolves the issue.

feckert, 2020-03-04:
@stintel should we backport this? Or should we wait for keepalived next release?

stintel, 2020-03-04:
> @stintel should we backport this? Or should we wait for keepalived next release?

As this breaks build with a valid combination of OpenWrt config settings this should be backported.

commit 7b552f29 (2020-03-05): keepalived: fix building with disable libipvs dynamic

Backport of upstream staging commit to fix compilation issue with x86_64 as
ppc64le build hosts.
https://github.com/pqarmitage/keepalived/commit/42c020aab6e51d09f22a2e4a33ce6fd73009e2dc

This fixes #10642.
closed 2020-03-05 by neheb