openwrt/packages #10644: procps-ng watch & pmap no longer being built for snapshot packages

notmyrealhandle, 2019-11-26:
Maintainer: @tiagogaspar8
Environment: arm_cortex-a15_neon-vfpv4 (others as well?), snapshot

Description:
about a month or two ago I noticed that procps-ng-watch & procps-ng-pmap are no longer in snapshot packages [here](http://downloads.openwrt.org/snapshots/packages/arm_cortex-a15_neon-vfpv4/packages/) (I think procps-ng sub packages are also missing).

Does anyone know why? @jow suggested a build failure but I don't see any fail logs for these packages [here](https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a15_neon-vfpv4/packages/).

I did find [this procps-ng commit](https://git.openwrt.org/?p=feed/packages.git;a=commit;h=071b15ecc0e7dd631e287ad670690d3fdb3a29d3) from 1 Oct 2019 which changes the default procps-ng package selection from all to none and I'm wondering if this caused (most of) them to no longer be built by the build system.

Is there someother way to get the functionality of these packages when creating a snapshot image with imagebuilder? (I know I could get watch inside busybox but I belive this requires compiling an image).

There is a forum topic [here](https://forum.openwrt.org/t/procps-ng-watch-procps-ng-pmap-no-longer-in-snapshot-packages/48525); however, I seem to be the only person who has noticed and bothered to comment/ask about it...

tiagogaspar8, 2019-11-26:
I'm actually not the maintainer of these packages but I have an idea of why these packages are not compiled.
So, I made that commit, it was because when some software needed some of the sub-packages, it automatically enabled all the sub-packages by default, leading to unneeded build, so I disabled it.
And for the packages not being built I believe that as there aren't any packages selecting it as a dependency, it isn't built because it doesn't get selected.
Am I right?

notmyrealhandle, 2019-11-26:
@tiagogaspar8 wrote:
> I'm actually not the maintainer of these packages

Apologies, I wasn't sure based on the makefile history... perhaps @neheb can point to the correct person.

As to how it happened, if it should be fixed and how are questions for someone familiar with openwrt's build system.  Your speculation above sounds reasonable to me.

tiagogaspar8, 2019-11-26:
Yes maybe he can help, but from what I saw in the makefile @kissg1988 is the actual maintainer of this package.
But why do you need that package? Is there any specific reason? Any package that needs it?

notmyrealhandle, 2019-11-26:
> But why do you need that package? Is there any specific reason? Any package that needs it?

I use watch fairly frequently (writing a sh script with a print/sleep command is a pita) and I'd rather not have to build an image just to get it.  I used pmap less (maby twice on openwrt in the past year that I can remember) but it shows up in the forums occasionally as a tool to help trouble shoot.

If I'm the only user who cares, then perhaps its time to let these packages go...

tiagogaspar8, 2019-11-26:
Don't get me wrong, I see your point! but we have to find out a way for it to build all the packages in the buildbot without when listed as a dependency of some package and for it to not select all sub-packages when listed as a dependency for some package.

micmac1, 2019-11-26:
I'm gonna go with @RussellSenior in #10124 ("Why?"). If somebody doesn't want the applets they can deselect them manually, but the applets (packages) should all be packaged by the build bots. I'm writing "packaged" on purpose, because the binaries are _built_ anyway. The only step "saved" is packaging them. This commit should be reverted IMHO.

neheb, 2019-11-26:
@micmac1 the reason I applied it is because selecting procps as a package made all of them available, when only pskill or something was needed.

The better question is, why does the buildbot not select all of them?

micmac1, 2019-11-26:
The commit only changed DEFAULT:=y to DEFAULT:=n, so I guess it must be that? ;) The only other package that uses DEFAULT:=n is rp-pppoe and it behaves the same (the DEFAULT:=n package isn't selected on the build bots).

Probably just removing the DEFAULT line altogether would work. Then all applets should be packaged by the build bots but not automatically when you compile a package that depends on procps-ng.

micmac1, 2019-12-01:
Hi all,

I checked the snapshot builds and watch and pmap are now packaged. This issue can be closed.

Kind regards,
Seb

closed 2019-12-01 by neheb

notmyrealhandle, 2019-12-01:
ty