openwrt/packages #10647: wifidog: failing to compile due to wolfssl

oxr463, 2019-11-26:
Maintainer: Michael Haas <haas@computerlinguist.org>
Environment: x86_64,18.06.5

Description: cannot compile wifidog due to wolfssl failing to build

```
FROM alpine:3
MAINTAINER Lucas Ramage <ramage.lucas@protonmail.com>

# https://openwrt.org/docs/guide-developer/build-system/install-buildsystem
RUN apk update && \
    apk add alpine-sdk \
            bash \
            bzip2 \
            coreutils \
            file \
            findutils \
            gawk \
            grep \
            linux-headers \
            ncurses-dev \
            perl \
            python2 \
            rsync \
            unzip \
            wget \
            xz \
            zlib-dev

ENV OPENWRT_MIRROR='https://downloads.openwrt.org/releases/18.06.5/targets/x86/64' \
    OPENWRT_SDK='openwrt-sdk-18.06.5-x86-64_gcc-7.3.0_musl.Linux-x86_64'

WORKDIR /tmp
RUN addgroup buildroot && \
    adduser -G buildroot -s /bin/bash -D buildroot && \
    echo "buildroot:buildroot"| /usr/sbin/chpasswd && \
    echo "buildroot    ALL=(ALL) NOPASSWD:ALL" >> /etc/sudoers && \
    wget "${OPENWRT_MIRROR}/${OPENWRT_SDK}.tar.xz" && \
    tar -xf "${OPENWRT_SDK}.tar.xz" && \
    mv "${OPENWRT_SDK}" /home/buildroot/openwrt-sdk && \
    chown -R buildroot:buildroot /home/buildroot/openwrt-sdk

USER buildroot
WORKDIR /home/buildroot/openwrt-sdk
RUN ./scripts/feeds update -a && \
    ./scripts/feeds install -a && \
    make defconfig && \
    make -j1 V=sc package/wifidog/compile
```

```
1590 make[2]: Entering directory '/home/buildroot/openwrt-sdk/feeds/base/package/libs/wolfssl'
1591 mkdir -p /home/buildroot/openwrt-sdk/dl
1592 SHELL= flock /home/buildroot/openwrt-sdk/tmp/.wolfssl-3.15.3-stable.zip.flock -c '  	/home/buildroot/openwrt-sdk/scripts/download.pl "/home/buildroot/openwrt-sdk/dl" "wolfssl-3.15.3-stable.zip" "dc97c07a7667b39a890e14f4b4a209f51524a4cabee7adb6c80822ee78c1f62a" "" "https://github.com/wolfSSL/wolfssl/archive/v3.15.3-stable"    '
1593 + curl -f --connect-timeout 20 --retry 5 --location --insecure https://github.com/wolfSSL/wolfssl/archive/v3.15.3-stable/wolfssl-3.15.3-stable.zip
1594   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
1595                                  Dload  Upload   Total   Spent    Left  Speed
1596 100   128    0   128    0     0    633      0 --:--:-- --:--:-- --:--:--   636
1597 100 5375k  100 5375k    0     0  11.2M      0 --:--:-- --:--:-- --:--:-- 11.2M
1598 touch /home/buildroot/openwrt-sdk/build_dir/target-x86_64_musl/wolfssl-3.15.3-stable/.prepared_52674977c652e218e6d8904aece84641_18f1e190c5d53547fed41a3eaa76e9e9_check
1599 . /home/buildroot/openwrt-sdk/include/shell.sh; unzip -q -d /home/buildroot/openwrt-sdk/build_dir/target-x86_64_musl/wolfssl-3.15.3-stable/.. /home/buildroot/openwrt-sdk/dl/wolfssl-3.15.3-stable.zip 
1600 [ ! -d ./src/ ] || cp -fpR ./src/. /home/buildroot/openwrt-sdk/build_dir/target-x86_64_musl/wolfssl-3.15.3-stable
1601 Applying ./patches/010-Make-RsaUnPad-constant-time-when-Block-Type-2-messag.patch using plaintext: 
1602 patching file src/internal.c
1603 patching file src/tls.c
1604 patching file wolfcrypt/src/integer.c
1605 patching file wolfcrypt/src/misc.c
1606 patching file wolfcrypt/src/rsa.c
1607 patching file wolfcrypt/src/sp_int.c
1608 patching file wolfcrypt/src/tfm.c
[... 29 lines cut ...]
1638 OpenWrt-libtoolize: linking file `m4/ltversion.m4'
1639 OpenWrt-libtoolize: linking file `m4/lt~obsolete.m4'
1640 autoreconf: running: /home/buildroot/openwrt-sdk/staging_dir/host/bin/autoconf --include=/home/buildroot/openwrt-sdk/staging_dir/target-x86_64_musl/host/share/aclocal --include=/home/buildroot/openwrt-sdk/staging_dir/hostpkg/share/aclocal --include=/home/buildroot/openwrt-sdk/staging_dir/target-x86_64_musl/usr/share/aclocal --include=m4 --include=. --prepend-include=/home/buildroot/openwrt-sdk/staging_dir/host/share/aclocal --force
1641 autoreconf: running: /home/buildroot/openwrt-sdk/staging_dir/host/bin/autoheader --include=/home/buildroot/openwrt-sdk/staging_dir/target-x86_64_musl/host/share/aclocal --include=/home/buildroot/openwrt-sdk/staging_dir/hostpkg/share/aclocal --include=/home/buildroot/openwrt-sdk/staging_dir/target-x86_64_musl/usr/share/aclocal --include=m4 --include=. --prepend-include=/home/buildroot/openwrt-sdk/staging_dir/host/share/aclocal --force
1642 autoreconf: running: /home/buildroot/openwrt-sdk/staging_dir/host/bin/automake --add-missing --force-missing
1643 configure.ac:21: installing 'build-aux/compile'
1644 configure.ac:23: installing 'build-aux/config.guess'
1645 configure.ac:23: installing 'build-aux/config.sub'
1646 configure.ac:26: installing 'build-aux/install-sh'
1647 configure.ac:26: installing 'build-aux/missing'
1648 Makefile.am: installing 'build-aux/depcomp'
1649 parallel-tests: installing 'build-aux/test-driver'
1650 autoreconf: Leaving directory `.'
1651 (cd /home/buildroot/openwrt-sdk/build_dir/target-x86_64_musl/wolfssl-3.15.3-stable/./; if [ -x ./configure ]; then find /home/buildroot/openwrt-sdk/build_dir/target-x86_64_musl/wolfssl-3.15.3-stable/ -name config.guess | xargs -r chmod u+w; find /home/buildroot/openwrt-sdk/build_dir/target-x86_64_musl/wolfssl-3.15.3-stable/ -name config.guess | xargs -r -n1 cp --remove-destination /home/buildroot/openwrt-sdk/scripts/config.guess; find /home/buildroot/openwrt-sdk/build_dir/target-x86_64_musl/wolfssl-3.15.3-stable/ -name config.sub | xargs -r chmod u+w; find /home/buildroot/openwrt-sdk/build_dir/target-x86_64_musl/wolfssl-3.15.3-stable/ -name config.sub | xargs -r -n1 cp --remove-destination /home/buildroot/openwrt-sdk/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/buildroot/openwrt-sdk/build_dir/target-x86_64_musl/wolfssl-3.15.3-stable:wolfssl-3.15.3-stable -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/buildroot/openwrt-sdk/build_dir/target-x86_64_musl/wolfssl-3.15.3-stable:wolfssl-3.15.3-stable -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/buildroot/openwrt-sdk/build_dir/target-x86_64_musl/wolfssl-3.15.3-stable:wolfssl-3.15.3-stable -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic " CPPFLAGS="-I/home/buildroot/openwrt-sdk/staging_dir/target-x86_64_musl/usr/include -I/home/buildroot/openwrt-sdk/staging_dir/target-x86_64_musl/include -I/home/buildroot/openwrt-sdk/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/usr/include -I/home/buildroot/openwrt-sdk/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/include/fortify -I/home/buildroot/openwrt-sdk/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/include " LDFLAGS="-L/home/buildroot/openwrt-sdk/staging_dir/target-x86_64_musl/usr/lib -L/home/buildroot/openwrt-sdk/staging_dir/target-x86_64_musl/lib -L/home/buildroot/openwrt-sdk/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/usr/lib -L/home/buildroot/openwrt-sdk/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/lib -znow -zrelro "   ./configure --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  --enable-opensslextra --enable-sni --enable-stunnel --disable-examples --disable-leanpsk --disable-leantls  --enable-ipv6 --enable-aesccm --disable-chacha --enable-ecc --enable-supportedcurves --enable-dh --enable-arc4 --enable-des3 --enable-psk --enable-session-ticket --enable-ocsp --enable-ocspstapling --enable-ocspstapling2 --enable-wpas --enable-sha512 --enable-fo [... cut ...]
1657 time: package/feeds/base/wolfssl/compile#9.51#1.51#12.73
1658 make[1]: *** [package/Makefile:113: package/feeds/base/wolfssl/compile] Error 2
1659 make[1]: Leaving directory '/home/buildroot/openwrt-sdk'
1660 make: *** [/home/buildroot/openwrt-sdk/include/toplevel.mk:211: package/wifidog/compile] Error 2
1661 The command '/bin/sh -c ./scripts/feeds update -a &&     ./scripts/feeds install -a &&     make defconfig &&     make -j1 V=sc package/wifidog/compile' returned a non-zero code: 2
```

oxr463, 2019-11-26:
See also: https://github.com/openwrt/packages/issues/1562

neheb, 2019-11-26:
Is this on an older branch?

In any case, ping @cotequeiroz

oxr463, 2019-11-26:
Nope, just run the following:

```sh
docker build -t openwrt/wifidog .
```

cotequeiroz, 2019-11-27:
It appears that `$TOPDIR` gets assigned to something with an unbalanced quote, leading to the syntax error.  The error is not in line 2 of the `./configure` script (a harmless comment), but in line 2 of `/home/buildroot/openwrt-sdk/include/site/x86_64`: `. $TOPDIR/include/site/linux`.
It probably blew up in wolfssl's hands because it was the first package to use autotools.  I'm short on time to debug this right now, but I'm hoping--if I'm right--this can point to the right direction.

neheb, 2020-07-23:
It seems this was fixed.

closed 2020-07-23 by neheb