openwrt/packages #10651: error: recursive dependency detected!

zhaochengwei, 2019-11-27:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:
[image]

after run 
```
./scripts/feeds update packages
./scripts/feeds install -a -p packages
```

I got 
`fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..HEAD
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..af57e55a88cca6eacb7eb8d2670fc2096119bb99
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..HEAD
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..af57e55a88cca6eacb7eb8d2670fc2096119bb99
WARNING: Makefile 'package/feeds/packages/bmx7-dnsupdate/Makefile' has a dependency on 'bmx7', which does not exist
WARNING: Makefile 'package/feeds/packages/bmx7-dnsupdate/Makefile' has a dependency on 'bmx7-json', which does not exist
WARNING: Makefile 'package/feeds/luci/freifunk-gwcheck/Makefile' has a dependency on 'olsrd-mod-dyn-gw-plain', which does not exist
WARNING: Makefile 'package/feeds/luci/freifunk-mapupdate/Makefile' has a dependency on 'olsrd-mod-nameservice', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-ahcp/Makefile' has a dependency on 'ahcpd', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-olsr/Makefile' has a dependency on 'olsrd', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-olsr/Makefile' has a dependency on 'olsrd-mod-jsoninfo', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-olsr-services/Makefile' has a dependency on 'olsrd', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-olsr-services/Makefile' has a dependency on 'olsrd-mod-nameservice', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-olsr-viz/Makefile' has a dependency on 'olsrd', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-olsr-viz/Makefile' has a dependency on 'olsrd-mod-txtinfo', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-mod-freifunk-community/Makefile' has a dependency on 'olsrd', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-mod-freifunk-community/Makefile' has a dependency on 'olsrd-mod-dyn-gw-plain', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-mod-freifunk-community/Makefile' has a dependency on 'olsrd-mod-jsoninfo', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-mod-freifunk-community/Makefile' has a dependency on 'olsrd-mod-nameservice', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-mod-freifunk-community/Makefile' has a dependency on 'olsrd-mod-watchdog', which does not exist
WARNING: Makefile 'package/feeds/packages/prometheus-node-exporter-lua/Makefile' has a dependency on 'bmx6', which does not exist
WARNING: Makefile 'package/feeds/packages/prometheus-node-exporter-lua/Makefile' has a dependency on 'bmx6-json', which does not exist
WARNING: Makefile 'package/feeds/packages/prometheus-node-exporter-lua/Makefile' has a dependency on 'bmx7', which does not exist
WARNING: Makefile 'package/feeds/packages/prometheus-node-exporter-lua/Makefile' has a dependency on 'bmx7-json', which does not exist
tmp/.config-package.in:55747:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:55747:   symbol PACKAGE_wget is selected by PACKAGE_wget
Segmentation fault (core dumped)
make: *** [/home/laiyi/phab/whc/mango_hub/whale_router/include/toplevel.mk:137: menuconfig] Error 139`

feckert, 2019-11-27:
> Please make sure that the issue subject starts with `<package-name>: `
>
> Also make sure that the package is maintained in this repository and not in base which should be submitted at [...]

Please update the template

neheb, 2020-01-17:
This is quite vague. Closing.

closed 2020-01-17 by neheb