openwrt/packages #10661: stubby: Could not parse config file "/var/etc/stubby/stubby.yml": Permission denied

yriorrio, 2019-11-28:
#9991 reports the same problem. It was opened on September 14 but closed by the reporting user on September 16 without any explanation.

Description: when starting stubby it fails to open /var/etc/stubby/stubby.yml due to insufficient permissions. Using chown and chmod to give root full access makes no difference. 

stubby version: 0.2.6-1
openwrt version: OpenWrt SNAPSHOT r11583-68fb38548b / LuCI Master git-19.327.83508-5e1253f (community firmware by Davidc502)
device: Linksys WRT1200AC

Temporary workaround: SSH into router and launch it with command **stubby -g** (keep in mind it will read configuration from /etc/stubby/stubby.yml and not /etc/config/stubby)

dibdot, 2019-12-02:
@jonathanunderwood @iamperson347

jonathanunderwood, 2019-12-10:
Was closed because we couldn't reproduce it. I still can't reproduce this. Can you give a solid reproducer?

schm0, 2020-01-07:
Starting stubby from a terminal session via /etc/init.d/stubby works fine but if you try to launch it from the luci gui it will fail with the permission error.

xPaw, 2020-01-10:
I just installed stubby on openwrt-19.07 and it fails to start after rebooting (or from luci) with the permission error.

Starting it manually from ssh does work.

neheb, 2020-02-13:
https://github.com/openwrt/packages/blob/master/net/stubby/files/stubby.init#L216

needs -m 0755 added i think.

net147, 2020-02-21:
When trying to start stubby from Startup tab of web interface, it gives the Permission denied error.

Logging into system and checking permissions:
# ls -l /var/etc/stubby/stubby.yml 
-rw-------    1 root     root           721 Feb 21 19:18 /var/etc/stubby/stubby.yml

I would prefer having it owned by root and read-only for other users.

jonathanunderwood, 2020-02-21:
Could those seeing this problem test the following changes to `/etc/init.d/stubby:

1. Change [Line 216](https://github.com/openwrt/packages/blob/master/net/stubby/files/stubby.init#L216) to read:

```bash
mkdir -p -m 0755 "$stubby_config_dir"
```

2. Add a line after [Line 227](https://github.com/openwrt/packages/blob/master/net/stubby/files/stubby.init#L227) which reads:

```bash
        chmod 0755 "$stubby_config"
```

and report back whether or not this solves the problem?

jonathanunderwood, 2020-02-21:
Nb. Edited the previous snippets - please see above, not the original email notification.

neheb, 2020-02-21:
It's some issue with the init script definitely.

It should be cleaned up.

closed 2020-02-25 by neheb