openwrt/packages #10663: ttyd: doesn't bind to (alias) interfaces properly

karlp, 2019-11-28:
Maintainer: @tsl0922
Environment: ath79, 19.07 branch

Description: When the ttyd config contains an interface alias like "@lan"
```
# uci show ttyd
ttyd.@ttyd[0]=ttyd
ttyd.@ttyd[0].interface='@lan'
ttyd.@ttyd[0].command='/usr/libexec/login.sh'
```

ttyd is incorrectly binding to the first address on that interface.  Given a network config like
```
config interface 'lan'
	option ifname 'eth0'
	option proto 'dhcp'

config interface 'plc'
	option proto 'static'
	option ifname 'eth0'
	option ipaddr '192.168.50.1'
	option netmask '255.255.255.0'
```

Which results in eth0 having multiple addresses, like so:
```
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether c4:93:00:03:b6:79 brd ff:ff:ff:ff:ff:ff
    inet 192.168.50.1/24 brd 192.168.50.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet 192.168.253.133/24 brd 192.168.253.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::c693:ff:fe03:b679/64 scope link 
       valid_lft forever preferred_lft forever
```

ttyd ends up binding to _only_ one of the ip addresses, eg:
```
# netstat -an | grep 7681
tcp        0      0 192.168.50.1:7681       0.0.0.0:*               LISTEN
```

Changing the config to explicitly bind to eth0 doesn't work either, it still binds to only the first address on the interface.  I had to change the setting to "unspecified" to make it work

ysc3839, 2019-11-28:
I don't think Linux supports binding to multiple address.
http://man7.org/linux/man-pages/man7/ip.7.html
> When a process wants to receive new incoming packets or connections,
it should bind a socket to a local interface address using bind(2).
In this case, only one IP socket may be bound to any given local
(address, port) pair.

karlp, 2019-11-28:
well, dnsmasq manages it :)  I think you actually need to bind multiple times or something.  I know it's complicated, but the current behaviour is awkward.

ysc3839, 2019-11-28:
This still is an upstream (Libwebsockets) issue. Not ttyd, and not OpenWrt.

closed 2025-11-14 by GeorgeSapkin