openwrt/packages #10679: dnsmasq-full: Cannot satisfy dependencies (19.07.0-rc1)

th0ma7, 2019-11-30:
Maintainer: @ldir-EDB0 
Environment: 
arch: mips
model: TP-link WDR4300
Openwrt: 19.07.0-rc1

Description:
I probably overlooked but hit the following issue where unable to install `dnsmasq-full`.  Actually downloading only already gives the following warning and wonder if I can/should force-install:
```
# opkg install dnsmasq-full --download-only
Downloading http://downloads.openwrt.org/releases/19.07.0-rc1/packages/mips_24kc/base/dnsmasq-full_2.80-14_mips_24kc.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for dnsmasq-full:
 * 	kernel (= 4.14.151-1-342af9e4f67b3447c53216ab8e3b12a1)
 * opkg_install_cmd: Cannot install package dnsmasq-full.
```

neheb, 2019-12-01:
this belongs at https://bugs.openwrt.org/

closed 2019-12-01 by neheb

th0ma7, 2019-12-01:
I thought it was related to a package more than basesystem..
Bug now created:
* https://bugs.openwrt.org/index.php?do=details&task_id=2644