openwrt/packages #10687: nut-cgi: fails to create var/etc/nut/hosts.conf after 18.06.5 update

reedmideke, 2019-11-30:
Maintainer: @cshoredaniel
Environment: openwrt-18.06.5-ipq806x-tplink_c2600  

installed nut packages are
```
nut - 2.7.4-10
nut-common - 2.7.4-10
nut-driver-usbhid-ups - 2.7.4-10
nut-server - 2.7.4-10
nut-upsc - 2.7.4-10
nut-upscmd - 2.7.4-10
nut-upsmon - 2.7.4-10
nut-upsrw - 2.7.4-10
nut-web-cgi - 2.7.4-10
```

Description:

After sysupgrade from 18.06.4, attempting to log into the nut web interface on /nut fails with 
```
Error: hosts.conf unavailable 
```
syslog contains
```
Fri Nov 29 01:01:38 2019 daemon.notice procd: /etc/rc.d/S51nut-cgi: /etc/rc.common: eval: line 1: can't create /var/etc/nut/hosts.conf: nonexistent directory
```

This appears to be a startup order issue: 
/etc/init.d/nut-cgi has START=51 and attempts to output with >> to /var/etc/nut/hosts.conf

/etc/init.d/nut-server has START=70 and creates the directory

These values appear in the current 18.06 source
https://github.com/openwrt/packages/blob/openwrt-18.06/net/nut/files/nut-cgi.init
https://github.com/openwrt/packages/blob/openwrt-18.06/net/nut/files/nut-server.init

Adjusting the nut-cgi START to 71, disabling and re-nabling nut-cgi resolved the issue

In 18.06.4, nut-server was 50.

In the 19.07 branch nut-server is 70 and nut-cgi is 87, so it's probably already fixed there (but I haven't tested)

cshoredaniel, 2019-12-01:
Are you saying it didt work in 18.60.4 and doesn't in 18.06.5?  That's rather odd since I didn't change anything between those two versions.  I did have patch (probably in master) to create the needed directory / file if it didn't exist, but didn't backport it because I forgot that fix, and thought the only changes I had made were a feature enhancement to the CGI boot script.

reedmideke, 2019-12-01:
I believe it worked in 18.06.4, and it does not out of the box in .5

I think the change that broke it was nut-server initscript changing to a later start order, not a change in nut-cgi, possibly this commit https://github.com/openwrt/packages/commit/a2ab989c121a1177536004eafbfd64165280edf6

cshoredaniel, 2019-12-01:
Hmmm...I thought that had all landed before 18.06.4 but maybe I'm mistaken.  Anyway I'll do up a patch, and verify all is well in master and 19.07

cshoredaniel, 2019-12-17:
All is well in master and 19.07 ... patch build and test in progress...

cshoredaniel, 2019-12-20:
@reedmideke Just pushed a PR that should fix this.  It's quite a small patch.

PR #10846 (2019-12-21): nut: Fix NUT CGI startup script
Commit c1aa1f784c737283b281da7a541921a88a81e684 which backported
a number of fixes from master, missed one fix for nut-cgi.  If
nut-cgi is installed standalone or nut-cgi starts before nut-server
(the default in 18.04.5) then the configuration file directory does
not get create and therefore there is no CGI configuration file.
This commit fixes that.

Closes: #10687

Maintainer: me
Compile tested: x86-64, Libvirt VM, 18.06-SNAPSHOT Dec 12 2019 
Run tested: same, brought up a separate VM for each of:

* nut-server (using dummy-ups  and data from my real UPS)
* nu [... cut]
commit 45c95e86 (2019-12-21): nut: Fix NUT CGI startup script

Commit c1aa1f784c737283b281da7a541921a88a81e684 which backported
a number of fixes from master, missed one fix for nut-cgi.  If
nut-cgi is installed standalone or nut-cgi starts before nut-server
(the default in 18.04.5) then the configuration file directory does
not get create and therefore there is no CGI configuration file.
This commit fixes that.

Closes: #10687
closed 2020-01-12 by neheb