openwrt/packages #10694: arp-scan returns no results

alexhenrie, 2019-12-02:
When I use arp-scan on OpenWrt, I get "0 responded":

```
# arp-scan -I br-lan -l
Interface: br-lan, datalink type: EN10MB (Ethernet)
WARNING: Cannot open MAC/Vendor file /usr/share/arp-scan/ieee-oui.txt: No such file or directory
WARNING: Cannot open MAC/Vendor file /usr/share/arp-scan/ieee-iab.txt: No such file or directory
WARNING: Cannot open MAC/Vendor file /usr/share/arp-scan/mac-vendor.txt: No such file or directory
Starting arp-scan 1.9.5 with 256 hosts (https://github.com/royhills/arp-scan)

15 packets received by filter, 0 packets dropped by kernel
Ending arp-scan 1.9.5: 256 hosts scanned in 1.852 seconds (138.23 hosts/sec). 0 responded
```

This appears to be the same bug as:
* https://github.com/royhills/arp-scan/issues/42
* https://bugs.launchpad.net/ubuntu/+source/arp-scan/+bug/1849740
* https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=944129

Updating OpenWrt's version of arp-scan from 1.9.5 to 1.9.7 would probably fix the problem.

alexhenrie, 2019-12-02:
Pinging package maintainer @urusha

closed 2019-12-02 by feckert

alexhenrie, 2019-12-02:
The updated package does indeed fix the problem, thank you!

cvocvo, 2020-04-23:
@alexhenrie @urusha Probably a dumb question, but how do I get the v1.9.7 package on OpenWRT 18.06? As near as I can tell 18.086 is still stuck on version `1.9-40-g69b2f70-1`