openwrt/packages #10712: Mwan3track not working properly on wireguard interface

wackejohn, 2019-12-07:
Maintainer: @feckert 
Environment: OpenWrt SNAPSHOT, r11625-a512123a4b 
                      Mwan3 version: 2.8.2-2
Description:
I'm using mwan3 and wireguard as client with latest openwrt snapshot.
And it seemed that the issue was caused by mwan3 itself, when mwan3 was running ,the `ping -I $Device` not working, and then mwan3 would always mark the wireguard interface down, but the wireguard interface actually working...
Below is the test resault:
```
root@HOME-Router:~# /usr/bin/ping -I wana 8.8.8.8
PING 8.8.8.8 (8.8.8.8) from 192.168.120.2 wana: 56(84) bytes of data.
^C
--- 8.8.8.8 ping statistics ---
6 packets transmitted, 0 received, 100% packet loss, time 5233ms

root@HOME-Router:~# mwan3 stop
root@HOME-Router:~# mwan3 status
Interface status:
 interface wan1 is offline and tracking is down
 interface wan1_6 is offline and tracking is down
 interface wan2 is offline and tracking is down
 interface wan2_6 is offline and tracking is down
 interface wan3 is offline and tracking is down
 interface wan3_6 is offline and tracking is down
 interface wan4 is offline and tracking is down
 interface wan4_6 is offline and tracking is down
 interface wana is offline and tracking is down
 interface wana_6 is offline and tracking is down
 interface wanb is offline and tracking is down
 interface wanb_6 is offline and tracking is down

Current ipv4 policies:

Current ipv6 policies:

Directly connected ipv4 networks:

Directly connected ipv6 networks:

Active ipv4 user rules:

Active ipv6 user rules:

root@HOME-Router:~# /usr/bin/ping -I wana 8.8.8.8
PING 8.8.8.8 (8.8.8.8) from 192.168.120.2 wana: 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=58 time=135 ms
64 bytes from 8.8.8.8: icmp_seq=2 ttl=58 time=135 ms
64 bytes from 8.8.8.8: icmp_seq=3 ttl=58 time=135 ms
64 bytes from 8.8.8.8: icmp_seq=4 ttl=58 time=135 ms
64 bytes from 8.8.8.8: icmp_seq=5 ttl=58 time=136 ms
^C
--- 8.8.8.8 ping statistics ---
6 packets transmitted, 5 received, 16.6667% packet loss, time 5007ms
rtt min/avg/max/mdev = 134.652/135.242/136.296/0.592 ms
root@HOME-Router:~# mwan3 restart
uci: Entry not found
uci: Entry not found
RTNETLINK answers: File exists
RTNETLINK answers: File exists
RTNETLINK answers: File exists
RTNETLINK answers: File exists
RTNETLINK answers: File exists
RTNETLINK answers: File exists
RTNETLINK answers: File exists
RTNETLINK answers: File exists
RTNETLINK answers: File exists
RTNETLINK answers: File exists
RTNETLINK answers: File exists
RTNETLINK answers: File exists
RTNETLINK answers: File exists
RTNETLINK answers: File exists
[... 285 lines cut ...]
RTNETLINK answers: File exists
RTNETLINK answers: File exists
RTNETLINK answers: File exists
RTNETLINK answers: File exists
RTNETLINK answers: File exists
RTNETLINK answers: File exists
RTNETLINK answers: File exists
RTNETLINK answers: File exists
RTNETLINK answers: File exists
RTNETLINK answers: File exists
RTNETLINK answers: File exists
RTNETLINK answers: File exists
RTNETLINK answers: File exists
RTNETLINK answers: File exists
RTNETLINK answers: File exists
uci: Entry not found
uci: Entry not found
root@HOME-Router:~# /usr/bin/ping -I wana 8.8.8.8
PING 8.8.8.8 (8.8.8.8) from 192.168.120.2 wana: 56(84) bytes of data.
^C
--- 8.8.8.8 ping statistics ---
8 packets transmitted, 0 received, 100% packet loss, time 7247ms
```
And there is a discuss on the forum:
https://forum.openwrt.org/t/mwan3track-not-working-properly-on-wireguard/49557

wackejohn, 2020-01-21:
@feckert Any update about this issue?

feckert, 2020-01-21:
Sorry I have not further explored the subject.
But the errors 
```
RTNETLINK answers: File exists
uci: Entry not found
```

are strange. First of all we should investigate why this message are shown during mwan3 restart.

wackejohn, 2020-01-21:
> Sorry I have not further explored the subject.
> But the errors
> 
> ```
> RTNETLINK answers: File exists
> uci: Entry not found
> ```
> 
> are strange. First of all we should investigate why this message are shown during mwan3 restart.

These errors are always there, maybe it is caused by my mwan3 and firewall configs. Is there any thing I can do for these errors?

wackejohn, 2020-03-14:
@feckert 
I'm trying to debug myself, and I've found that the `mwan3_set_general_iptables()` in `lib/mwan3/mwan3.sh` caused ping failure.
```
			$IPT -A mwan3_hook \
				-j CONNMARK --restore-mark --nfmask $MMX_MASK --ctmask $MMX_MASK
			$IPT -A mwan3_hook \
				-m mark --mark 0x0/$MMX_MASK \
				-j mwan3_ifaces_in
			$IPT -A mwan3_hook \
				-m mark --mark 0x0/$MMX_MASK \
				-j mwan3_connected
			$IPT -A mwan3_hook \
				-m mark --mark 0x0/$MMX_MASK \
				-j mwan3_rules
			$IPT -A mwan3_hook \
				-j CONNMARK --save-mark --nfmask $MMX_MASK --ctmask $MMX_MASK
			$IPT -A mwan3_hook \
				-m mark ! --mark $MMX_DEFAULT/$MMX_MASK \
				-j mwan3_connected
```
I'm not goot at iptables, is there any way to get the ping working with wireguard interfaces?

feckert, 2020-05-29:
I have seen that you do not use the ping from busybox. I think this has everything to do with the interface binding that doesn't work by ip-utils, because the version is over 10 years old. In the pullrequest https://github.com/openwrt/packages/pull/12229 from @aaronjg there are some changes that would fix this. Please give them a try. And if it does not work please reopen this Issue.

wackejohn, 2020-05-29:
@feckert 
Thanks for your reply.
I've tried pullrequest [#12229](https://github.com/openwrt/packages/pull/12229) , but the ping still not working (both from busybox and new ip-utils).  

```
BusyBox v1.31.1 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r13398-48e3449945
 -----------------------------------------------------
root@HOME-Router_CZ:~# /bin/ping
BusyBox v1.31.1 () multi-call binary.

Usage: ping [OPTIONS] HOST

Send ICMP ECHO_REQUEST packets to network hosts

        -4,-6           Force IP or IPv6 name resolution
        -c CNT          Send only CNT pings
        -s SIZE         Send SIZE data bytes in packets (default 56)
        -i SECS         Interval
        -A              Ping as soon as reply is recevied
        -t TTL          Set TTL
        -I IFACE/IP     Source interface or IP address
        -W SEC          Seconds to wait for the first response (default 10)
                        (after all -c CNT packets are sent)
        -w SEC          Seconds until ping exits (default:infinite)
                        (can exit earlier with -c CNT)
        -q              Quiet, only display output at start
                        and when finished
        -p HEXBYTE      Pattern to use for payload
root@HOME-Router_CZ:~# /usr/bin/ping -V
ping from iputils reboot-13398-g48e3449945+
root@HOME-Router_CZ:~# /bin/ping -I wana 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
^C
--- 8.8.8.8 ping statistics ---
5 packets transmitted, 0 packets received, 100% packet loss
root@HOME-Router_CZ:~# /usr/bin/ping -I wana 8.8.8.8
PING 8.8.8.8 (8.8.8.8) from 192.168.120.2 wana: 56(84) bytes of data.
^C
--- 8.8.8.8 ping statistics ---
7 packets transmitted, 0 received, 100% packet loss, time 6166ms
```

aaronjg, 2020-05-29:
Can you confirm that busybox ping works when mwan3 is not running?

If that doesn't work, can you provide some additional diagnostics:

- Output of `ip rule list`

- Then add the following logging rules to you firewall, do a `ping -c1 -I wana 1.1.1.1`, and post the result from the system log?

```                                      
iptables --table mangle -I POSTROUTING 1 -d 1.1.1.1 -j LOG --log-prefix "postroute start "
iptables --table mangle -A POSTROUTING  -d 1.1.1.1 -j LOG --log-prefix "postroute end  " 
iptables --table mangle -I OUTPUT 1 -d 1.1.1.1 -j LOG --log-prefix "output start "
iptables --table mangle -A OUTPUT  -d 1.1.1.1 -j LOG --log-prefix "output end "
```

wackejohn, 2020-05-30:
@aaronjg 
The busybox ping works when I stop the mwan3:
```
root@HOME-Router_CZ:~# mwan3 stop
root@HOME-Router_CZ:~# mwan3 interfaces
Interface status:
 interface wan1 is offline and tracking is down
 interface wan1_6 is offline and tracking is down
 interface wan2 is offline and tracking is down
 interface wan2_6 is offline and tracking is down
 interface wan3 is offline and tracking is down
 interface wan3_6 is offline and tracking is down
 interface wan4 is offline and tracking is down
 interface wan4_6 is offline and tracking is down
 interface wana is offline and tracking is down
 interface wana_6 is offline and tracking is down
 interface wanb is offline and tracking is down
 interface wanb_6 is offline and tracking is down

root@HOME-Router_CZ:~# /bin/ping
BusyBox v1.31.1 () multi-call binary.

Usage: ping [OPTIONS] HOST

Send ICMP ECHO_REQUEST packets to network hosts

        -4,-6           Force IP or IPv6 name resolution
        -c CNT          Send only CNT pings
        -s SIZE         Send SIZE data bytes in packets (default 56)
        -i SECS         Interval
        -A              Ping as soon as reply is recevied
        -t TTL          Set TTL
        -I IFACE/IP     Source interface or IP address
        -W SEC          Seconds to wait for the first response (default 10)
                        (after all -c CNT packets are sent)
        -w SEC          Seconds until ping exits (default:infinite)
                        (can exit earlier with -c CNT)
        -q              Quiet, only display output at start
                        and when finished
        -p HEXBYTE      Pattern to use for payload
root@HOME-Router_CZ:~# /bin/ping -I wana 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
64 bytes from 8.8.8.8: seq=0 ttl=58 time=136.848 ms
64 bytes from 8.8.8.8: seq=1 ttl=58 time=136.043 ms
64 bytes from 8.8.8.8: seq=2 ttl=58 time=136.566 ms
64 bytes from 8.8.8.8: seq=3 ttl=58 time=137.378 ms
64 bytes from 8.8.8.8: seq=4 ttl=58 time=136.683 ms
^C
--- 8.8.8.8 ping statistics ---
5 packets transmitted, 5 packets received, 0% packet loss
round-trip min/avg/max = 136.043/136.703/137.378 ms
root@HOME-Router_CZ:~# /bin/ping -I wanb 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
64 bytes from 8.8.8.8: seq=0 ttl=55 time=141.829 ms
64 bytes from 8.8.8.8: seq=1 ttl=55 time=142.050 ms
64 bytes from 8.8.8.8: seq=2 ttl=55 time=141.761 ms
64 bytes from 8.8.8.8: seq=3 ttl=55 time=141.551 ms
64 bytes from 8.8.8.8: seq=4 ttl=55 time=141.491 ms
^C
--- 8.8.8.8 ping statistics ---
5 packets transmitted, 5 packets received, 0% packet loss
round-trip min/avg/max = 141.491/141.736/142.050 ms
```

And the output when mwan3 was running:
```
root@HOME-Router_CZ:~# ip rule list
0:      from all lookup local
1001:   from all iif pppoe-wan1 lookup 1
1003:   from all iif pppoe-wan2 lookup 3
1005:   from all iif pppoe-wan3 lookup 5
1007:   from all iif pppoe-wan4 lookup 7
1009:   from all iif wana lookup 9
1011:   from all iif wanb lookup 11
2001:   from all fwmark 0x100/0x3f00 lookup 1
2003:   from all fwmark 0x300/0x3f00 lookup 3
2005:   from all fwmark 0x500/0x3f00 lookup 5
2007:   from all fwmark 0x700/0x3f00 lookup 7
2009:   from all fwmark 0x900/0x3f00 lookup 9
2011:   from all fwmark 0xb00/0x3f00 lookup 11
2061:   from all fwmark 0x3d00/0x3f00 blackhole
2062:   from all fwmark 0x3e00/0x3f00 unreachable
32766:  from all lookup main
32767:  from all lookup default
root@HOME-Router_CZ:~# iptables --table mangle -I POSTROUTING 1 -d 8.8.8.8 -j LOG --log-prefix "postroute start "
root@HOME-Router_CZ:~# iptables --table mangle -A POSTROUTING  -d 8.8.8.8 -j LOG --log-prefix "postroute end  " 
root@HOME-Router_CZ:~# iptables --table mangle -I OUTPUT 1 -d 8.8.8.8 -j LOG --log-prefix "output start "
root@HOME-Router_CZ:~# iptables --table mangle -A OUTPUT  -d 8.8.8.8 -j LOG --log-prefix "output end "
root@HOME-Router_CZ:~# /bin/ping -c1 -I wana 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
^C
--- 8.8.8.8 ping statistics ---
1 packets transmitted, 0 packets received, 100% packet loss
root@HOME-Router_CZ:~# logread |grep "postroute start"
Sat May 30 08:40:14 2020 kern.warn kernel: [34815.895080] postroute start IN= OUT=wana SRC=192.168.120.2 DST=8.8.8.8 LEN=84 TOS=0x00 PREC=0x00 TTL=64 ID=49580 DF PROTO=ICMP TYPE=8 CODE=0 ID=37655 SEQ=0 MARK=0x900
root@HOME-Router_CZ:~# logread |grep "postroute end"
Sat May 30 08:40:14 2020 kern.warn kernel: [34815.910087] postroute end  IN= OUT=wana SRC=192.168.120.2 DST=8.8.8.8 LEN=84 TOS=0x00 PREC=0x00 TTL=64 ID=49580 DF PROTO=ICMP TYPE=8 CODE=0 ID=37655 SEQ=0 MARK=0x900
root@HOME-Router_CZ:~# logread |grep "output start"
Sat May 30 08:40:14 2020 kern.warn kernel: [34815.866656] output start IN= OUT=wana SRC=192.168.120.2 DST=8.8.8.8 LEN=84 TOS=0x00 PREC=0x00 TTL=64 ID=49580 DF PROTO=ICMP TYPE=8 CODE=0 ID=37655 SEQ=0
root@HOME-Router_CZ:~# logread |grep "output end"
Sat May 30 08:40:14 2020 kern.warn kernel: [34815.880484] output end IN= OUT=wana SRC=192.168.120.2 DST=8.8.8.8 LEN=84 TOS=0x00 PREC=0x00 TTL=64 ID=49580 DF PROTO=ICMP TYPE=8 CODE=0 ID=37655 SEQ=0 MARK=0x900
```

aaronjg, 2020-05-30:
Is that the correct src ip? It looks like it is a wan, but has a private IP address.

Can you share the output of 
`ip route list`
and
`ip route list table 9`

I am wondering if your VPN set up two half internet routes rather than one default route. Those would then be copied over to the other routing tables and could cause issues.

wackejohn, 2020-05-30:
@aaronjg 
wana and wanb were the wireguard vpn and worked as internet connections, so the src ip was correct.
The output:
```
root@HOME-Router_CZ:~# ip route list
default via 180.116.118.1 dev pppoe-wan1 proto static metric 10 
default via 114.228.153.1 dev pppoe-wan2 proto static metric 15 
default via 223.66.36.1 dev pppoe-wan3 proto static metric 20 
default via 183.213.154.1 dev pppoe-wan4 proto static metric 25 
114.228.153.1 dev pppoe-wan2 proto kernel scope link src 114.228.153.21 
180.116.118.1 dev pppoe-wan1 proto kernel scope link src 180.116.118.230 
183.213.154.1 dev pppoe-wan4 proto kernel scope link src 183.213.155.69 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 
192.168.2.0/24 dev vpn1 proto static scope link metric 110 
192.168.3.0/24 dev vpn1 proto static scope link metric 110 
192.168.120.0/24 dev wana proto static scope link metric 30 
192.168.150.0/24 dev wanb proto static scope link metric 40 
192.168.195.0/24 dev vpn0 proto static scope link metric 100 
192.168.195.2 dev vpn0 proto static scope link metric 100 
192.168.195.3 dev vpn0 proto static scope link metric 100 
192.168.195.4 dev vpn0 proto static scope link metric 100 
192.168.195.5 dev vpn0 proto static scope link metric 100 
192.168.195.6 dev vpn0 proto static scope link metric 100 
192.168.196.0/24 dev vpn1 proto static scope link metric 110 
192.168.196.2 dev vpn1 proto static scope link metric 110 
192.168.196.3 dev vpn1 proto static scope link metric 110 
223.66.36.1 dev pppoe-wan3 proto kernel scope link src 223.66.36.129 
root@HOME-Router_CZ:~# ip route list table 9
default dev wana scope link metric 30 
114.228.153.1 dev pppoe-wan2 proto kernel scope link src 114.228.153.21 
180.116.118.1 dev pppoe-wan1 proto kernel scope link src 180.116.118.230 
183.213.154.1 dev pppoe-wan4 proto kernel scope link src 183.213.155.69 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 
192.168.2.0/24 dev vpn1 proto static scope link metric 110 
192.168.3.0/24 dev vpn1 proto static scope link metric 110 
192.168.120.0/24 dev wana proto static scope link metric 30 
192.168.150.0/24 dev wanb proto static scope link metric 40 
192.168.195.0/24 dev vpn0 proto static scope link metric 100 
192.168.195.2 dev vpn0 proto static scope link metric 100 
192.168.195.3 dev vpn0 proto static scope link metric 100 
192.168.195.4 dev vpn0 proto static scope link metric 100 
192.168.195.5 dev vpn0 proto static scope link metric 100 
192.168.195.6 dev vpn0 proto static scope link metric 100 
192.168.196.0/24 dev vpn1 proto static scope link metric 110 
192.168.196.2 dev vpn1 proto static scope link metric 110 
192.168.196.3 dev vpn1 proto static scope link metric 110 
223.66.36.1 dev pppoe-wan3 proto kernel scope link src 223.66.36.129 
```

aaronjg, 2020-05-30:
It looks like you have no routes beyond the subnet for the wireguard interface. If you don't have a default route, then mwan3 isn't going to be able to copy anything over into the relevant table.

I think you need to configure your wireguard to add the default routes, or do it manually after the link comes up.

I'm curious how it is working without mwan3 enabled. Can you turn mwan3 off and post the same results from the iptables logs?

wackejohn, 2020-05-30:
The log when mwan3 stoped:
```
root@HOME-Router_CZ:~# /bin/ping -c1 -I wana 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
64 bytes from 8.8.8.8: seq=0 ttl=58 time=193.008 ms

--- 8.8.8.8 ping statistics ---
1 packets transmitted, 1 packets received, 0% packet loss
round-trip min/avg/max = 193.008/193.008/193.008 ms
root@HOME-Router_CZ:~# logread |grep "postroute start"
Sat May 30 10:14:51 2020 kern.warn kernel: [40492.986831] postroute start IN= OUT=wana SRC=192.168.120.2 DST=8.8.8.8 LEN=84 TOS=0x00 PREC=0x00 TTL=64 ID=5275 DF PROTO=ICMP TYPE=8 CODE=0 ID=18496 SEQ=0
root@HOME-Router_CZ:~# logread |grep "postroute end"
Sat May 30 10:14:51 2020 kern.warn kernel: [40493.000785] postroute end  IN= OUT=wana SRC=192.168.120.2 DST=8.8.8.8 LEN=84 TOS=0x00 PREC=0x00 TTL=64 ID=5275 DF PROTO=ICMP TYPE=8 CODE=0 ID=18496 SEQ=0
root@HOME-Router_CZ:~# logread |grep "output start"
Sat May 30 10:14:51 2020 kern.warn kernel: [40492.959591] output start IN= OUT=wana SRC=192.168.120.2 DST=8.8.8.8 LEN=84 TOS=0x00 PREC=0x00 TTL=64 ID=5275 DF PROTO=ICMP TYPE=8 CODE=0 ID=18496 SEQ=0
root@HOME-Router_CZ:~# logread |grep "output end"
Sat May 30 10:14:51 2020 kern.warn kernel: [40492.973292] output end IN= OUT=wana SRC=192.168.120.2 DST=8.8.8.8 LEN=84 TOS=0x00 PREC=0x00 TTL=64 ID=5275 DF PROTO=ICMP TYPE=8 CODE=0 ID=18496 SEQ=0
```
And how to add the default route for wireguard interface?

aaronjg, 2020-05-30:
Sorry, I don't use wireguard so I can't help you there. With OtherVPNs, the routes are often pushed to you and the VPN client then sets them up. You could do it manually with
`ip route add ...`

wackejohn, 2020-05-30:
I think it's not about the default thing, I've add the defalut route for wireguard interface, and restarted the mwan3, but the ping still not working.
```
BusyBox v1.31.1 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r13398-48e3449945
 -----------------------------------------------------
root@HOME-Router_CZ:~# ip route list
default via 180.116.118.1 dev pppoe-wan1 proto static metric 10 
default via 114.228.153.1 dev pppoe-wan2 proto static metric 15 
default via 223.66.36.1 dev pppoe-wan3 proto static metric 20 
default via 183.213.154.1 dev pppoe-wan4 proto static metric 25 
default via 192.168.120.1 dev wana metric 30 
default via 192.168.150.1 dev wanb metric 40 
114.228.153.1 dev pppoe-wan2 proto kernel scope link src 114.228.153.21 
180.116.118.1 dev pppoe-wan1 proto kernel scope link src 180.116.118.230 
183.213.154.1 dev pppoe-wan4 proto kernel scope link src 183.213.155.69 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 
192.168.2.0/24 dev vpn1 proto static scope link metric 110 
192.168.3.0/24 dev vpn1 proto static scope link metric 110 
192.168.120.0/24 dev wana proto static scope link metric 30 
192.168.150.0/24 dev wanb proto static scope link metric 40 
192.168.195.0/24 dev vpn0 proto static scope link metric 100 
192.168.195.2 dev vpn0 proto static scope link metric 100 
192.168.195.3 dev vpn0 proto static scope link metric 100 
192.168.195.4 dev vpn0 proto static scope link metric 100 
192.168.195.5 dev vpn0 proto static scope link metric 100 
192.168.195.6 dev vpn0 proto static scope link metric 100 
192.168.196.0/24 dev vpn1 proto static scope link metric 110 
192.168.196.2 dev vpn1 proto static scope link metric 110 
192.168.196.3 dev vpn1 proto static scope link metric 110 
223.66.36.1 dev pppoe-wan3 proto kernel scope link src 223.66.36.129 
root@HOME-Router_CZ:~# /bin/ping -c1 -I wana 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes

--- 8.8.8.8 ping statistics ---
1 packets transmitted, 0 packets received, 100% packet loss
```

wackejohn, 2020-05-30:
@feckert 
Can you reopen this issue for me please? thanks.

aaronjg, 2020-05-30:
It looks like you are running a development build. Can you try it with the latest stable release?

wackejohn, 2020-05-30:
@aaronjg 
The wireguard endpoint route was deleted and it's working well without the static out, but the mwan3track not working with or without the static route.
My device running the development build for years, so it's inconvenient to switch to stable release, but i think it will be the same situation.

aaronjg, 2020-06-15:
@wackejohn I just ran into this issue myself. It appears that something is handled differently when the outgoing interface is 192.168.X.X.

Can you try applying this commit, and see if it fixes things for you?

https://github.com/aaronjg/openwrt-packages/commit/3f1152b46da5e66254b7aee234877d2b6d4f0fb3

wackejohn, 2020-06-16:
@aaronjg 
I've tried the commit, but still no lucky...
```
root@HOME-Router_HB:~# /bin/ping
BusyBox v1.31.1 () multi-call binary.

Usage: ping [OPTIONS] HOST

Send ICMP ECHO_REQUEST packets to network hosts

        -4,-6           Force IP or IPv6 name resolution
        -c CNT          Send only CNT pings
        -s SIZE         Send SIZE data bytes in packets (default 56)
        -i SECS         Interval
        -A              Ping as soon as reply is recevied
        -t TTL          Set TTL
        -I IFACE/IP     Source interface or IP address
        -W SEC          Seconds to wait for the first response (default 10)
                        (after all -c CNT packets are sent)
        -w SEC          Seconds until ping exits (default:infinite)
                        (can exit earlier with -c CNT)
        -q              Quiet, only display output at start
                        and when finished
        -p HEXBYTE      Pattern to use for payload
root@HOME-Router_HB:~# /bin/ping -I wana 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
^C
--- 8.8.8.8 ping statistics ---
5 packets transmitted, 0 packets received, 100% packet loss
root@HOME-Router_HB:~# /bin/ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
64 bytes from 8.8.8.8: seq=1 ttl=116 time=37.902 ms
64 bytes from 8.8.8.8: seq=2 ttl=116 time=37.761 ms
64 bytes from 8.8.8.8: seq=4 ttl=116 time=37.525 ms
64 bytes from 8.8.8.8: seq=5 ttl=116 time=37.710 ms
64 bytes from 8.8.8.8: seq=7 ttl=116 time=38.157 ms
^C
--- 8.8.8.8 ping statistics ---
9 packets transmitted, 5 packets received, 44% packet loss
root@HOME-Router_HB:~# traceroute 8.8.8.8
traceroute to 8.8.8.8 (8.8.8.8), 30 hops max, 38 byte packets
 1  192.168.120.1 (192.168.120.1)  135.816 ms  138.259 ms  138.762 ms
 2  172.22.61.200 (172.22.61.200)  154.769 ms  152.138 ms  154.381 ms
 3  72.14.222.37 (72.14.222.37)  146.069 ms  139.710 ms  146.242 ms
 4  108.170.247.225 (108.170.247.225)  138.113 ms  108.170.247.129 (108.170.247.129)  143.132 ms  108.170.247.225 (108.170.247.225)  182.618 ms
 5  209.85.143.255 (209.85.143.255)  198.962 ms  66.249.94.217 (66.249.94.217)  254.052 ms  142.250.226.109 (142.250.226.109)  224.975 ms
 6  dns.google (8.8.8.8)  141.630 ms  141.739 ms  146.930 ms
```
And it seemed that the icmp packages didn't pass through the vpn.

aaronjg, 2020-06-16:
So, with the trace route it looks like the packets are going through your vpn (192.168.120.1) when you don’t specify the interface, but not when you do? Looks like the issue I ran into is different from the issue I had.

wackejohn, 2020-06-16:
@aaronjg 
As my mwan3 config:
```
config rule 'AntiGFW_DNS'
        option proto 'all'
        option dest_ip '8.8.8.8,8.8.4.4,208.67.222.222,208.67.220.220,199.91.73.222,178.79.131.110'
        option sticky '0'
        option family 'ipv4'
        option use_policy 'wana_wanb'
```
the icmp packets should pass through the vpn (wana or wanb) wether specify the interface or not.

aaronjg, 2020-06-16:
yes, but there are different rules that are applied to route them out of the interface when you specify the interface vs when you don't.

I am surprised that they are being blocked with the new patch. That should make the ICMP echo  packets skip the mwan3 rules entirely when you are specifying the bound interface.

What kernel version are you using? I'll do the best I can to help, but I may not be able to reproduce since you are on the snapshot...

A few more diagnostics - can you repeat the firewall logging with the new patch? I want to see if these packets are somehow still getting marked.

What is the output of:
`ip -4 addr ls wana`
`ipset list mwan3_source_v4`
`iptables --table mangle -S mwan3_hook`

jamesmacwhite, 2020-06-16:
So, not to try and create noise here, but I recently switched from OpenVPN to Wireguard with mwan3, evaluating Wireguard vs OpenVPN, I found the IPv6 side was a bit off, but the latest changes from @aaronjg, fixes those.

I have two Wireguard VPN interfaces as clients load balanced with Mullvad and it works OK. I'm on 19.07.3 on a Linksys WRT3200ACM. Currently running 2.8.6, but I'm aware you've tried the changes and it hasn't worked.

Have you tried stripping back the mwan3 config to a very minimal setup?

aaronjg, 2020-06-17:
James, not noise at all. Thanks for sharing your experience.  What sort of
IPv4 address are you getting from Mullvad? I wonder if some of john’s
problems is that he is getting a private, 192.168.X.x address from his vpn.

jamesmacwhite, 2020-06-17:
@aaronjg Mullvad provides an IPv4 in the 10.0.0.0/8 range and a single /128 fc00::/7 ULA as they use NAT66.

In order to have IPv4 and IPv6 working with mwan3, I have a `wg6` and `wgb6` as an alias to the main `wg` and `wgb` interfaces with the /128 ULA configured as a static, so I can have both IPv4 and IPv6 control. As with anything Wireguard currently, you always get a static IPv4 and IPv6 based on your private/public key.

The IPv6 side seems unrelated in this case, because it looks like @wackejohn can't even use the Wireguard inferface once mwan3 is enabled, but in my case:

```
root@linksys-wrt3200acm:~# traceroute -i wg openwrt.org
traceroute to openwrt.org (139.59.209.225), 30 hops max, 38 byte packets
 1  10.64.0.1 (10.64.0.1)  20.292 ms  25.235 ms  15.759 ms
 2  vlan817.bb2.lon7.uk.m247.com (185.200.118.99)  17.486 ms  15.898 ms  16.395 ms
 3  37.120.220.58 (37.120.220.58)  18.236 ms  21.289 ms  48.244 ms
 4  xe-2-0-0-0.core1.lon2.uk.m247.com (176.10.82.158)  19.059 ms  xe-2-2-2-0.core1.lon2.uk.m247.com (83.97.21.236)  24.275 ms  xe-2-0-0-0.core1.lon2.uk.m247.com (176.10.82.158)  20.991 ms
 5  89.44.212.108 (89.44.212.108)  38.722 ms  89.44.212.110 (89.44.212.110)  31.225 ms  89.44.212.4 (89.44.212.4)  35.174 ms
 6  37.120.128.253 (37.120.128.253)  29.299 ms  32.342 ms  27.820 ms
 7  fra2-edge1.digitalocean.com (80.81.195.151)  31.283 ms  33.836 ms  35.757 ms
 8  *  *  *
 9  *  *  *
10  wiki-01.infra.openwrt.org (139.59.209.225)  42.591 ms  31.464 ms  32.306 ms
````

```
root@linksys-wrt3200acm:~# traceroute -i wgb openwrt.org
traceroute to openwrt.org (139.59.209.225), 30 hops max, 38 byte packets
 1  10.64.0.1 (10.64.0.1)  31.416 ms  22.988 ms  33.543 ms
 2  vlan74.xs1a.man4.uk.m247.com (89.238.183.58)  24.190 ms  23.847 ms  23.517 ms
 3  eth-1-0.core-dc1-agg1.man4.uk.m247.com (77.243.185.82)  23.718 ms  22.940 ms  23.778 ms
 4  xe-1-1-1-0.core1.man4.uk.m247.com (77.243.185.66)  24.027 ms  24.940 ms  22.331 ms
 5  te-12-5-0.core-dc2.man4.uk.m247.com (83.97.21.151)  43.766 ms  te-9-3-0.bb1.lon1.uk.m247.com (212.103.51.17)  45.725 ms  te-13-4-0.core-dc2.man4.uk.m247.com (77.243.176.47)  41.174 ms
 6  xe-3-3-1-0.core1.lon2.uk.m247.com (193.9.115.143)  31.692 ms  xe-3-0-2-0.core1.lon2.uk.m247.com (77.243.185.215)  32.758 ms  xe-5-0-0-0.core1.lon2.uk.m247.com (176.10.82.22)  30.290 ms
 7  89.44.212.110 (89.44.212.110)  38.744 ms  xe-4-0-1-0.core1.fra2.de.m247.com (193.9.115.137)  46.178 ms  46.206 ms
 8  37.120.128.253 (37.120.128.253)  41.575 ms  40.368 ms  41.920 ms
 9  *  *  *
10  *  *  *
11  *  *  *
12  wiki-01.infra.openwrt.org (139.59.209.225)  42.859 ms  41.151 ms  39.960 ms
```

Perhaps it is related to the 192.168.0.0 range.

wackejohn, 2020-06-19:
@aaronjg 
So, I've changed the vpn address to 10.10.1.0/24, but it's still not working.
```
root@HOME-Router_HB:~# ip -4 addr ls wana
27: wana: <POINTOPOINT,NOARP,UP,LOWER_UP> mtu 1300 qdisc noqueue state UNKNOWN group default qlen 1000
    inet 10.10.1.3/24 brd 10.10.1.255 scope global wana
       valid_lft forever preferred_lft forever
root@HOME-Router_HB:~# ipset list mwan3_source_v4
Name: mwan3_source_v4
Type: hash:net
Revision: 6
Header: family inet hashsize 1024 maxelem 65536
Size in memory: 704
References: 1
Number of entries: 6
Members:
192.168.196.2
10.10.1.3
117.85.168.51
192.168.2.1
58.219.110.163
192.168.150.3
root@HOME-Router_HB:~# iptables --table mangle -S mwan3_hook
-N mwan3_hook
-A mwan3_hook -p icmp -m set --match-set mwan3_source_v4 src -m icmp --icmp-type 8 -j RETURN
-A mwan3_hook -j CONNMARK --restore-mark --nfmask 0xff00 --ctmask 0xff00
-A mwan3_hook -m mark --mark 0x0/0xff00 -j mwan3_ifaces_in
-A mwan3_hook -m mark --mark 0x0/0xff00 -j mwan3_connected
-A mwan3_hook -m mark --mark 0x0/0xff00 -j mwan3_rules
-A mwan3_hook -j CONNMARK --save-mark --nfmask 0xff00 --ctmask 0xff00
-A mwan3_hook -m mark ! --mark 0xff00/0xff00 -j mwan3_connected
root@HOME-Router_HB:~# /bin/ping 199.85.126.10
PING 199.85.126.10 (199.85.126.10): 56 data bytes
64 bytes from 199.85.126.10: seq=0 ttl=57 time=29.872 ms
64 bytes from 199.85.126.10: seq=1 ttl=57 time=29.876 ms
64 bytes from 199.85.126.10: seq=2 ttl=57 time=30.275 ms
64 bytes from 199.85.126.10: seq=3 ttl=57 time=29.513 ms
^C
--- 199.85.126.10 ping statistics ---
4 packets transmitted, 4 packets received, 0% packet loss
round-trip min/avg/max = 29.513/29.884/30.275 ms
root@HOME-Router_HB:~# /bin/ping -i wana 199.85.126.10
ping: invalid number 'wana'
root@HOME-Router_HB:~# /bin/ping -I wana 199.85.126.10
PING 199.85.126.10 (199.85.126.10): 56 data bytes
^C
--- 199.85.126.10 ping statistics ---
7 packets transmitted, 0 packets received, 100% packet loss
root@HOME-Router_HB:~# traceroute -i wana 199.85.126.10
traceroute to 199.85.126.10 (199.85.126.10), 30 hops max, 38 byte packets
 1  *  *  *
 2  *  *  *
 3  *  *  *
 4  *  *  *
 5  *  *  *
 6^C
```
My openwrt version:
```

BusyBox v1.31.1 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r13578-b7a8a54542
 -----------------------------------------------------
```
Kernel version:
```
Linux HOME-Router_HB 5.4.45 #0 SMP Mon Jun 15 19:16:18 2020 armv7l GNU/Linux
```

jamesmacwhite, 2020-06-19:
@wackejohn My only thoughts would be the fact your on the snapshot builds and the kernel branch is different 4.x vs 5.x. I'll happy provide any configuration of the Wireguard clients and network config I have setup to see if you find anything different that jumps out, but that's the only area I can see at that moment that could explain the difference in behaviour.

aaronjg, 2020-06-19:
If private/public address are handled differently, changing the ip address to 10.x.x.x likely won't affect anything.

After applying the patch, the firewall log looks essentially the same with and without mwan3 enabled, so it is surprising that it would work in one case and not the other.  Can you use tcpdump to figure out if the packet is actually going out? 

Another possibility is that mwan3 is trying to route the raw vpn traffic through the vpn, but it looks like you are only using the vpn for dns? Though one of the ips, didn't look like a dns server 178.79.131.110), so unless that is your wireguard endpoint, or you have additional mwan3 rules, this seems unlikely.

Other than that, perhaps trying a 4.x kernel as @jamesmacwhite suggests could fix it. If so, it would be good to know so that the issue can be fixed before mainline moves to 5.X.

wackejohn, 2020-06-20:
@aaronjg 
I've tried the latest stable release on my another router, and seemed that the issue didn't exist.
```
BusyBox v1.30.1 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 19.07.3, r11063-85e04e9f46
 -----------------------------------------------------
root@OpenWrt:~# uname -a
Linux OpenWrt 4.14.180 #0 Sat May 16 18:32:20 2020 mips GNU/Linux
root@OpenWrt:~# ping -I wana 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
64 bytes from 8.8.8.8: seq=0 ttl=121 time=137.162 ms
64 bytes from 8.8.8.8: seq=1 ttl=121 time=137.205 ms
64 bytes from 8.8.8.8: seq=2 ttl=121 time=139.210 ms
64 bytes from 8.8.8.8: seq=3 ttl=121 time=137.581 ms
64 bytes from 8.8.8.8: seq=4 ttl=121 time=137.172 ms
^C
--- 8.8.8.8 ping statistics ---
5 packets transmitted, 5 packets received, 0% packet loss
round-trip min/avg/max = 137.162/137.666/139.210 ms
```
Then I tried port the 5.x kernel from snapshot to stable release, the issue showed up.
```

BusyBox v1.30.1 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 19.07.3, r11063-85e04e9f46
 -----------------------------------------------------
root@OpenWrt:~# uname -a
Linux OpenWrt 5.4.46 #0 Sat May 16 18:32:20 2020 mips GNU/Linux
root@OpenWrt:~# ping -I wana 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
^C
--- 8.8.8.8 ping statistics ---
9 packets transmitted, 0 packets received, 100% packet loss
```
So, it's about the kernel, snapshot was on 4.19.x and now on 5.4.x, how to fix this issue? I want to use the 5.4.x kernel for my mvebu devices.

jamesmacwhite, 2020-06-20:
This is good to know. Potentially now it's an OpenWrt bug in core? Needs reporting here: https://bugs.openwrt.org/.

aaronjg, 2020-06-20:
@wackejohn Thanks for doing the diagnostics. 

With the mainline kernel, were you able to get things to work without my mwan3_source_v4 patch?

Hard to say exactly where the bug is. It could be an issue with the kernel, in which case you could do a git-bisect to try to locate which kernel commit broke everything.

It could also be an issue with mwan3 is working in the 4.14 kernel, but only because there was a bug in the kernel, so the proper solution could still be to patch mwan3.

Can't really say until we figure out the route cause.

Now that you have a working and broken implementation, it will be a bit easier to diagnose.

Can you use iptables logging and tcpdump to figure out exactly where the behaviors diverge?

- Are the firewall marks, interfaces and outgoing ip address the same at the end of postrouting on both routers?
- using tcpdump are the packets leaving the vpn interface on both routers?
- Is there wireguard traffic on the WAN interface in both cases?

Unfortunately, I don't have a router that I can run the snapshot build on, so I won't be able to reproduce.

wackejohn, 2020-06-24:
@aaronjg 
> Are the firewall marks, interfaces and outgoing ip address the same at the end of postrouting on both routers?

All the settings are the same on both routers.

>using tcpdump are the packets leaving the vpn interface on both routers?

Tcpdump on kernel 5.4.x, the icmp packtes can't leaving the vpn interface.
```
root@HOME-Router_CZ:~# /bin/ping -I wana 139.59.209.225
PING 139.59.209.225 (139.59.209.225): 56 data bytes
^C
--- 139.59.209.225 ping statistics ---
478 packets transmitted, 0 packets received, 100% packet loss
```
```
root@HOME-Router_CZ:~# tcpdump -i wana -nn host 139.59.209.225
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wana, link-type RAW (Raw IP), capture size 262144 bytes
19:36:13.309118 IP 10.10.1.2 > 139.59.209.225: ICMP echo request, id 65311, seq 27, length 64
19:36:14.309218 IP 10.10.1.2 > 139.59.209.225: ICMP echo request, id 65311, seq 28, length 64
19:36:15.309335 IP 10.10.1.2 > 139.59.209.225: ICMP echo request, id 65311, seq 29, length 64
19:36:16.309431 IP 10.10.1.2 > 139.59.209.225: ICMP echo request, id 65311, seq 30, length 64
19:36:17.309521 IP 10.10.1.2 > 139.59.209.225: ICMP echo request, id 65311, seq 31, length 64
19:36:18.309609 IP 10.10.1.2 > 139.59.209.225: ICMP echo request, id 65311, seq 32, length 64
19:36:19.309699 IP 10.10.1.2 > 139.59.209.225: ICMP echo request, id 65311, seq 33, length 64
19:36:20.309804 IP 10.10.1.2 > 139.59.209.225: ICMP echo request, id 65311, seq 34, length 64
19:36:21.309922 IP 10.10.1.2 > 139.59.209.225: ICMP echo request, id 65311, seq 35, length 64
19:36:22.310020 IP 10.10.1.2 > 139.59.209.225: ICMP echo request, id 65311, seq 36, length 64
19:36:23.310144 IP 10.10.1.2 > 139.59.209.225: ICMP echo request, id 65311, seq 37, length 64
19:36:24.310263 IP 10.10.1.2 > 139.59.209.225: ICMP echo request, id 65311, seq 38, length 64
19:36:25.310353 IP 10.10.1.2 > 139.59.209.225: ICMP echo request, id 65311, seq 39, length 64
19:36:26.310574 IP 10.10.1.2 > 139.59.209.225: ICMP echo request, id 65311, seq 40, length 64
19:36:27.310679 IP 10.10.1.2 > 139.59.209.225: ICMP echo request, id 65311, seq 41, length 64
19:36:28.310781 IP 10.10.1.2 > 139.59.209.225: ICMP echo request, id 65311, seq 42, length 64
19:36:29.310899 IP 10.10.1.2 > 139.59.209.225: ICMP echo request, id 65311, seq 43, length 64
19:36:30.311003 IP 10.10.1.2 > 139.59.209.225: ICMP echo request, id 65311, seq 44, length 64
19:36:31.311097 IP 10.10.1.2 > 139.59.209.225: ICMP echo request, id 65311, seq 45, length 64
19:36:32.311194 IP 10.10.1.2 > 139.59.209.225: ICMP echo request, id 65311, seq 46, length 64
19:36:33.311280 IP 10.10.1.2 > 139.59.209.225: ICMP echo request, id 65311, seq 47, length 64
19:36:34.311369 IP 10.10.1.2 > 139.59.209.225: ICMP echo request, id 65311, seq 48, length 64
```
```
root@VPS-Server:~# tcpdump -i wg0 -nn host 139.59.209.225
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wg0, link-type RAW (Raw IP), capture size 262144 bytes
^C
0 packets captured
0 packets received by filter
0 packets dropped by kernel
```
>Hard to say exactly where the bug is. It could be an issue with the kernel, in which case you could do a git-bisect to try to locate which kernel commit broke everything.

The snapshot kernel version was from 4.14 directly to 4.19, so it will be a huge work.

aaronjg, 2020-06-24:
yeah, git bisect would probably take several kernel flashes, so not the easiest way to diagnose. Let's keep exploring other possibilities.

It looks like the ping packets are being handled properly to go out of the VPN interface, but the VPN traffic is not being routed properly.

Can you add similar iptables logging to track the wireguard packets, and share the outptu? In particular, I want to see if they are getting different firewall marks with the 5.X kernel vs. the 4.X kernel.  You'll have to do some other rule to filter them, perhaps based on destination address or port.

Is the ip address of your wireguard server in the `mwan3_connected_v4` ipset? (you can check either in the output of `ipset list mwan3_connected_v4` or `ipset test mwan3_connected_v4 $SERVER_IP`

With the mainline kernel, were you able to get things to work without my mwan3_source_v4 patch?

wackejohn, 2020-06-27:
@aaronjg 
>Can you add similar iptables logging to track the wireguard packets, and share the outptu?

The output (strange thing was that the fist ping succeeded):
```
root@OpenWrt:~# mwan3 interfaces
Interface status:
 interface wan is online and tracking is active
 interface wan_6 is online and tracking is active
 interface wana is online and tracking is not enabled

root@OpenWrt:~# ping -I wana -c 1 openwrt.org
ping: connect: Permission denied
root@OpenWrt:~# /bin/ping -I wana -c 1 openwrt.org
PING openwrt.org (139.59.209.225): 56 data bytes
64 bytes from 139.59.209.225: seq=0 ttl=52 time=545.257 ms

--- openwrt.org ping statistics ---
1 packets transmitted, 1 packets received, 0% packet loss
round-trip min/avg/max = 545.257/545.257/545.257 ms
root@OpenWrt:~# /bin/ping -I wana -c 1 openwrt.org
PING openwrt.org (139.59.209.225): 56 data bytes

--- openwrt.org ping statistics ---
1 packets transmitted, 0 packets received, 100% packet loss

root@OpenWrt:~# logread |grep "postroute start"
Sat Jun 27 23:36:52 2020 kern.warn kernel: [  253.142138] postroute startIN= OUT=pppoe-wan SRC=180.116.118.253 DST=173.242.119.141 LEN=176 TOS=0x08 PREC=0x80 TTL=64 ID=1880 PROTO=UDP SPT=36686 DPT=51888 LEN=156 MARK=0x100
Sat Jun 27 23:36:52 2020 kern.warn kernel: [  253.342776] postroute startIN= OUT=pppoe-wan SRC=180.116.118.253 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=1889 PROTO=UDP SPT=36686 DPT=51888 LEN=136 MARK=0x100
Sat Jun 27 23:36:56 2020 kern.warn kernel: [  257.613739] postroute startIN= OUT=wana SRC=180.116.118.253 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=2153 PROTO=UDP SPT=36686 DPT=51888 LEN=136 MARK=0x100
Sat Jun 27 23:37:08 2020 kern.warn kernel: [  268.944923] postroute startIN= OUT=pppoe-wan SRC=180.116.118.253 DST=173.242.119.141 LEN=204 TOS=0x00 PREC=0xC0 TTL=64 ID=4435 PROTO=ICMP TYPE=3 CODE=3 [SRC=173.242.119.141 DST=180.116.118.253 LEN=176 TOS=0x00 PREC=0x00 TTL=54 ID=3942 PROTO=UDP SPT=51888 DPT=36686 LEN=156 ] MARK=0x100
Sat Jun 27 23:37:12 2020 kern.warn kernel: [  273.124283] postroute startIN= OUT=pppoe-wan SRC=180.116.118.253 DST=173.242.119.141 LEN=176 TOS=0x08 PREC=0x80 TTL=64 ID=3389 PROTO=UDP SPT=36686 DPT=51888 LEN=156 MARK=0x100
Sat Jun 27 23:37:12 2020 kern.warn kernel: [  273.321857] postroute startIN= OUT=pppoe-wan SRC=180.116.118.253 DST=173.242.119.141 LEN=60 TOS=0x00 PREC=0x00 TTL=64 ID=3406 PROTO=UDP SPT=36686 DPT=51888 LEN=40 MARK=0x100
root@OpenWrt:~# logread |grep "postroute end"
Sat Jun 27 23:36:52 2020 kern.warn kernel: [  253.158003] postroute endIN= OUT=pppoe-wan SRC=180.116.118.253 DST=173.242.119.141 LEN=176 TOS=0x08 PREC=0x80 TTL=64 ID=1880 PROTO=UDP SPT=36686 DPT=51888 LEN=156 MARK=0x100
Sat Jun 27 23:36:52 2020 kern.warn kernel: [  253.358639] postroute endIN= OUT=pppoe-wan SRC=180.116.118.253 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=1889 PROTO=UDP SPT=36686 DPT=51888 LEN=136 MARK=0x100
Sat Jun 27 23:36:56 2020 kern.warn kernel: [  257.629166] postroute endIN= OUT=wana SRC=180.116.118.253 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=2153 PROTO=UDP SPT=36686 DPT=51888 LEN=136 MARK=0x100
Sat Jun 27 23:37:08 2020 kern.warn kernel: [  268.970377] postroute endIN= OUT=pppoe-wan SRC=180.116.118.253 DST=173.242.119.141 LEN=204 TOS=0x00 PREC=0xC0 TTL=64 ID=4435 PROTO=ICMP TYPE=3 CODE=3 [SRC=173.242.119.141 DST=180.116.118.253 LEN=176 TOS=0x00 PREC=0x00 TTL=54 ID=3942 PROTO=UDP SPT=51888 DPT=36686 LEN=156 ] MARK=0x100
Sat Jun 27 23:37:12 2020 kern.warn kernel: [  273.140148] postroute endIN= OUT=pppoe-wan SRC=180.116.118.253 DST=173.242.119.141 LEN=176 TOS=0x08 PREC=0x80 TTL=64 ID=3389 PROTO=UDP SPT=36686 DPT=51888 LEN=156 MARK=0x100
Sat Jun 27 23:37:12 2020 kern.warn kernel: [  273.337545] postroute endIN= OUT=pppoe-wan SRC=180.116.118.253 DST=173.242.119.141 LEN=60 TOS=0x00 PREC=0x00 TTL=64 ID=3406 PROTO=UDP SPT=36686 DPT=51888 LEN=40 MARK=0x100
root@OpenWrt:~# logread |grep "output start"
Sat Jun 27 23:36:52 2020 kern.warn kernel: [  253.111960] output startIN= OUT=pppoe-wan SRC=180.116.118.253 DST=173.242.119.141 LEN=176 TOS=0x08 PREC=0x80 TTL=64 ID=1880 PROTO=UDP SPT=36686 DPT=51888 LEN=156
Sat Jun 27 23:36:52 2020 kern.warn kernel: [  253.312617] output startIN= OUT=pppoe-wan SRC=180.116.118.253 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=1889 PROTO=UDP SPT=36686 DPT=51888 LEN=136
Sat Jun 27 23:36:56 2020 kern.warn kernel: [  257.583575] output startIN= OUT=pppoe-wan SRC=180.116.118.253 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=2153 PROTO=UDP SPT=36686 DPT=51888 LEN=136
Sat Jun 27 23:37:08 2020 kern.warn kernel: [  268.895579] output startIN= OUT=pppoe-wan SRC=180.116.118.253 DST=173.242.119.141 LEN=204 TOS=0x00 PREC=0xC0 TTL=64 ID=4435 PROTO=ICMP TYPE=3 CODE=3 [SRC=173.242.119.141 DST=180.116.118.253 LEN=176 TOS=0x00 PREC=0x00 TTL=54 ID=3942 PROTO=UDP SPT=51888 DPT=36686 LEN=156 ]
Sat Jun 27 23:37:12 2020 kern.warn kernel: [  273.094108] output startIN= OUT=pppoe-wan SRC=180.116.118.253 DST=173.242.119.141 LEN=176 TOS=0x08 PREC=0x80 TTL=64 ID=3389 PROTO=UDP SPT=36686 DPT=51888 LEN=156
Sat Jun 27 23:37:12 2020 kern.warn kernel: [  273.292050] output startIN= OUT=pppoe-wan SRC=180.116.118.253 DST=173.242.119.141 LEN=60 TOS=0x00 PREC=0x00 TTL=64 ID=3406 PROTO=UDP SPT=36686 DPT=51888 LEN=40
root@OpenWrt:~# logread |grep "output end"
[... cut ...]
[... 83 lines cut ...]
Revision: 6
Header: family inet hashsize 1024 maxelem 65536
Size in memory: 1208
References: 1
Number of entries: 15
Members:
180.116.118.253
127.255.255.255
192.168.1.0/24
180.116.118.1
10.10.1.0/24
127.0.0.1
10.10.1.0
10.10.1.5
192.168.1.255
192.168.1.0
192.168.1.1
224.0.0.0/3
127.0.0.0/8
127.0.0.0
10.10.1.255
```
>With the mainline kernel, were you able to get things to work without my mwan3_source_v4 patch?

Acctully, kernel above 4.19, the `ping -I ` won't work with or without your patch.

aaronjg, 2020-06-28:
What do you get for:
`ip route list table 1`  this is how it is trying to route out the traffic to your firewall server.

Normally, the vpn server would have a static route and be in the connected ipset list, and ignored. Earlier, you said that route was deleted. Why?

> Acctully, kernel above 4.19, the ping -I won't work with or without your patch.

Yes, but what about the earlier kernels, do you need my patch to make that work?

wackejohn, 2020-06-29:
>Normally, the vpn server would have a static route and be in the connected ipset list, and ignored. Earlier, you said that route was deleted. Why?

Cause I have multi ISPs for internet connection, and I want to use policy to control the wireguard connection with mwan3, so I delete the static route (snapshot version of wireguard have an option "nohostroute" do the same thing).

>Yes, but what about the earlier kernels, do you need my patch to make that work?

On earlier kernels (4.14 and below ), the `ping -I ` is working and no need your patch.

aaronjg, 2020-07-03:
Can you describe exactly what you are trying to do and post your complete mwan3 configuration and the output of `ip route list #` for every '#` in  `ip rule list` (I think 1,3,5,7 and 9 from your previous post).

Mwan3 needs that route to include it in the mwan3_connected set so that it knows to let the traffic go to the VPN server. Otherwise, it may end up trying to route the traffic that should be _to_ the VPN server _through_ the VPN server, which won't work.

If you post your configuration and what it is exactly you are trying to accomplish, there may be a way to still do that with additional firewall rules. However, this is a bit outside the box of the 'normal' mwan3 setup.

wackejohn, 2020-07-04:
@aaronjg 
>Can you describe exactly what you are trying to do 

I'm trying to bypass the ISP firewall through wireguard vpn, including DNS and Routing, and then use mwan3 to achieve the failover policy for two wireguard vpn connections.

My mwan3 config (will be very complicated):
```
config rule 'AntiGFW_DNS'
        option proto 'all'
        option family 'ipv4'
        option dest_ip '8.8.8.8,8.8.4.4,208.67.222.222,208.67.220.220,199.91.73.222,178.79.131.110'
        option sticky '1'
        option timeout '600'
        option use_policy 'wana_wanb'

config rule 'ChinaNet_DNS'
        option dest_ip '218.4.4.4,218.2.2.2,114.114.114.114,223.5.5.5'
        option family 'ipv4'
        option proto 'all'
        option sticky '1'
        option timeout '600'
        option use_policy 'wan1_wan2'

config rule 'CM_DNS'
        option dest_ip '112.4.0.55,221.131.143.69,114.114.115.115,223.6.6.6'
        option family 'ipv4'
        option proto 'all'
        option sticky '1'
        option timeout '600'
        option use_policy 'wan3_wan4'

config rule 'Clock_VPS'
        option family 'ipv4'
        option proto 'all'
        option dest_ip '119.23.32.83,31.40.214.127,173.242.119.141'
        option use_policy 'wan1_2_default'
        option sticky '0'

config rule 'IPV4_PTlist'
        option proto 'all'
        option family 'ipv4'
        option ipset 'ptlist'
        option sticky '0'
        option use_policy 'wan3_wan4'

config rule 'IPV4_GFW'
        option proto 'all'
        option family 'ipv4'
        option ipset 'gfwlist'
        option timeout '300'
        option sticky '0'
        option use_policy 'wana_wanb'

config rule 'CM_IPV4'
        option proto 'all'
        option family 'ipv4'
        option ipset 'cmcc'
        option timeout '600'
        option sticky '1'
        option use_policy 'wan3_wan4'

config rule 'ChinaNet_IPV4'
        option proto 'all'
        option family 'ipv4'
        option timeout '600'
        option sticky '1'
        option use_policy 'wan1_wan2'
        option ipset 'cnnocmcc'

config rule 'IPV4_WEB'
[... 666 lines cut ...]
1003:   from all iif pppoe-wan2 lookup 3
1005:   from all iif pppoe-wan3 lookup 5
1007:   from all iif pppoe-wan4 lookup 7
1009:   from all iif wana lookup 9
1011:   from all iif wanb lookup 11
2001:   from all fwmark 0x100/0x3f00 lookup 1
2003:   from all fwmark 0x300/0x3f00 lookup 3
2005:   from all fwmark 0x500/0x3f00 lookup 5
2007:   from all fwmark 0x700/0x3f00 lookup 7
2009:   from all fwmark 0x900/0x3f00 lookup 9
2011:   from all fwmark 0xb00/0x3f00 lookup 11
2061:   from all fwmark 0x3d00/0x3f00 blackhole
2062:   from all fwmark 0x3e00/0x3f00 unreachable
32766:  from all lookup main
32767:  from all lookup default
```
As above, now I keep the static route of wireguard, but the `ping -I` is also not working, so it's not about the static route thing.
```
root@HOME-Router_CZ:~# ping -I wana 8.8.8.8
PING 8.8.8.8 (8.8.8.8) from 10.10.1.2 wana: 56(84) bytes of data.
^C
--- 8.8.8.8 ping statistics ---
5 packets transmitted, 0 received, 100% packet loss, time 4192ms

```

aaronjg, 2020-07-05:
Can you post the `ip route list table X` for each of the routing tables?  When you keep the static route, does the VPN ip address show up in `ipset list mwan3_source_v4` ?

wackejohn, 2020-07-06:
@aaronjg 
output:
```
root@HOME-Router_CZ:~# ip route list table 1
default via 180.115.231.1 dev pppoe-wan1 metric 10 
10.10.1.0/24 dev wana proto static scope link metric 30 
31.40.214.127 via 180.115.231.1 dev pppoe-wan1 proto static metric 10 
112.1.96.1 dev pppoe-wan3 proto kernel scope link src 112.1.96.195 
112.3.84.1 dev pppoe-wan4 proto kernel scope link src 112.3.87.193 
114.228.153.1 dev pppoe-wan2 proto kernel scope link src 114.228.153.129 
173.242.119.141 via 180.115.231.1 dev pppoe-wan1 proto static metric 10 
180.115.231.1 dev pppoe-wan1 proto kernel scope link src 180.115.231.180 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 
192.168.2.0/24 dev vpn1 proto static scope link metric 110 
192.168.3.0/24 dev vpn1 proto static scope link metric 110 
192.168.150.0/24 dev wanb proto static scope link metric 40 
192.168.195.0/24 dev vpn0 proto static scope link metric 100 
192.168.195.2 dev vpn0 proto static scope link metric 100 
192.168.195.3 dev vpn0 proto static scope link metric 100 
192.168.195.4 dev vpn0 proto static scope link metric 100 
192.168.195.5 dev vpn0 proto static scope link metric 100 
192.168.195.6 dev vpn0 proto static scope link metric 100 
192.168.196.0/24 dev vpn1 proto static scope link metric 110 
192.168.196.2 dev vpn1 proto static scope link metric 110 
192.168.196.3 dev vpn1 proto static scope link metric 110 
root@HOME-Router_CZ:~# ip route list table 3
default via 114.228.153.1 dev pppoe-wan2 metric 15 
10.10.1.0/24 dev wana proto static scope link metric 30 
31.40.214.127 via 180.115.231.1 dev pppoe-wan1 proto static metric 10 
112.1.96.1 dev pppoe-wan3 proto kernel scope link src 112.1.96.195 
112.3.84.1 dev pppoe-wan4 proto kernel scope link src 112.3.87.193 
114.228.153.1 dev pppoe-wan2 proto kernel scope link src 114.228.153.129 
173.242.119.141 via 180.115.231.1 dev pppoe-wan1 proto static metric 10 
180.115.231.1 dev pppoe-wan1 proto kernel scope link src 180.115.231.180 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 
192.168.2.0/24 dev vpn1 proto static scope link metric 110 
192.168.3.0/24 dev vpn1 proto static scope link metric 110 
192.168.150.0/24 dev wanb proto static scope link metric 40 
192.168.195.0/24 dev vpn0 proto static scope link metric 100 
192.168.195.2 dev vpn0 proto static scope link metric 100 
192.168.195.3 dev vpn0 proto static scope link metric 100 
192.168.195.4 dev vpn0 proto static scope link metric 100 
192.168.195.5 dev vpn0 proto static scope link metric 100 
192.168.195.6 dev vpn0 proto static scope link metric 100 
192.168.196.0/24 dev vpn1 proto static scope link metric 110 
192.168.196.2 dev vpn1 proto static scope link metric 110 
192.168.196.3 dev vpn1 proto static scope link metric 110 
root@HOME-Router_CZ:~# ip route list table 5
default via 112.1.96.1 dev pppoe-wan3 metric 20 
10.10.1.0/24 dev wana proto static scope link metric 30 
31.40.214.127 via 180.115.231.1 dev pppoe-wan1 proto static metric 10 
112.1.96.1 dev pppoe-wan3 proto kernel scope link src 112.1.96.195 
112.3.84.1 dev pppoe-wan4 proto kernel scope link src 112.3.87.193 
114.228.153.1 dev pppoe-wan2 proto kernel scope link src 114.228.153.129 
173.242.119.141 via 180.115.231.1 dev pppoe-wan1 proto static metric 10 
180.115.231.1 dev pppoe-wan1 proto kernel scope link src 180.115.231.180 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 
192.168.2.0/24 dev vpn1 proto static scope link metric 110 
192.168.3.0/24 dev vpn1 proto static scope link metric 110 
192.168.150.0/24 dev wanb proto static scope link metric 40 
192.168.195.0/24 dev vpn0 proto static scope link metric 100 
192.168.195.2 dev vpn0 proto static scope link metric 100 
192.168.195.3 dev vpn0 proto static scope link metric 100 
192.168.195.4 dev vpn0 proto static scope link metric 100 
192.168.195.5 dev vpn0 proto static scope link metric 100 
192.168.195.6 dev vpn0 proto static scope link metric 100 
192.168.196.0/24 dev vpn1 proto static scope link metric 110 
192.168.196.2 dev vpn1 proto static scope link metric 110 
192.168.196.3 dev vpn1 proto static scope link metric 110 
root@HOME-Router_CZ:~# ip route list table 7
[... 74 lines cut ...]
default via 180.115.231.1 dev pppoe-wan1 proto static metric 10 
default via 114.228.153.1 dev pppoe-wan2 proto static metric 15 
default via 112.1.96.1 dev pppoe-wan3 proto static metric 20 
default via 112.3.84.1 dev pppoe-wan4 proto static metric 25 
10.10.1.0/24 dev wana proto static scope link metric 30 
31.40.214.127 via 180.115.231.1 dev pppoe-wan1 proto static metric 10 
112.1.96.1 dev pppoe-wan3 proto kernel scope link src 112.1.96.195 
112.3.84.1 dev pppoe-wan4 proto kernel scope link src 112.3.87.193 
114.228.153.1 dev pppoe-wan2 proto kernel scope link src 114.228.153.129 
173.242.119.141 via 180.115.231.1 dev pppoe-wan1 proto static metric 10 
180.115.231.1 dev pppoe-wan1 proto kernel scope link src 180.115.231.180 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 
192.168.2.0/24 dev vpn1 proto static scope link metric 110 
192.168.3.0/24 dev vpn1 proto static scope link metric 110 
192.168.150.0/24 dev wanb proto static scope link metric 40 
192.168.195.0/24 dev vpn0 proto static scope link metric 100 
192.168.195.2 dev vpn0 proto static scope link metric 100 
192.168.195.3 dev vpn0 proto static scope link metric 100 
192.168.195.4 dev vpn0 proto static scope link metric 100 
192.168.195.5 dev vpn0 proto static scope link metric 100 
192.168.195.6 dev vpn0 proto static scope link metric 100 
192.168.196.0/24 dev vpn1 proto static scope link metric 110 
192.168.196.2 dev vpn1 proto static scope link metric 110 
192.168.196.3 dev vpn1 proto static scope link metric 110 
```

aaronjg, 2020-07-06:
>  Seems that there is no mwan3_source_v4?

My mistake - I meant to ask about `mwan3_connected_v4`

aaronjg, 2020-07-06:
Also, could you rerun the diagnostics from this comment with the static route added back in?

https://github.com/openwrt/packages/issues/10712#issuecomment-650578269

There are a couple of weird things one:
- This is the weird thing that is going on: the packet starts out being sent to the vpn correctly from `ppoe-wan` but after the mwan3 rule is applied, it is trying to come out of `wana`, which obviously won't work.   In a more typical mwan3 setup, these packets would be unmarked, and go through the default routing table - however, it seems like your setup should work too.

- In the iptables log, you have packets coming out of `ppoe-wan` but I don't see any entries for `ppoe-wan` in the routing tables, just `ppoe-wan1` through `ppoe-wan4`

This is the weird iptables behavior:
```
Sat Jun 27 23:36:56 2020 kern.warn kernel: [  257.583575] output startIN= OUT=pppoe-wan SRC=180.116.118.253 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=2153 PROTO=UDP SPT=36686 DPT=51888 LEN=136

Sat Jun 27 23:36:56 2020 kern.warn kernel: [  257.598261] output endIN= OUT=pppoe-wan SRC=180.116.118.253 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=2153 PROTO=UDP SPT=36686 DPT=51888 LEN=136 MARK=0x100

Sat Jun 27 23:36:56 2020 kern.warn kernel: [  257.613739] postroute startIN= OUT=wana SRC=180.116.118.253 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=2153 PROTO=UDP SPT=36686 DPT=51888 LEN=136 MARK=0x100

Sat Jun 27 23:36:56 2020 kern.warn kernel: [  257.629166] postroute endIN= OUT=wana SRC=180.116.118.253 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=2153 PROTO=UDP SPT=36686 DPT=51888 LEN=136 MARK=0x100
```

wackejohn, 2020-07-07:
@aaronjg 
```
root@HOME-Router_CZ:~# ipset list mwan3_connected_v4
Name: mwan3_connected_v4
Type: hash:net
Revision: 6
Header: family inet hashsize 1024 maxelem 65536
Size in memory: 2832
References: 1
Number of entries: 44
Members:
192.168.1.0/24
173.242.119.141
192.168.150.2
127.0.0.1
114.228.153.129
192.168.195.0/24
112.3.84.1
192.168.195.2
192.168.195.0
10.10.1.255
192.168.1.1
192.168.3.0/24
192.168.150.255
10.10.1.0/24
192.168.195.3
114.228.153.1
192.168.196.3
180.115.231.180
192.168.150.0
192.168.195.255
192.168.195.6
112.3.87.193
127.0.0.0/8
192.168.1.255
10.10.1.0
127.255.255.255
31.40.214.127
112.1.96.1
180.115.231.1
192.168.1.0
192.168.195.1
192.168.196.1
192.168.195.4
192.168.196.0/24
192.168.196.2
192.168.196.0
10.10.1.2
192.168.195.5
127.0.0.0
224.0.0.0/3
192.168.150.0/24
112.1.96.195
192.168.196.255
192.168.2.0/24
```
As above, the vpn server ip was in the `mwan3_connected_v4`

>Also, could you rerun the diagnostics from this comment with the static route added back in?

Actually, it was another router with minimal configuration for test, please ignore that output.

aaronjg, 2020-07-07:
>  Actually, it was another router with minimal configuration for test, please ignore that output.

I see. It will certainly be easier to debug with the minimal configuration.  Can you post the full mwan3 config from the minimal configuration, the routing tables as above from the minimal configuration setup, and rerun the iptables logging now that the static route is in place? 

I'm trying to see why the VPN packets are getting a firewall mark of 0x100 - normally these should get marked 0x3f00 and then not match any further rules.

wackejohn, 2020-07-11:
@aaronjg 
The minimal mwan3 config:
```
config interface 'wan'
        option enabled '1'
        option family 'ipv4'
        option reliability '2'
        option count '1'
        option timeout '2'
        option interval '5'
        option down '3'
        option up '8'
        option initial_state 'online'
        list track_ip '114.114.114.114'
        list track_ip '114.114.115.115'
        option track_method 'ping'
        option size '56'
        option max_ttl '60'
        option check_quality '0'
        option failure_interval '5'
        option recovery_interval '5'

config interface 'wan_6'
        option enabled '1'
        list track_ip '240e:5a::6666'
        list track_ip '240e:5b::6666'
        option family 'ipv6'
        option reliability '2'
        option count '1'
        option timeout '2'
        option interval '5'
        option down '3'
        option up '8'

config interface 'wana'
        option family 'ipv4'
        option reliability '1'
        option count '1'
        option timeout '2'
        option interval '5'
        option down '3'
        option up '8'
        option enabled '1'
        option initial_state 'online'
        option track_method 'ping'
        option size '56'
        option max_ttl '60'
        option check_quality '0'
        option failure_interval '5'
        option recovery_interval '5'

config member 'wan_m1_w1'
        option interface 'wan'
        option metric '1'
        option weight '1'

config member 'wana_m1_w1'
        option interface 'wana'
        option metric '1'
        option weight '1'

config member 'wan_6_m1_w1'
        option interface 'wan_6'
        option metric '1'
        option weight '1'

config policy 'wan_only'
        list use_member 'wan_m1_w1'
        list use_member 'wan_6_m1_w1'

config policy 'wana_only'
        list use_member 'wana_m1_w1'

config policy 'balanced'
        list use_member 'wan_m1_w1'

config rule 'https'
        option sticky '1'
        option dest_port '443'
        option proto 'tcp'
        option use_policy 'balanced'

config rule 'default_rule'
        option dest_ip '0.0.0.0/0'
        option use_policy 'balanced'
```

Routing table:
```
root@OpenWrt:~# ip route show
default via 117.85.168.1 dev pppoe-wan proto static 
10.10.1.0/24 dev wana proto static scope link metric 10 
117.85.168.1 dev pppoe-wan proto kernel scope link src 117.85.170.4 
173.242.119.141 via 117.85.168.1 dev pppoe-wan proto static 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 
root@OpenWrt:~# ip route list table 1
default via 117.85.168.1 dev pppoe-wan 
10.10.1.0/24 dev wana proto static scope link metric 10 
117.85.168.1 dev pppoe-wan proto kernel scope link src 117.85.170.4 
173.242.119.141 via 117.85.168.1 dev pppoe-wan proto static 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 
root@OpenWrt:~# ip route list table 3
default dev wana scope link metric 10 
10.10.1.0/24 dev wana proto static scope link metric 10 
117.85.168.1 dev pppoe-wan proto kernel scope link src 117.85.170.4 
173.242.119.141 via 117.85.168.1 dev pppoe-wan proto static 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 
```

And the iptable log:
```

[  294.718366] output startIN= OUT=pppoe-wan SRC=117.85.170.4 DST=173.242.119.141 LEN=176 TOS=0x08 PREC=0x80 TTL=64 ID=50646 PROTO=UDP SPT=57303 DPT=51888 LEN=156 
[  294.732883] output endIN= OUT=pppoe-wan SRC=117.85.170.4 DST=173.242.119.141 LEN=176 TOS=0x08 PREC=0x80 TTL=64 ID=50646 PROTO=UDP SPT=57303 DPT=51888 LEN=156 MARK=0x3f00 
[  294.748276] postroute startIN= OUT=pppoe-wan SRC=117.85.170.4 DST=173.242.119.141 LEN=176 TOS=0x08 PREC=0x80 TTL=64 ID=50646 PROTO=UDP SPT=57303 DPT=51888 LEN=156 MARK=0x3f00 
[  294.764055] postroute endIN= OUT=pppoe-wan SRC=117.85.170.4 DST=173.242.119.141 LEN=176 TOS=0x08 PREC=0x80 TTL=64 ID=50646 PROTO=UDP SPT=57303 DPT=51888 LEN=156 MARK=0x3f00 
[  294.919942] output startIN= OUT=pppoe-wan SRC=117.85.170.4 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=50661 PROTO=UDP SPT=57303 DPT=51888 LEN=136 
[  294.934433] output endIN= OUT=pppoe-wan SRC=117.85.170.4 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=50661 PROTO=UDP SPT=57303 DPT=51888 LEN=136 MARK=0x3f00 
```

Seemed that with the static route in place, the firewall mark was `0x3f00` .

aaronjg, 2020-07-11:
Okay, so with the static route in place, this is starting to look like a more 'normal' mwan3 config, so it will be easier to figure out teh root cause, rather than get distracted by anamolies of your more complex config.

Is that all of the relevant output from the iptables log? something looks very strange - the first packet appears to be routed correctly out of ppoe-wan, but the second one appears to go through the output routing table, but never make it to the postrouting table. Yet, as far as I can tell, aside from the length, the packets are identical. Very strange. 

Without mwan3, what do the packets look like going through the firewall?  It also is interesting that the UDP packet length of the first packet is the IP packet length of the second. If the iptable logs look different, could you share the headers of these two UDP packets and the corresponding packets without mwan3?

Also, now that you have the minimal config with the static route, can you check with tcpdump if packets are going through the ppoe interface and the wireguard interface, as you did here: https://github.com/openwrt/packages/issues/10712#issuecomment-648792167

wackejohn, 2020-07-12:
@aaronjg 
Mwan3 was running:
```
root@OpenWrt:~# iptables --table mangle -I POSTROUTING 1 -d 173.242.119.141 -j LOG --log-prefix "postroute start"
root@OpenWrt:~# iptables --table mangle -A POSTROUTING  -d 173.242.119.141 -j LOG --log-prefix "postroute end"
root@OpenWrt:~# iptables --table mangle -I OUTPUT 1 -d 173.242.119.141 -j LOG --log-prefix "output start"
root@OpenWrt:~# iptables --table mangle -A OUTPUT  -d 173.242.119.141 -j LOG --log-prefix "output end"
root@OpenWrt:~# ping -I wana -c 5 139.59.209.225
PING 139.59.209.225 (139.59.209.225): 56 data bytes
64 bytes from 139.59.209.225: seq=0 ttl=52 time=556.206 ms

--- 139.59.209.225 ping statistics ---
5 packets transmitted, 1 packets received, 80% packet loss
round-trip min/avg/max = 556.206/556.206/556.206 ms

[  668.808017] output startIN= OUT=pppoe-wan SRC=180.112.126.158 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=42894 PROTO=UDP SPT=52640 DPT=51888 LEN=136 
[  668.822786] output endIN= OUT=pppoe-wan SRC=180.112.126.158 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=42894 PROTO=UDP SPT=52640 DPT=51888 LEN=136 
[  668.837364] postroute startIN= OUT=pppoe-wan SRC=180.112.126.158 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=42894 PROTO=UDP SPT=52640 DPT=51888 LEN=136 
[  668.852348] postroute endIN= OUT=pppoe-wan SRC=180.112.126.158 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=42894 PROTO=UDP SPT=52640 DPT=51888 LEN=136 
[  669.872475] output startIN= OUT=pppoe-wan SRC=180.112.126.158 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=42936 PROTO=UDP SPT=52640 DPT=51888 LEN=136 
[  669.887223] output endIN= OUT=pppoe-wan SRC=180.112.126.158 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=42936 PROTO=UDP SPT=52640 DPT=51888 LEN=136 
[  669.901799] postroute startIN= OUT=pppoe-wan SRC=180.112.126.158 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=42936 PROTO=UDP SPT=52640 DPT=51888 LEN=136 
[  669.916781] postroute endIN= OUT=pppoe-wan SRC=180.112.126.158 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=42936 PROTO=UDP SPT=52640 DPT=51888 LEN=136 
[  670.873132] output startIN= OUT=pppoe-wan SRC=180.112.126.158 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=42961 PROTO=UDP SPT=52640 DPT=51888 LEN=136 
[  670.887883] output endIN= OUT=pppoe-wan SRC=180.112.126.158 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=42961 PROTO=UDP SPT=52640 DPT=51888 LEN=136 
[  670.902462] postroute startIN= OUT=pppoe-wan SRC=180.112.126.158 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=42961 PROTO=UDP SPT=52640 DPT=51888 LEN=136 
[  670.917445] postroute endIN= OUT=pppoe-wan SRC=180.112.126.158 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=42961 PROTO=UDP SPT=52640 DPT=51888 LEN=136 
[  671.873635] output startIN= OUT=pppoe-wan SRC=180.112.126.158 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=43059 PROTO=UDP SPT=52640 DPT=51888 LEN=136 
[  671.888386] output endIN= OUT=pppoe-wan SRC=180.112.126.158 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=43059 PROTO=UDP SPT=52640 DPT=51888 LEN=136 
[  671.902967] postroute startIN= OUT=pppoe-wan SRC=180.112.126.158 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=43059 PROTO=UDP SPT=52640 DPT=51888 LEN=136 
[  671.917950] postroute endIN= OUT=pppoe-wan SRC=180.112.126.158 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=43059 PROTO=UDP SPT=52640 DPT=51888 LEN=136 
[  672.874291] output startIN= OUT=pppoe-wan SRC=180.112.126.158 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=43104 PROTO=UDP SPT=52640 DPT=51888 LEN=136 
[  672.889043] output endIN= OUT=pppoe-wan SRC=180.112.126.158 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=43104 PROTO=UDP SPT=52640 DPT=51888 LEN=136 
[  672.903621] postroute startIN= OUT=pppoe-wan SRC=180.112.126.158 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=43104 PROTO=UDP SPT=52640 DPT=51888 LEN=136 
[  672.918603] postroute endIN= OUT=pppoe-wan SRC=180.112.126.158 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=43104 PROTO=UDP SPT=52640 DPT=51888 LEN=136 
[  683.322226] output startIN= OUT=pppoe-wan SRC=180.112.126.158 DST=173.242.119.141 LEN=60 TOS=0x00 PREC=0x00 TTL=64 ID=43606 PROTO=UDP SPT=52640 DPT=51888 LEN=40 
[  683.336797] output endIN= OUT=pppoe-wan SRC=180.112.126.158 DST=173.242.119.141 LEN=60 TOS=0x00 PREC=0x00 TTL=64 ID=43606 PROTO=UDP SPT=52640 DPT=51888 LEN=40 
[  683.351198] postroute startIN= OUT=pppoe-wan SRC=180.112.126.158 DST=173.242.119.141 LEN=60 TOS=0x00 PREC=0x00 TTL=64 ID=43606 PROTO=UDP SPT=52640 DPT=51888 LEN=40 
[  683.366005] postroute endIN= OUT=pppoe-wan SRC=180.112.126.158 DST=173.242.119.141 LEN=60 TOS=0x00 PREC=0x00 TTL=64 ID=43606 PROTO=UDP SPT=52640 DPT=51888 LEN=40 

root@OpenWrt:~# tcpdump -i wana -nn host 139.59.209.225 -vv -X
tcpdump: listening on wana, link-type RAW (Raw IP), capture size 262144 bytes
01:30:25.659203 IP (tos 0x0, ttl 64, id 62017, offset 0, flags [DF], proto ICMP (1), length 84)
    10.10.1.5 > 139.59.209.225: ICMP echo request, id 27162, seq 0, length 64
        0x0000:  4500 0054 f241 4000 4001 e03b 0a0a 0105  E..T.A@.@..;....
        0x0010:  8b3b d1e1 0800 52c4 6a1a 0000 7c9c be84  .;....R.j...|...
        0x0020:  0000 0000 0000 0000 0000 0000 0000 0000  ................
        0x0030:  0000 0000 0000 0000 0000 0000 0000 0000  ................
        0x0040:  0000 0000 0000 0000 0000 0000 0000 0000  ................
[... cut ...]
        0x0000:  4500 0054 93cf 0000 3401 8aae 8b3b d1e1  E..T....4....;..
        0x0010:  0a0a 0105 0000 7f02 4825 0004 0744 3190  ........H%...D1.
        0x0020:  0000 0000 0000 0000 0000 0000 0000 0000  ................
        0x0030:  0000 0000 0000 0000 0000 0000 0000 0000  ................
        0x0040:  0000 0000 0000 0000 0000 0000 0000 0000  ................
        0x0050:  0000 0000                                ....
^C
10 packets captured
10 packets received by filter
0 packets dropped by kernel
```

aaronjg, 2020-07-12:
Okay - very strange that the src ip changed with mwan3 running and not. 180.112.126.158 v 180.112.50.137 

Also, it seems strange that this ip is not showing up in the routing tables. Did the routing tables change between posts? Before it looked like the packets were coming from 173.242.119.141. 

If your configuration is changing - it becomes difficult to diagnose because we can't cross reference with the previous data that you have posted.

wackejohn, 2020-07-12:
@aaronjg 
>Okay - very strange that the src ip changed with mwan3 running and not. 180.112.126.158 v 180.112.50.137

My ISP provide the dynamic pppoe connection, the ip addr will change on ervery reconnect. It was my mistake that I copied the previous log.

The right log:
```
root@OpenWrt:/# [  249.122255] output startIN= OUT=pppoe-wan SRC=180.112.50.137 DST=173.242.119.141 LEN=176 TOS=0x08 PREC=0x80 TTL=64 ID=29806 PROTO=UDP SPT=53766 DPT=51888 LEN=156 
[  249.136949] output endIN= OUT=pppoe-wan SRC=180.112.50.137 DST=173.242.119.141 LEN=176 TOS=0x08 PREC=0x80 TTL=64 ID=29806 PROTO=UDP SPT=53766 DPT=51888 LEN=156 MARK=0x3f00 
[  249.152521] postroute startIN= OUT=pppoe-wan SRC=180.112.50.137 DST=173.242.119.141 LEN=176 TOS=0x08 PREC=0x80 TTL=64 ID=29806 PROTO=UDP SPT=53766 DPT=51888 LEN=156 MARK=0x3f00 
[  249.168472] postroute endIN= OUT=pppoe-wan SRC=180.112.50.137 DST=173.242.119.141 LEN=176 TOS=0x08 PREC=0x80 TTL=64 ID=29806 PROTO=UDP SPT=53766 DPT=51888 LEN=156 MARK=0x3f00 
[  249.333152] output startIN= OUT=pppoe-wan SRC=180.112.50.137 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=29814 PROTO=UDP SPT=53766 DPT=51888 LEN=136 
[  249.347822] output endIN= OUT=pppoe-wan SRC=180.112.50.137 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=29814 PROTO=UDP SPT=53766 DPT=51888 LEN=136 MARK=0x3f00 
[  249.363379] postroute startIN= OUT=pppoe-wan SRC=180.112.50.137 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=29814 PROTO=UDP SPT=53766 DPT=51888 LEN=136 MARK=0x3f00 
[  249.379328] postroute endIN= OUT=pppoe-wan SRC=180.112.50.137 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=29814 PROTO=UDP SPT=53766 DPT=51888 LEN=136 MARK=0x3f00 
[  250.186954] output startIN= OUT=pppoe-wan SRC=180.112.50.137 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=29827 PROTO=UDP SPT=53766 DPT=51888 LEN=136 
[  250.201625] output endIN= OUT=pppoe-wan SRC=180.112.50.137 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=29827 PROTO=UDP SPT=53766 DPT=51888 LEN=136 MARK=0x3f00 
[  250.217190] postroute startIN= OUT=wana SRC=180.112.50.137 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=29827 PROTO=UDP SPT=53766 DPT=51888 LEN=136 MARK=0x3f00 
[  250.232704] postroute endIN= OUT=wana SRC=180.112.50.137 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=29827 PROTO=UDP SPT=53766 DPT=51888 LEN=136 MARK=0x3f00 
[  251.186778] output startIN= OUT=pppoe-wan SRC=180.112.50.137 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=29829 PROTO=UDP SPT=53766 DPT=51888 LEN=136 
[  251.201465] output endIN= OUT=pppoe-wan SRC=180.112.50.137 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=29829 PROTO=UDP SPT=53766 DPT=51888 LEN=136 MARK=0x3f00 
[  251.217046] postroute startIN= OUT=wana SRC=180.112.50.137 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=29829 PROTO=UDP SPT=53766 DPT=51888 LEN=136 MARK=0x3f00 
[  251.232565] postroute endIN= OUT=wana SRC=180.112.50.137 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=29829 PROTO=UDP SPT=53766 DPT=51888 LEN=136 MARK=0x3f00 
[  251.327170] output startIN= OUT=pppoe-wan SRC=180.112.50.137 DST=173.242.119.141 LEN=220 TOS=0x00 PREC=0x00 TTL=64 ID=29838 PROTO=UDP SPT=53766 DPT=51888 LEN=200 
[  251.341842] output endIN= OUT=pppoe-wan SRC=180.112.50.137 DST=173.242.119.141 LEN=220 TOS=0x00 PREC=0x00 TTL=64 ID=29838 PROTO=UDP SPT=53766 DPT=51888 LEN=200 MARK=0x3f00 
[  251.357411] postroute startIN= OUT=wana SRC=180.112.50.137 DST=173.242.119.141 LEN=220 TOS=0x00 PREC=0x00 TTL=64 ID=29838 PROTO=UDP SPT=53766 DPT=51888 LEN=200 MARK=0x3f00 
[  251.372925] postroute endIN= OUT=wana SRC=180.112.50.137 DST=173.242.119.141 LEN=220 TOS=0x00 PREC=0x00 TTL=64 ID=29838 PROTO=UDP SPT=53766 DPT=51888 LEN=200 MARK=0x3f00 
[  251.475861] output startIN= OUT=pppoe-wan SRC=180.112.50.137 DST=173.242.119.141 LEN=284 TOS=0x00 PREC=0x00 TTL=64 ID=29839 PROTO=UDP SPT=53766 DPT=51888 LEN=264 
[  251.490533] output endIN= OUT=pppoe-wan SRC=180.112.50.137 DST=173.242.119.141 LEN=284 TOS=0x00 PREC=0x00 TTL=64 ID=29839 PROTO=UDP SPT=53766 DPT=51888 LEN=264 MARK=0x3f00 
[  251.506101] postroute startIN= OUT=wana SRC=180.112.50.137 DST=173.242.119.141 LEN=284 TOS=0x00 PREC=0x00 TTL=64 ID=29839 PROTO=UDP SPT=53766 DPT=51888 LEN=264 MARK=0x3f00 
[  251.521614] postroute endIN= OUT=wana SRC=180.112.50.137 DST=173.242.119.141 LEN=284 TOS=0x00 PREC=0x00 TTL=64 ID=29839 PROTO=UDP SPT=53766 DPT=51888 LEN=264 MARK=0x3f00 
[  251.556966] output startIN= OUT=pppoe-wan SRC=180.112.50.137 DST=173.242.119.141 LEN=348 TOS=0x00 PREC=0x00 TTL=64 ID=29841 PROTO=UDP SPT=53766 DPT=51888 LEN=328 
[  251.571631] output endIN= OUT=pppoe-wan SRC=180.112.50.137 DST=173.242.119.141 LEN=348 TOS=0x00 PREC=0x00 TTL=64 ID=29841 PROTO=UDP SPT=53766 DPT=51888 LEN=328 MARK=0x3f00 
[  251.587199] postroute startIN= OUT=wana SRC=180.112.50.137 DST=173.242.119.141 LEN=348 TOS=0x00 PREC=0x00 TTL=64 ID=29841 PROTO=UDP SPT=53766 DPT=51888 LEN=328 MARK=0x3f00 
[  251.602711] postroute endIN= OUT=wana SRC=180.112.50.137 DST=173.242.119.141 LEN=348 TOS=0x00 PREC=0x00 TTL=64 ID=29841 PROTO=UDP SPT=53766 DPT=51888 LEN=328 MARK=0x3f00 
[  251.618711] output startIN= OUT=pppoe-wan SRC=180.112.50.137 DST=173.242.119.141 LEN=412 TOS=0x00 PREC=0x00 TTL=64 ID=29842 PROTO=UDP SPT=53766 DPT=51888 LEN=392 
[  251.633392] output endIN= OUT=pppoe-wan SRC=180.112.50.137 DST=173.242.119.141 LEN=412 TOS=0x00 PREC=0x00 TTL=64 ID=29842 PROTO=UDP SPT=53766 DPT=51888 LEN=392 MARK=0x3f00 
[... cut ...]
[  272.043384] output endIN= OUT=pppoe-wan SRC=180.112.50.137 DST=173.242.119.141 LEN=60 TOS=0x00 PREC=0x00 TTL=64 ID=31070 PROTO=UDP SPT=53766 DPT=51888 LEN=40 MARK=0x3f00 
[  272.058765] postroute startIN= OUT=pppoe-wan SRC=180.112.50.137 DST=173.242.119.141 LEN=60 TOS=0x00 PREC=0x00 TTL=64 ID=31070 PROTO=UDP SPT=53766 DPT=51888 LEN=40 MARK=0x3f00 
[  272.074541] postroute endIN= OUT=pppoe-wan SRC=180.112.50.137 DST=173.242.119.141 LEN=60 TOS=0x00 PREC=0x00 TTL=64 ID=31070 PROTO=UDP SPT=53766 DPT=51888 LEN=40 MARK=0x3f00
```

aaronjg, 2020-07-12:
Got it. So now we can see that that it is for some reason trying to route the encrypted vpn traffic back through the VPN. You can see that the log with mwan3 is much longer than the log without. After the first successful ping,  each ping attempt triggers a cascade of 17 packets, each 64 bytes longer than the previous.

Is your VPN perhaps connected on port 443? If so can you delete the rule titled 'https.'

wackejohn, 2020-07-13:
@aaronjg 
>Is your VPN perhaps connected on port 443? If so can you delete the rule titled 'https?'

Nope.  My VPN was configred on port 51888 udp, as the log showed, while the https is tcp port.

aaronjg, 2020-07-13:
Many VPNs use port 443 (https port) to avoid being blocked by ISP firewalls. Is any part of your connection going over 443 or https? You included it in the minimal config - so I was wondering what it is needed for.

Something still seems wrong from the IP tables log:
```
SRC:180.112.50.137 DEST: 173.242.119.141
```

From the routing table:
```
117.85.168.1 dev pppoe-wan proto kernel scope link src 117.85.170.4 
173.242.119.141 via 117.85.168.1 dev pppoe-wan proto static 
```

Do you have any idea why the packets are getting assigned a source address of 180.112.50.137.

Another thing that looks weird is your mwan3 config has both a `wan` and a `wana` device, but the `wana` device is not part of any used policy.  Can you confirm that this is in fact the minimal configuration to reproduce the bug?

wackejohn, 2020-07-13:
@aaronjg 
>Many VPNs use port 443 (https port) to avoid being blocked by ISP firewalls. Is any part of your connection going over 443 or https? 

I'm using the wireguard vpn, and when I use other vpns with tun device, the `ping -I ` actually working, so I open this issue.

>You included it in the minimal config - so I was wondering what it is needed for.

The https rule was generated by the mwan3, so it was one of the default rule of mwan3.

>117.85.168.1 dev pppoe-wan proto kernel scope link src 117.85.170.4 
>173.242.119.141 via 117.85.168.1 dev pppoe-wan proto static 

Because I capture the routing table and the log at the different time, so the ip addr was different, but this shouldn't matter, please ignore the different ip addr.

> Another thing that looks weird is your mwan3 config has both a wan and a wana device, but the wana device is not part of any used policy. Can you confirm that this is in fact the minimal configuration to reproduce the bug?

Yes, with the default rule of mwan3, the `ping -I` won't working until stop the mwan3 (or use the 4.14 kernel and below).

aaronjg, 2020-07-13:
> Because I capture the routing table and the log at the different time, so the ip addr was different, but this shouldn't matter, please ignore the different ip addr.

When things change like this, it makes it very difficult to help. To identify the issue, we have to look for anomolies and these changes look very much like something when wrong in mwan3.

> Yes, with the default rule of mwan3, the ping -I won't working until stop the mwan3 (or use the 4.14 kernel and below).
A "minimal configuration" is the smallest possible configuration which gives the problematic behavior. To get a minimal configuration, you need to delete sections of the configuration until you can no longer reproduce the bug. (Either things start working again, or it breaks because some key configuration is missing). Some of these sections appear extraneous (eg policies that are not part of any rule). Which is why I asked if it was the minimal configuration. 

Anyway, since you have added back the direct link route, and confirmed that the IP address from the log matches the correct outgoing IP address of your ppoe device - it appears that mwan3 is doing everything correctly. Unfortunately, this means that you have found a bug in wireguard or the kernel routing.

You can actually see it here:

```
[  250.186954] output startIN= OUT=pppoe-wan SRC=180.112.50.137 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=29827 PROTO=UDP SPT=53766 DPT=51888 LEN=136 
[  250.201625] output endIN= OUT=pppoe-wan SRC=180.112.50.137 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=29827 PROTO=UDP SPT=53766 DPT=51888 LEN=136 MARK=0x3f00 
[  250.217190] postroute startIN= OUT=wana SRC=180.112.50.137 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=29827 PROTO=UDP SPT=53766 DPT=51888 LEN=136 MARK=0x3f00 
[  250.232704] postroute endIN= OUT=wana SRC=180.112.50.137 DST=173.242.119.141 LEN=156 TOS=0x00 PREC=0x00 TTL=64 ID=29827 PROTO=UDP SPT=53766 DPT=51888 LEN=136 MARK=0x3f00 
```

So this packet gets marked correctly by iptables with 0x3f0.  You never shared the output of `ip rule list` from the smaller (minimal?) configuration. But assuming nothing crazy is going on here, that firewall mark should skip the tables mwan3 added and match the main routing table.

The main routing table is (with 117.85.170.4 replaced with 180.112.50.137 presumably, and the corresponding gateway also replaced.):
```
default via 117.85.168.1 dev pppoe-wan proto static 
10.10.1.0/24 dev wana proto static scope link metric 10 
117.85.168.1 dev pppoe-wan proto kernel scope link src 117.85.170.4 
173.242.119.141 via 117.85.168.1 dev pppoe-wan proto static 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 
```

So that packet should match the fourth rule in the table and be routed out of ppoe. Somehow it appears to not matching that rule and being routed out of wana - which causes it to go back through ppoe. 

So it appears that this is an issue upstream, either with wireguard or the kernel. You should try to create a minimal configuration without mwan3 to reproduce this, and post a bug report on the main openwrt project.  A minimal configuration will likely contain:
- One `iptable` command to mark the packets
- A few `ip route add ... table 1`  commands to create a new routing table. experiment with this to see which rules are needed to reproduce this
- One `ip rule add ... table 1` command to add a rule to the routing table.

If you can't get it with those rules, you may need a few more commands to reproduce the behavior - but try to recreate the behavior with as few commands as possible since that will make it easier for the openwrt developers to help out.

ptpt52, 2020-08-08:
@wackejohn  hi
in `ip route`
why not `default route` for `wana`? this is not correct
so in `ip route list table 9`
```
default dev wana scope link metric 30  <<== This line is not correct

10.10.1.0/24 dev wana proto static scope link metric 30 
31.40.214.127 via 180.115.231.1 dev pppoe-wan1 proto static metric 10 
112.1.96.1 dev pppoe-wan3 proto kernel scope link src 112.1.96.195 
112.3.84.1 dev pppoe-wan4 proto kernel scope link src 112.3.87.193 
114.228.153.1 dev pppoe-wan2 proto kernel scope link src 114.228.153.129 
173.242.119.141 via 180.115.231.1 dev pppoe-wan1 proto static metric 10 
180.115.231.1 dev pppoe-wan1 proto kernel scope link src 180.115.231.180 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 
192.168.2.0/24 dev vpn1 proto static scope link metric 110 
192.168.3.0/24 dev vpn1 proto static scope link metric 110 
192.168.150.0/24 dev wanb proto static scope link metric 40 
192.168.195.0/24 dev vpn0 proto static scope link metric 100 
192.168.195.2 dev vpn0 proto static scope link metric 100 
192.168.195.3 dev vpn0 proto static scope link metric 100 
192.168.195.4 dev vpn0 proto static scope link metric 100 
192.168.195.5 dev vpn0 proto static scope link metric 100 
192.168.195.6 dev vpn0 proto static scope link metric 100 
192.168.196.0/24 dev vpn1 proto static scope link metric 110 
192.168.196.2 dev vpn1 proto static scope link metric 110 
192.168.196.3 dev vpn1 proto static scope link metric 110 
```

It should be like: `default via 10.10.1.xx dev wana proto static metric 40`
you need to add it to `main` table and restart mwan3
and mwan3rtmon should copy it to `table 9`

and if you don't want all traffic go out via `wana`, you could set the metric large number

aaronjg, 2020-08-10:
> you need to add it to `main` table and restart mwan3
> and mwan3rtmon should copy it to `table 9`

mwan3 does not copy over the default routes, it just inserts its own.

> and if you don't want all traffic go out via wana, you could set the metric large number

With only one default route, changing the metric on the default route will not change the traffic routing

wackejohn, 2020-08-20:
Thanks @aaronjg, with this [pr](https://github.com/openwrt/packages/pull/13169) and add default route of wireguard interface to main routing table ,this issue is fixed now.

ptpt52, 2020-10-26:
ping -I should work, so it is bug in wireguard
@wackejohn  
Please try test my new fix for it
https://github.com/x-wrt/x-wrt/blob/master/package/network/services/wireguard/patches/100-skb-sock-sync-sk_bound_dev_if.patch

wackejohn, 2020-10-26:
@ptpt52 
Tried, but still have bugs working with two or more wireguard links.

The ttl was different from wana (121) to wanb (116):
```
root@HOME-Router_CZ:/etc/config# ping -I wana 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
64 bytes from 8.8.8.8: seq=0 ttl=121 time=135.858 ms
64 bytes from 8.8.8.8: seq=1 ttl=116 time=136.189 ms
64 bytes from 8.8.8.8: seq=2 ttl=116 time=136.097 ms
64 bytes from 8.8.8.8: seq=3 ttl=116 time=135.867 ms
64 bytes from 8.8.8.8: seq=4 ttl=116 time=137.416 ms
64 bytes from 8.8.8.8: seq=5 ttl=116 time=137.491 ms
64 bytes from 8.8.8.8: seq=6 ttl=116 time=135.676 ms
64 bytes from 8.8.8.8: seq=7 ttl=116 time=135.557 ms
64 bytes from 8.8.8.8: seq=8 ttl=116 time=135.830 ms
64 bytes from 8.8.8.8: seq=9 ttl=116 time=139.063 ms
64 bytes from 8.8.8.8: seq=10 ttl=116 time=135.592 ms
64 bytes from 8.8.8.8: seq=11 ttl=116 time=135.746 ms
^C
--- 8.8.8.8 ping statistics ---
12 packets transmitted, 12 packets received, 0% packet loss
round-trip min/avg/max = 135.557/136.365/139.063 ms
root@HOME-Router_CZ:/etc/config# ping -I wana 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
64 bytes from 8.8.8.8: seq=0 ttl=121 time=133.802 ms
64 bytes from 8.8.8.8: seq=1 ttl=116 time=136.936 ms
64 bytes from 8.8.8.8: seq=2 ttl=116 time=136.612 ms
64 bytes from 8.8.8.8: seq=3 ttl=116 time=138.964 ms
64 bytes from 8.8.8.8: seq=4 ttl=116 time=136.239 ms
^C
--- 8.8.8.8 ping statistics ---
5 packets transmitted, 5 packets received, 0% packet loss
round-trip min/avg/max = 133.802/136.510/138.964 ms
root@HOME-Router_CZ:/etc/config# ping -I wana 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
64 bytes from 8.8.8.8: seq=0 ttl=121 time=133.633 ms
64 bytes from 8.8.8.8: seq=1 ttl=116 time=136.060 ms
64 bytes from 8.8.8.8: seq=2 ttl=116 time=135.999 ms
64 bytes from 8.8.8.8: seq=3 ttl=116 time=136.289 ms
64 bytes from 8.8.8.8: seq=4 ttl=116 time=135.918 ms
64 bytes from 8.8.8.8: seq=5 ttl=116 time=135.588 ms
64 bytes from 8.8.8.8: seq=6 ttl=116 time=135.867 ms
64 bytes from 8.8.8.8: seq=7 ttl=116 time=136.036 ms
64 bytes from 8.8.8.8: seq=8 ttl=116 time=135.913 ms
^C
--- 8.8.8.8 ping statistics ---
9 packets transmitted, 9 packets received, 0% packet loss
round-trip min/avg/max = 133.633/135.700/136.289 ms
root@HOME-Router_CZ:/etc/config# mwan3 ifup wana
root@HOME-Router_CZ:/etc/config# ping -I wanb 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
64 bytes from 8.8.8.8: seq=0 ttl=116 time=136.261 ms
64 bytes from 8.8.8.8: seq=1 ttl=121 time=137.569 ms
64 bytes from 8.8.8.8: seq=2 ttl=121 time=135.139 ms
64 bytes from 8.8.8.8: seq=3 ttl=121 time=133.616 ms
64 bytes from 8.8.8.8: seq=4 ttl=121 time=134.030 ms
64 bytes from 8.8.8.8: seq=5 ttl=121 time=133.919 ms
64 bytes from 8.8.8.8: seq=6 ttl=121 time=134.292 ms
64 bytes from 8.8.8.8: seq=7 ttl=121 time=133.984 ms
64 bytes from 8.8.8.8: seq=8 ttl=121 time=139.795 ms
64 bytes from 8.8.8.8: seq=9 ttl=121 time=133.475 ms
64 bytes from 8.8.8.8: seq=10 ttl=121 time=135.308 ms
64 bytes from 8.8.8.8: seq=11 ttl=121 time=135.950 ms
64 bytes from 8.8.8.8: seq=12 ttl=121 time=134.733 ms
64 bytes from 8.8.8.8: seq=13 ttl=121 time=137.229 ms
64 bytes from 8.8.8.8: seq=14 ttl=121 time=134.602 ms
64 bytes from 8.8.8.8: seq=15 ttl=121 time=139.457 ms
64 bytes from 8.8.8.8: seq=16 ttl=121 time=133.962 ms
64 bytes from 8.8.8.8: seq=17 ttl=121 time=137.164 ms
64 bytes from 8.8.8.8: seq=18 ttl=121 time=135.839 ms
^C
--- 8.8.8.8 ping statistics ---
19 packets transmitted, 19 packets received, 0% packet loss
round-trip min/avg/max = 133.475/135.596/139.795 ms
root@HOME-Router_CZ:/etc/config# ifdown wanb
root@HOME-Router_CZ:/etc/config# ping -I wanb 8.8.8.8
ping: bad address 'wanb'
root@HOME-Router_CZ:/etc/config# ping -I wana 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
64 bytes from 8.8.8.8: seq=0 ttl=121 time=134.590 ms
64 bytes from 8.8.8.8: seq=1 ttl=121 time=134.660 ms
64 bytes from 8.8.8.8: seq=2 ttl=121 time=133.850 ms
64 bytes from 8.8.8.8: seq=3 ttl=121 time=140.986 ms
64 bytes from 8.8.8.8: seq=4 ttl=121 time=135.351 ms
64 bytes from 8.8.8.8: seq=5 ttl=121 time=133.815 ms
64 bytes from 8.8.8.8: seq=6 ttl=121 time=141.610 ms
64 bytes from 8.8.8.8: seq=7 ttl=121 time=133.842 ms
64 bytes from 8.8.8.8: seq=8 ttl=121 time=133.694 ms
64 bytes from 8.8.8.8: seq=9 ttl=121 time=133.862 ms
^C
--- 8.8.8.8 ping statistics ---
10 packets transmitted, 10 packets received, 0% packet loss
round-trip min/avg/max = 133.694/135.626/141.610 ms
root@HOME-Router_CZ:/etc/config# ifup wanb
root@HOME-Router_CZ:/etc/config# ifdown wana
root@HOME-Router_CZ:/etc/config# ping -I wanb 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
64 bytes from 8.8.8.8: seq=0 ttl=116 time=136.513 ms
64 bytes from 8.8.8.8: seq=1 ttl=116 time=137.159 ms
64 bytes from 8.8.8.8: seq=2 ttl=116 time=136.090 ms
64 bytes from 8.8.8.8: seq=3 ttl=116 time=137.653 ms
64 bytes from 8.8.8.8: seq=4 ttl=116 time=136.160 ms
64 bytes from 8.8.8.8: seq=5 ttl=116 time=136.246 ms
64 bytes from 8.8.8.8: seq=6 ttl=116 time=135.954 ms
64 bytes from 8.8.8.8: seq=7 ttl=116 time=136.978 ms
64 bytes from 8.8.8.8: seq=8 ttl=116 time=136.383 ms
64 bytes from 8.8.8.8: seq=9 ttl=116 time=135.880 ms
64 bytes from 8.8.8.8: seq=10 ttl=116 time=136.031 ms
^C
--- 8.8.8.8 ping statistics ---
11 packets transmitted, 11 packets received, 0% packet loss
round-trip min/avg/max = 135.880/136.458/137.653 ms
```

ptpt52, 2020-10-26:
@wackejohn  ttl is the issue?
Seems to be normal, in a real network environment

[11 more replies]

PR #13169 (2020-10-21): mwan3: mwan3track via default routing table and use procd from mwan3track & mwan3rtmon
Maintainer:  @feckert 
Compile tested: 19.07.3-ipq806x and snapshot-7673df79 MT76x8 
Run tested: 19.07.3 and snapshot - 7673df79

Description:
This P/R has a few fixes for the 19.07.3

1) Use the default routing tables for all mwan3track outgoing requests. This fixes some issues on the 5.4.x kernel where packets are not routed properly unless they go through a routing table that has the gateway. This also allows us to remove some hacks that were in place for ipv6 mwan3track
2) Comply with source based routin [... cut]