openwrt/packages #10714: NoDDos error when reboot!!!

S8D, 2019-12-07:
This page show error when upgrade package and reboot. To fix this error, reinstall NoDDos -.-
https://xxx/cgi-bin/luci/admin/status/noddos

Failed to execute template dispatcher target for entry '/admin/status/noddos'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'noddos/clients'.
A runtime error occurred: [string "/usr/lib/lua/luci/view/noddos/clients.htm"]:16: bad argument #1 to 'input' (/var/lib/noddos/DeviceProfiles.json: No such file or directory)
stack traceback:
	[C]: in function 'input'
	[string "/usr/lib/lua/luci/view/noddos/clients.htm"]:16: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function </usr/lib/lua/luci/template.lua:85>
	(tail call): ?
	/usr/lib/lua/luci/dispatcher.lua:920: in function </usr/lib/lua/luci/dispatcher.lua:919>

micmac1, 2019-12-08:
Hello @S8D,

The file `/var/lib/noddos/DeviceProfiles.json` aren't provided by the package. When the noddos init script starts it checks if the file exists, if not it starts getnoddosdeviceprofiles. Please do some basic checks and see if noddos gets started properly. If not try to find out where the problem comes from. Maybe run getnoddosdeviceprofiles manually to get more info.

Kind regards,
Seb

S8D, 2019-12-11:
Hello @micmac1 ,
getnoddosdeviceprofiles get error when download
https://www.noddos.io/config/DeviceProfiles.json.sha256 through wget/curl because they do not skip certificate checks
wget --no-check-certificate and curl -k
Please fix this error in next version.

micmac1, 2019-12-11:
Hi @S8D 

noddos package depends on wget and ca-bundle. So everything should be in place to download via https.

The problem seems to be with https://www.noddos.io. The certificate has expired.

Pinging maintainer @StevenHessing

micmac1, 2019-12-26:
Certificate got renewed and is valid till March 2020. Please try again. If solved consider closing this issue.

S8D, 2019-12-27:
Thanks you, i want to leech files on NoDDos's sever to my hosts.

S8D, 2020-02-19:
Recognized Clients
500 Internal Server Error

Sorry, the server encountered an unexpected error.

Failed to execute template dispatcher target for entry '/admin/status/noddos'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'noddos/clients'.
A runtime error occurred: [string "/usr/lib/lua/luci/view/noddos/clients.htm"]:25: bad argument #1 to 'ipairs' (table expected, got nil)
stack traceback:
	[C]: in function 'ipairs'
	[string "/usr/lib/lua/luci/view/noddos/clients.htm"]:25: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 
	(tail call): ?
	/usr/lib/lua/luci/dispatcher.lua:920: in function

anand2000zz, 2020-08-03:
hi, is the certificate renewed or any other way to response for  https://www.noddos.io/config/DeviceProfiles.json.sha256
 for  /var/lib/noddos/DeviceProfiles.json  from the script:  getnoddosdeviceprofiles

Strykar, 2020-12-19:
The noddos project should be considered dead/abandoned, see https://github.com/openwrt/packages/issues/14248
Please close this.

closed 2020-12-20 by neheb