openwrt/packages #10715: mosquitto-ssl install failed for conflicting

lincoln310, 2019-12-07:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

```
Formating code blocks by wrapping them with pairs of ```
```

micmac1, 2019-12-08:
Hi @lincoln310,

I don't think this is a bug, certainly not of mosquitto-ssl. You happen to have libopenssl installed and mostquito-ssl pulling in libopenssl1.1 as a dependency. Probably you have libopenssl from before it was packaged with API_VERSION (hence the soname suffix in libopenssl1.1).

You can probably just

opkg-remove -f libopenssl

and then install it again (so you get the version with the soname suffix).

Best regards,
Seb

lincoln310, 2019-12-08:
@micmac1 thanks, i will try it. I don't know which program used libopenssl, so whether it will cause some error if i reinstalled it with mosquitto. But still i will try it!

micmac1, 2019-12-08:
So list-installed lists libopenssl? Here's the depends from mosquitto-ssl:

`Depends: libc, librt, libopenssl1.1, libwebsockets-openssl`

libopenssl and libopenssl1.1 aren't the same, althought the latter PROVIDES the former.

BKPepe, 2020-02-22:
Closing due to inactivity of the submitter.

closed 2020-02-22 by BKPepe