openwrt/packages #10731: nlbwmon: Unable to save database: Invalid argument

fantom-x, 2019-12-09:
19.07-SNAPSHOT

The database does not save with `option database_compress '0'`. When the value is `1`, the database gets saved.

neheb, 2020-02-28:
Can you strace this?

fantom-x, 2020-02-28:
Do you mind helping with instructions?

neheb, 2020-02-28:
ps | grep nlbwmon to get the PID number

strace -p PID

Force the error to happen and see what strace says.

neheb, 2020-02-29:
Also, is this on a little or big endian system?

fantom-x, 2020-02-29:
It was on a GL-B1300 (ipq40xx).

neheb, 2020-02-29:
looks like little.

fantom-x, 2020-02-29:
```
--- SIGTERM {si_signo=SIGTERM, si_code=SI_USER, si_pid=1, si_uid=0} ---
munmap(0xb6f54000, 23)                  = 0
open("/etc/TZ", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 9
fstat64(9, {st_mode=S_IFREG|0644, st_size=23, ...}) = 0
mmap2(NULL, 23, PROT_READ, MAP_SHARED, 9, 0) = 0xb6f54000
close(9)                                = 0
stat64("/root/nlbwmon/20200201.db", 0xbee435d8) = -1 ENOENT (No such file or directory)
open("/root/nlbwmon/20200201.db", O_RDWR|O_CREAT|O_LARGEFILE, 0640) = 9
ftruncate64(9, 40)                      = 0
mmap2(NULL, 40, PROT_WRITE, MAP_SHARED, 9, 0) = -1 EINVAL (Invalid argument)
close(9)                                = 0
unlink("/root/nlbwmon/20200201.db")     = 0
open("/root/nlbwmon/20200201.db", O_RDWR|O_CREAT|O_LARGEFILE, 0640) = 9
ftruncate64(9, 40)                      = 0
mmap2(NULL, 40, PROT_WRITE, MAP_SHARED, 9, 0) = -1 EINVAL (Invalid argument)
close(9)                                = 0
unlink("/root/nlbwmon/20200201.db")     = 0
writev(2, [{iov_base="Unable to save database: Invalid"..., iov_len=42}, {iov_base=NULL, iov_len=0}], 2) = 42
unlink("/tmp/0.db")                     = -1 ENOENT (No such file or directory)
rt_sigaction(SIGINT, NULL, {sa_handler=0x15a4c, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0xb6f1e008}, 8) = 0
rt_sigaction(SIGTERM, NULL, {sa_handler=0x15a4c, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0xb6f1e008}, 8) = 0
rt_sigaction(SIGCHLD, NULL, {sa_handler=0xb6eaffc4, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0xb6f1e008}, 8) = 0
rt_sigaction(SIGCHLD, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0xb6f1e008}, NULL, 8) = 0
rt_sigaction(SIGPIPE, NULL, {sa_handler=SIG_IGN, sa_mask=[], sa_flags=0}, 8) = 0
close(4)                                = 0
epoll_ctl(-1, EPOLL_CTL_DEL, 5, NULL)   = -1 EBADF (Bad file descriptor)
close(6)                                = 0
close(5)                                = 0
exit_group(0)                           = ?
+++ exited with 0 +++

```

fantom-x, 2020-02-29:
```
epoll_pwait(4, [{EPOLLIN, {u32=164444, u64=164444}}], 10, 254018, NULL, 8) = 1
recvmsg(7, {msg_name={sa_family=AF_NETLINK, nl_pid=0, nl_groups=0x000001}, msg_namelen=12, msg_iov=[{iov_base={{len=148, type=0x100 /* NLMSG_??? */, flags=0x600 /* NLM_F_??? */, seq=0, pid=0}, "\x02\x00\x00\x00\x34\x00\x01\x80\x14\x00\x01\x80\x08\x00\x01\x00\x0a\xac\x24\x33\x08\x00\x02\x00\x0a\xac\x24\xc8\x1c\x00\x02\x80"...}, iov_len=16384}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 148
munmap(0xb6f54000, 23)                  = 0
open("/etc/TZ", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 9
fstat64(9, {st_mode=S_IFREG|0644, st_size=23, ...}) = 0
mmap2(NULL, 23, PROT_READ, MAP_SHARED, 9, 0) = 0xb6f54000
close(9)                                = 0
epoll_pwait(4, [{EPOLLIN, {u32=164444, u64=164444}}], 10, 249190, NULL, 8) = 1
recvmsg(7, {msg_name={sa_family=AF_NETLINK, nl_pid=0, nl_groups=0x000001}, msg_namelen=12, msg_iov=[{iov_base={{len=148, type=0x100 /* NLMSG_??? */, flags=0x600 /* NLM_F_??? */, seq=0, pid=0}, "\x02\x00\x00\x00\x34\x00\x01\x80\x14\x00\x01\x80\x08\x00\x01\x00\x7f\x00\x00\x01\x08\x00\x02\x00\x7f\x00\x00\x01\x1c\x00\x02\x80"...}, iov_len=16384}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 148
munmap(0xb6f54000, 23)                  = 0
open("/etc/TZ", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 9
fstat64(9, {st_mode=S_IFREG|0644, st_size=23, ...}) = 0
mmap2(NULL, 23, PROT_READ, MAP_SHARED, 9, 0) = 0xb6f54000
close(9)                                = 0
epoll_pwait(4, [{EPOLLIN, {u32=164444, u64=164444}}], 10, 249173, NULL, 8) = 1
recvmsg(7, {msg_name={sa_family=AF_NETLINK, nl_pid=0, nl_groups=0x000001}, msg_namelen=12, msg_iov=[{iov_base={{len=148, type=0x100 /* NLMSG_??? */, flags=0x600 /* NLM_F_??? */, seq=0, pid=0}, "\x02\x00\x00\x00\x34\x00\x01\x80\x14\x00\x01\x80\x08\x00\x01\x00\x0a\xac\x24\x33\x08\x00\x02\x00\x0a\xac\x24\xc8\x1c\x00\x02\x80"...}, iov_len=16384}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 148
munmap(0xb6f54000, 23)                  = 0
open("/etc/TZ", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 9
fstat64(9, {st_mode=S_IFREG|0644, st_size=23, ...}) = 0
mmap2(NULL, 23, PROT_READ, MAP_SHARED, 9, 0) = 0xb6f54000
close(9)                                = 0
epoll_pwait(4, [{EPOLLIN, {u32=164444, u64=164444}}], 10, 248182, NULL, 8) = 1
recvmsg(7, {msg_name={sa_family=AF_NETLINK, nl_pid=0, nl_groups=0x000001}, msg_namelen=12, msg_iov=[{iov_base={{len=148, type=0x100 /* NLMSG_??? */, flags=0x600 /* NLM_F_??? */, seq=0, pid=0}, "\x02\x00\x00\x00\x34\x00\x01\x80\x14\x00\x01\x80\x08\x00\x01\x00\x7f\x00\x00\x01\x08\x00\x02\x00\x7f\x00\x00\x01\x1c\x00\x02\x80"...}, iov_len=16384}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 148
munmap(0xb6f54000, 23)                  = 0
open("/etc/TZ", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 9
fstat64(9, {st_mode=S_IFREG|0644, st_size=23, ...}) = 0
mmap2(NULL, 23, PROT_READ, MAP_SHARED, 9, 0) = 0xb6f54000
close(9)                                = 0
epoll_pwait(4, [{EPOLLIN, {u32=164444, u64=164444}}], 10, 248166, NULL, 8) = 1
recvmsg(7, {msg_name={sa_family=AF_NETLINK, nl_pid=0, nl_groups=0x000004}, msg_namelen=12, msg_iov=[{iov_base={{len=196, type=0x102 /* NLMSG_??? */, flags=0, seq=0, pid=0}, "\x02\x00\x00\x00\x34\x00\x01\x80\x14\x00\x01\x80\x08\x00\x01\x00\x7f\x00\x00\x01\x08\x00\x02\x00\x7f\x00\x00\x01\x1c\x00\x02\x80"...}, iov_len=16384}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 196
munmap(0xb6f54000, 23)                  = 0
open("/etc/TZ", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 9
fstat64(9, {st_mode=S_IFREG|0644, st_size=23, ...}) = 0
mmap2(NULL, 23, PROT_READ, MAP_SHARED, 9, 0) = 0xb6f54000
close(9)                                = 0
epoll_pwait(4, [{EPOLLIN, {u32=164444, u64=164444}}], 10, 247943, NULL, 8) = 1
recvmsg(7, {msg_name={sa_family=AF_NETLINK, nl_pid=0, nl_groups=0x000004}, msg_namelen=12, msg_iov=[{iov_base={{len=196, type=0x102 /* NLMSG_??? */, flags=0, seq=0, pid=0}, "\x02\x00\x00\x00\x34\x00\x01\x80\x14\x00\x01\x80\x08\x00\x01\x00\x0a\xac\x24\x33\x08\x00\x02\x00\x0a\xac\x24\xc8\x1c\x00\x02\x80"...}, iov_len=16384}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 196
munmap(0xb6f54000, 23)                  = 0
open("/etc/TZ", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 9
fstat64(9, {st_mode=S_IFREG|0644, st_size=23, ...}) = 0
mmap2(NULL, 23, PROT_READ, MAP_SHARED, 9, 0) = 0xb6f54000
close(9)                                = 0
epoll_pwait(4, [{EPOLLIN, {u32=164444, u64=164444}}], 10, 247371, NULL, 8) = 1
recvmsg(7, {msg_name={sa_family=AF_NETLINK, nl_pid=0, nl_groups=0x000001}, msg_namelen=12, msg_iov=[{iov_base={{len=148, type=0x100 /* NLMSG_??? */, flags=0x600 /* NLM_F_??? */, seq=0, pid=0}, "\x02\x00\x00\x00\x34\x00\x01\x80\x14\x00\x01\x80\x08\x00\x01\x00\x0a\xac\x24\x33\x08\x00\x02\x00\x0a\xac\x24\xc8\x1c\x00\x02\x80"...}, iov_len=16384}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 148
munmap(0xb6f54000, 23)                  = 0
open("/etc/TZ", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 9
fstat64(9, {st_mode=S_IFREG|0644, st_size=23, ...}) = 0
mmap2(NULL, 23, PROT_READ, MAP_SHARED, 9, 0) = 0xb6f54000
close(9)                                = 0
epoll_pwait(4, [{EPOLLIN, {u32=164444, u64=164444}}], 10, 247355, NULL, 8) = 1
recvmsg(7, {msg_name={sa_family=AF_NETLINK, nl_pid=0, nl_groups=0x000001}, msg_namelen=12, msg_iov=[{iov_base={{len=148, type=0x100 /* NLMSG_??? */, flags=0x600 /* NLM_F_??? */, seq=0, pid=0}, "\x02\x00\x00\x00\x34\x00\x01\x80\x14\x00\x01\x80 [... cut ...]
rt_sigaction(SIGCHLD, NULL, {sa_handler=0xb6eaffc4, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0xb6f1e008}, 8) = 0
rt_sigaction(SIGCHLD, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0xb6f1e008}, NULL, 8) = 0
rt_sigaction(SIGPIPE, NULL, {sa_handler=SIG_IGN, sa_mask=[], sa_flags=0}, 8) = 0
close(4)                                = 0
epoll_ctl(-1, EPOLL_CTL_DEL, 5, NULL)   = -1 EBADF (Bad file descriptor)
close(6)                                = 0
close(5)                                = 0
exit_group(0)                           = ?
+++ exited with 0 +++

```

fantom-x, 2020-02-29:
Not related, but it is opening **"/etc/TZ"** way too often. I got many pages of that in just a few seconds.

neheb, 2020-02-29:
man page for the failure says:

EINVAL (Various platforms where the page size is not 4096 bytes.)  offset * 4096 is not a multiple of the system page size.

A small git grep indicates the page size to be 2048. This might be a musl bug honestly.

I have no idea how to fix.

neheb, 2020-02-29:
/etc/TZ reading comes from https://github.com/openwrt/openwrt/blob/master/toolchain/musl/patches/110-read_timezone_from_fs.patch

I have no idea why that patch is even there. /etc/TZ is a uClibc extension. It would make more sense to set the TZ environment variable globally.

fantom-x, 2020-02-29:
This router is using JFFS2 compressed partition in case that provides any hints

fantom-x, 2020-02-29:
It works with compressed files, so why would page size break things only in one case.

neheb, 2020-02-29:
Because nlbwmon is using mmap when compression is disabled. mmap is the problem.

jow-, 2020-02-29:
The `mmap()` syscall is not supported by JFFS2 filesystems. Either put the database file on an ext4 filesystem (e.g. an USB thumb drive) or enable compression as a workaround to inhibit the use of memory mapping.

fantom-x, 2020-02-29:
> enable compression

Yeah, that is what I did. It is just compressing files on a compressed file systems is a waste and slows things down, so I wanted to avoid the overhead. Otherwise, it is all working fine.

neheb, 2020-02-29:
@jow- could be a good idea to make it conditional. Or throw an error indicating that JFFS2 is not supported.

fantom-x, 2020-02-29:
Better yet, log a warning and default to compressed.

neheb, 2020-03-09:
I'll close this for now. This should be continued at https://github.com/jow-/nlbwmon

closed 2020-03-09 by neheb