openwrt/packages #10755: yggdrasil: use sighup for reload instead of using restart

stephen304, 2019-12-10:
Maintainer: @wfleurant
Environment: ARM64

Description:
If possible, it would be nice to do the equivalent of `ExecReload=/bin/kill -HUP $MAINPID` when reload is run, instead of doing restart, which does stop and then start, which will remove the tun interface and destroy any modifications made to the interface. This will bring it into line with how the yggdrasil systemd service operates (https://github.com/yggdrasil-network/yggdrasil-go/blob/master/contrib/systemd/yggdrasil.service)

wfleurant, 2019-12-15:
`kill -HUP $(pidof yggdrasil)`  

but yggdrasil says  

`Reloading config at runtime is only possible with -useconffile`  

I think we should pursue reloading with sighup. we would need to test/ask how sighup handles a bunch of things. The first config change we could test is changing yggdrasil interface name -- what happens?

wfleurant, 2023-10-20:
this looks like it belongs in PR https://github.com/openwrt/packages/pull/20626

Saiv46, 2025-07-13:
@wfleurant Would you like to close this issue as completed? (See reply above)

wfleurant, 2025-07-15:
ok can we close it @stephen304