openwrt/packages #10783: rstrip.sh breaks private rpaths via PATCHELF --set-rpath

Andy2244, 2019-12-13:
Maintainer: @jow- @neheb 
Environment: all

Description:
Samba4 by default uses two rpaths, one public (/usr/lib) and a private (/usr/lib/samba), so all bins have those set to: `/usr/lib/samba:/usr/lib`
The default RSTRIP macro will break those paths via rstrip.sh to:
`/usr/lib/samba XXXXXXXX `

This results in broken bin's and my temp fix is this:
`RSTRIP:=:`

Just wanted to ask if this is a bug we want to fix, by allowing multiple rpaths in bins, if they are defined or keep as samba4 specific quirk?

PS: The main use-case is to avoid/circumvent build issues and let samba use some of its own bundled system lib versions, like tevent or tdb, com_err and than make those private to avoid breaking other packages.

jow-, 2019-12-16:
The `rstrip.sh` script is supposed to retain paths starting with `/lib` or `/usr/lib` so I'm not sure whats wrong there. Maybe try adding sone debug echos to `rstrip.sh`

neheb, 2020-01-17:
RSTRIP:=: is used in multiple places. I assume for similar reasons.

In any case, if this is really a bug, it belongs at https://bugs.openwrt.org

closed 2020-01-17 by neheb