openwrt/packages #10807: ModemManager : In /etc/config/network, support designating a /dev/ device, instead of /sys/devices/...

jonesmz, 2019-12-17:
Maintainer: @nickberry17

Description:

While the rational for primarily supporting paths under /sys/ makes sense for the "device" key in /etc/config/network when there are more than one modem on an OpenWRT machine. In many use cases there is only one modem, and as such using the kernel assigned device name, such as /dev/cdc-wdm0, makes just as much sense, and is easier on the end user.

The modemmanager netifd script can detect if it's given a path under /dev/, or under /sys/, and adapt /dev/ paths to the appropriate /sys/ paths for end user convenience.

Another way to look at this for supporting different models of OpenWRT compatible mainboards, which each both expose the modem under different paths in /sys/, but each model exposes the modem as /dev/cdc-wdm0, which can simplify firmware configuration.

nickberry17, 2019-12-17:
I think you can already use a /dev/ device.  It's just not generally advised because, as you know, that is not the real sysfs path to the device, and could change.

jonesmz, 2019-12-17:
At the moment, the UCI key "device" only permits the full sysfs path.

Using a /dev/ path doesn't work. mmcli throws an error, and the netifd modemmanager proto script uses the text from UCI directly in the --modem parameter to mmcli

Thanks for the scriptlet though :-)

jonesmz, 2019-12-18:
Currently I'm using this script function

```
# translate the name of an object in /dev/ to its appropriate entry in /sys
lookup_sysfs_path() {
	local name="${1}"
	if [ -z "${name}" ]; then
		if [ "${name:0:1}" = '/' ]; then
			echo ${name}
		else
			readlink -f /sys/class/usbmisc/$name/../../../
		fi
	fi
}
```

And then changing

```
json_get_vars device apn username password pincode iptype metric
```

to 

```
json_get_vars device apn username password pincode iptype metric
device=$(lookup_sysfs_path ${device})
```

And so far it seems happy, but I have additional testing to do.

jonesmz, 2019-12-18:
A slightly more robust method

```
mm_find_physdev_sysfs_path() {
        local tmp_path="$1"

        while true; do
                tmp_path=$(dirname "${tmp_path}")

                # avoid infinite loops iterating
                [ -z "${tmp_path}" ] || [ "${tmp_path}" = "/" ] && return

                # the physical device will be that with a idVendor and idProduct pair of files
                [ -f "${tmp_path}"/idVendor ] && [ -f "${tmp_path}"/idProduct ] && {
                        tmp_path=$(readlink -f "$tmp_path")
                        echo "${tmp_path}"
                        return
                }
        done
}

# translate the name of an object in /dev/ to its appropriate entry in /sys
lookup_sysfs_path() {
        local name="${1}"
        if [ ! -z "${name}" ]; then
                if [ "${name:0:1}" = '/' ]; then
                        echo ${name}
                else
                        local wait_seconds="10"
                        until [ $((wait_seconds--)) -eq 0 ] || [ -e "/sys/class/usbmisc/$name" ] ; do sleep 1; done
                        mm_find_physdev_sysfs_path `readlink -f /sys/class/usbmisc/$name`
                fi
        fi
}
```

aleksander0m, 2020-05-15:
The sysfs path is associated to the physical layout of the hardware. A modem that is exposed in a given sysfs path will always have the same sysfs path as long as it's not plugged in e.g. a different USB slot. If multiple modems are available, each of them will have its own stable sysfs path.

The /dev name is assigned by the kernel during runtime, and it MAY NOT be always the same. It depends on whether other modems are available and detected before, and sometimes, when the modem resets itself, it may also happen that the cdc-wdmX number bumps because the previous port exposed isn't fully cleared up. Do not rely on the /dev name, and do not rely on the wwan interface name either!

If this was solely my decision, I would not allow using /dev names in /etc/config/network, because it's not robust enough in all possible scenarios. Using the sysfs path is the most robust solution in all cases.

My question would be, why do you want to use /dev names? What is the main reason?

jonesmz, 2020-05-15:
> Another way to look at this for supporting different models of OpenWRT compatible mainboards, which each both expose the modem under different paths in /sys/, but each model exposes the modem as /dev/cdc-wdm0, which can simplify firmware configuration.

aleksander0m, 2020-05-15:
> Another way to look at this for supporting different models of OpenWRT compatible mainboards, which each both expose the modem under different paths in /sys/, but each model exposes the modem as /dev/cdc-wdm0, which can simplify firmware configuration.

ModemManager does not try to use one single port to control the modem. The fact that you would specify one of the ports of the modem to refer to it, e.g. cdc-wdm0 or ttyUSB3 wouldn't mean that ModemManager would use that port to control the device. ModemManager doesn't expect configuration "per port", it expects configuration "per device". Not even the data port used during connection is fixed in ModemManager, that is decided during runtime based on different things.

Also, as said, even in one single setup with one single modem, there is truly no guarantee that the modem will always be exposed with the same cdc-wdm port name. Referring the modem by a fixed cdc-wdm or ttyUSB or ttyACM port name is not a secure solution. Referring the modem by its sysfs path is secure, as the sysfs path will never change, as long as the hardware layout doesn't change.

Saying that referring to a modem by its cdc-wdm interface name in the configuration would allow the same setting to be applied in different hardware layouts, is a flawed logic IMO. What about the APN? Will that also be the same in different hardwares and due to that we can fix it to some value? The configuration of a given modem in a given setup cannot be used in a different setup with a different modem; it may work in the same way it may not work.

I understand that this way of configuring the device is very different to other modem management protocol handlers in netifd; but it was truly done on purpose.

If you want to build firmware that can be accommodated on different hardware platforms all sharing the same modem settings but with different sysfs paths, a simple uci-defaults script would help to handle that very very easily, just detecting which is the hardware layout present during the first boot after flashing openwrt, and configuring the modem based on that.

feckert, 2023-09-13:
I think the reasoning should be enough why the `sysfspath` is better than a `dev` path.

closed 2023-09-13 by feckert